thesofproject/sof #6935: [BUG][MTL][FW]: mtl-003 release FW - RVP speaker and headset playback resulting distorted audio

udaymb, 2023-01-11:
**Describe the bug**
Headset and speaker playback with MTL FW release - [MTL-003 2023WW02.zip](https://github.com/thesofproject/sof/files/10390077/MTL-003.2023WW02.zip) is resulting in distorted audio

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)
**command line**
Speaker - aplay -D hw:0,1 -c 2 -r 48000 -f s16_le <audio file>
Headset - aplay -D hw:0,0 -c 2 -r 48000 -f s16_le <audio file> 
**UI**
Login to UI and play youtube video or audio or video files

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Always reproducible

Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.
Consistently reproducible

**Expected behavior**
Audio should be clear with playback on speaker and headset

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
High priority - need fix to verify the features

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA} - https://github.com/intel-innersource/os.chrome.internal.kernel-bleeding-edge - branch - mtl - commit 6e8e0e3eb7191c582d79f398835c92a62c87725c
    * SOF: {SHA} - [MTL-003 2023WW02.zip](https://github.com/thesofproject/sof/files/10390077/MTL-003.2023WW02.zip)

md5sum /lib/firmware/intel/sof-ipc4/mtl/sof-mtl.ri
a58034d27305a67aeecd5523bc46e998  /lib/firmware/intel/sof-ipc4/mtl/sof-mtl.ri

2) Name of the topology file
    * Topology: {FILE} - [sof-mtl-max98357a-rt5682.tplg.zip](https://github.com/thesofproject/sof/files/10390126/sof-mtl-max98357a-rt5682.tplg.zip) -  9c077c288605e8c581c6d25ef131a8dc  sof-mtl-max98357a-rt5682.tplg

md5sum /lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682.tplg
9c077c288605e8c581c6d25ef131a8dc  /lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682.tplg

3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM} - MTL

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
captured the speaker playback audio - [hs-cap-mtl-003.zip](https://github.com/thesofproject/sof/files/10390295/hs-cap-mtl-003.zip)
there are no errors observed in dmesg - [dmesg-mtl-003.log](https://github.com/thesofproject/sof/files/10390315/dmesg-mtl-003.log)

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

keqiaozhang, 2023-01-12:
We also observed this issue on MTL-NOCODEC platform. https://github.com/thesofproject/sof/issues/6837

mengdonglin, 2023-01-17:
@udaymb Are you good to close this issue with the hot fixes?

closed 2023-02-01 by mengdonglin