thesofproject/sof #6947: [BUG][MTL] tplg failed to load on MTL-SDW

keqiaozhang, 2023-01-12:
**Describe the bug**
This is a regression and caused by:
(https://github.com/thesofproject/sof/commit/f963fa0e49c2a29835ae4a9bad8bb01849b7a920)
```
commit f963fa0e49c2a29835ae4a9bad8bb01849b7a920
Author: Bard Liao <yung-chuan.liao@linux.intel.com>
Date:   Mon Dec 19 14:19:14 2022 +0800

    topology2: set NUM_HDMIS=3 topologies

    Set NUM_HDMIS=3 for all topologies.

    Signed-off-by: Bard Liao <yung-chuan.liao@linux.intel.com>
```

**DMESG**
```
[   75.275539] kernel: snd_sof:sof_widget_ready: sof-audio-pci-intel-mtl 0000:00:1f.3: tplg: widget 17 (copier.host.50.1) is ready [type: 23, pipe: 50, pins: 1 / 4, stream: HDMI1 Playback]
[   75.275563] kernel: snd_sof:sof_widget_ready: sof-audio-pci-intel-mtl 0000:00:1f.3: tplg: widget 18 (pipeline.51) is ready [type: 32, pipe: 51, pins: 0 / 0, stream: HDMI1 stream]
[   75.275580] kernel: snd_sof:sof_widget_ready: sof-audio-pci-intel-mtl 0000:00:1f.3: tplg: widget 19 (copier.HDA.51.1) is ready [type: 27, pipe: 51, pins: 1 / 4, stream: iDisp1]
[   75.275600] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: can't connect DAI copier.HDA.51.1 stream iDisp1
[   75.275641] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : copier.HDA.51.1 stream iDisp1
[   75.275683] kernel: sof_sdw sof_sdw: ASoC: failed to load widget copier.HDA.51.1
[   75.275708] kernel: sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[   75.275831] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22
[   75.275875] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to load DSP topology -22
[   75.275905] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[   75.276008] kernel: sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[   75.276745] kernel: sof_sdw sof_sdw: snd_soc_register_card failed -22
[   75.276871] kernel: sof_sdw: probe of sof_sdw failed with error -22
```

**To Reproduce**
Check the kernel log after booting

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: main/https://github.com/thesofproject/sof/commit/f963fa0e49c2a29835ae4a9bad8bb01849b7a920
2) Name of the topology file
    * Topology: sof-mtl-rt711-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-SDW

keqiaozhang, 2023-01-12:
A potential fix: https://github.com/thesofproject/sof/pull/6946

miRoox, 2023-01-13:
We found this issue also happened on ADLP_RVP_SDW + IPC4

miRoox, 2023-01-19:
This issue has been fixed by https://github.com/thesofproject/sof/pull/6946, but we found a new issue https://github.com/thesofproject/sof/issues/6968

closed 2023-01-19 by miRoox