thesofproject/sof #6963: [Need Info] Pipeline creation failing on Apollo Lake / Gemini Lake after booting SOF

coolstar, 2023-01-18:
Hi there, I am writing an audio driver for a non-Linux OS and am hoping to use SOF (mainly for Gemini Lake chromebooks, but my test hardware is Apollo Lake chromebooks)

The intel-signed APL sof binary does have working audio on my chromebook in Linux, however with my own driver on the other OS, I'm noticing a strange issue on Apollo Lake

After booting SOF (and successfully getting the FW_READY message)  I'm attempting to create a pipeline with the following parameters:

```c
	uint32_t pipeline_id = 1;
	uint32_t comp_id = 6;

	struct sof_ipc_pipe_new pipe_msg;
	bzero(&pipe_msg, sizeof(pipe_msg));

	pipe_msg.hdr.cmd = SOF_IPC_GLB_TPLG_MSG | SOF_IPC_TPLG_PIPE_NEW;
	pipe_msg.hdr.size = sizeof(pipe_msg);

	pipe_msg.comp_id = comp_id;
	pipe_msg.pipeline_id = pipeline_id;
	pipe_msg.sched_id = comp_id - 1;

	pipe_msg.period = 1000;
	pipe_msg.period_mips = 5000;

	pipe_msg.time_domain = SOF_TIME_DOMAIN_TIMER;

	struct sof_ipc_comp_reply reply;
	int status = ipc3_tx_msg(&pipe_msg, sizeof(pipe_msg), &reply, sizeof(reply), FALSE);
	printf("Create Pipe? 0x%x %d\n", status, reply.rhdr.error);
```

However, even though the IPC handler returns success (you can see the status parameter returns 0, SOF is returning -22)

My driver code is working on Comet Lake / Jasper Lake / Tiger Lake / Alder Lake, so not sure if there's anything special needed for APL / GLK. The same issue happens on a Gemini Lake chromebook w/ community-signed firmware (though unfortunately I don't have GLK hardware on hand so it's not as easy for me to test it there)

[sof-driver-log.txt](https://github.com/thesofproject/sof/files/10445062/sof-driver-log.txt)

lgirdwood, 2023-01-18:
@coolstar APL/GLK is an older version of DSP IP than the others and so there are differences around boot, FW loading and performance.  That being said, the IPC3 mechanism used between the SOF Linux driver and FW is exactly the same on all current Chromebooks.

Some things I would check
1) The SOF Linux driver is NOT sending a PM gate D3 IPC (it does this automatically after 3 seconds of idle, can be switched OFF as kernel module param IIUC).
2) The SOF Linux driver has a IPC log/trace mode, I would enable this to see a normal IPC flow for topology load (which can the be copied).

I also have a WIP userspace driver in my ```lrg/topic/plugin``` branch that can load topology and play audio as an ALSA plugin. This can also be used to dump the required IPC flows. See it's ReadMe.md for usage, but it does load topology and playback works on test topologies.

plbossart, 2023-01-18:
@coolstar the IPC mechanism and registers changed between GLK (cAVS1.5) and CML..TGL (cAVS 1.8+), so it's possible that your driver does not use the 'right' IPC registers. 

Unless you have a specific need for GLK I would stick to more recent platforms which have better support. Specifically TGL is the only case where multi-core is supported, and what most developers use.

coolstar, 2023-01-19:
Seems like the issue was the memcpyto / memcpyfrom functions in my driver. Seems Comet Lake+ is fine with the DSP being read/written 4 bytes at a time, but Apollo Lake / Gemini Lake needs 1 byte to be written at a time. I guess copying 4 bytes at a time may have been mangling the input going to the DSP?

SOF is working now for me on APL, albeit only with dmic and speakers since it appears SSP1 is not configured properly (even though I'm using the configuration from the topology, there's no errors reported, just no sound). I'll have someone test my driver on GLK and see if SOF is working correctly there.

coolstar, 2023-01-20:
Seems like the issue with headphones on Apollo Lake was since it's missing the SSP mclk id in its topology. After fixing it (basing off GLK's topology), it now works on both :D

Working SOF port in action on Apollo Lake chromebook (and I tested that it worked on Gemini Lake earlier)! I know this will make quite a lot of chromebook modders happy.

https://www.youtube.com/watch?v=mtPHPJKPp6o

closed 2023-01-20 by coolstar