thesofproject/sof #6968: [BUG] failed to set DMIC configuration on MTLP_RVP_SDW

miRoox, 2023-01-19:
**Describe the bug**
After https://github.com/thesofproject/sof/issues/6947 got fixed, we meet an IO error when capturing on the DMIC on MTLP_RVP_SDW. If we check the mtrace, it shows `Failed to set the requested configuration.`

**To Reproduce**
```
arecord  -Dhw:0,10 -r 48000 -c 4 -f S32_LE -d 3 /dev/null -v -q
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/bfdc6b889069
    * SOF: https://github.com/thesofproject/sof/commit/4cc9d5418b0e
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-rt711-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_SDW

**Screenshots or console output**

**mtrace**
```
00:01:19.067,786] <inf> ipc: ipc_cmd: rx	: 0x40000004|0x306
[00:01:19.067,825] <inf> ipc: buffer_new: buffer new size 0x600 id 0.0 flags 0x0
[00:01:19.067,868] <inf> dai_intel_dmic: dai_dmic_probe: dmic_probe()
[00:01:19.067,891] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[00:01:19.067,901] <inf> ipc: dai_config: comp:0 0x40000 dai_config() dai type = 2 index = 0 dd 0x400efb40
[00:01:19.067,915] <inf> dai_intel_dmic: dai_dmic_set_config: dmic_set_config()
[00:01:19.067,931] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt: dmic_set_config_nhlt(): OUTCONTROL0 = 003a0003
[00:01:19.067,938] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   tie=0, sip=0, finit=0, fci=0
[00:01:19.067,945] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   bfth=3, of=2, ipm=4, th=3
[00:01:19.067,953] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   ipms1=0, ipms2=0, ipms3=0, ipms4=0
[00:01:19.067,958] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   ipms_mode=0
[00:01:19.067,966] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt: dmic_set_config_nhlt(): OUTCONTROL1 = 003a0003
[00:01:19.067,973] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   tie=0, sip=0, finit=0, fci=0
[00:01:19.067,980] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   bfth=3, of=2, ipm=4, th=3
[00:01:19.067,988] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   ipms1=0, ipms2=0, ipms3=0, ipms4=0
[00:01:19.067,993] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt:   ipms_mode=0
[00:01:19.068,001] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt: dmic_set_config_nhlt(): OUTCONTROL0 = 023a0003
[00:01:19.068,013] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt: dmic_set_config_nhlt(): clkdiv = 16, mcic = 25, mfir_a = 2, len = 101
[00:01:19.068,033] <inf> dai_intel_dmic_nhlt: dai_dmic_set_config_nhlt: dmic_set_config_nhlt(): clkdiv = 16, mcic = 25, mfir_a = 2, len = 101
[00:01:19.068,050] <err> dai_intel_dmic_nhlt: nhlt_dmic_dai_params_get(): Illegal IPM PDM controllers count 4
[00:01:19.068,058] <err> dai_intel_dmic: dmic_set_config(): Failed to set the requested configuration.
[00:01:19.068,083] <err> copier: comp:0 0x40000 failed to create dai
[00:01:19.068,088] <err> copier: unable to create dai
[00:01:19.068,106] <err> ipc: error: failed to init module 4 : 0
[00:01:19.068,111] <err> ipc: ipc4: MODULE_MSG failed with err 104
```

**dmesg**
```
[  949.147341] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget copier.DMIC.101.1 instance 0 - pipe 101 - core 0
[  949.147359] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x40000004|0x306: MOD_INIT_INSTANCE [data size: 3096]
[  949.148499] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x60000068|0x306: MOD_INIT_INSTANCE
[  949.148530] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 104 - Other failure of module instance initialization request
[  949.148813] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x40000004|0x306
[  949.148907] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to create module copier.DMIC.101.1
[  949.149032] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  949.149471] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x32000000|0x0: GLB_DELETE_PIPELINE
[  949.149546] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  949.149574] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: widget pipeline.101 freed
[  949.149587] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
```

miRoox, 2023-01-20:
Today's daily test started before #6967 got merged, let's keep an eye on it.

fredoh9, 2023-01-23:
Working fine after https://github.com/thesofproject/sof/pull/6967

PR #6967 (2023-01-19): topology2: sof-mtl-rt711-4ch: specify PLATFORM=mtl
It worked because we defined DMIC_DRIVER_VERSION 3 as the default value. We have to specify PLATFORM=mtl when the default value was removed.

Fixes: #6968
closed 2023-01-23 by fredoh9