thesofproject/sof #6975: [BUG] FW reported error: 3018 - Unknown for 0xe070000|0x0 (GLB_CHAIN_DMA) when multiple-pause-resume on TGLU_UP_HDA

miRoox, 2023-01-20:
**Describe the bug**
After we enabled Chain DMA on TGL (https://github.com/thesofproject/sof/pull/6656), multiple-pause-resume failed at the first round on TGLU_UP_HDA + IPC4.

**To Reproduce**
```
TPLG=/lib/firmware/intel/avs-tplg/sof-hda-generic.tplg MODEL=TGLU_UP_HDA_IPC4ZPH ~/sof-test/test-case/multiple-pause-resume.sh -r 1
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/cff3bce51a58
    * SOF: https://github.com/thesofproject/sof/commit/507ad53b4830
2) Name of the topology file
    * Topology: avs-tplg/sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_UP_HDA

**Screenshots or console output**

**console output**
```
2023-01-19 22:31:22 UTC [REMOTE_INFO] HDMI1 to command: aplay -D hw:0,3 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
spawn aplay -D hw:0,3 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
2023-01-19 22:31:22 UTC [REMOTE_INFO] HDMI2 to command: aplay -D hw:0,4 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
2023-01-19 22:31:22 UTC [REMOTE_INFO] wait for expect process finished
spawn aplay -D hw:0,4 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
Hardware PCM card 0 'sof-hda-dsp' device 4 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0

#+                                                 | 00%
(1/50) pcm'HDMI2' cmd'aplay' id'1': Wait for 36 ms before pause
Hardware PCM card 0 'sof-hda-dsp' device 3 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0

#+                                                 | 00%
(1/50) pcm'HDMI1' cmd'aplay' id'0': Wait for 41 ms before pause
 
=== PAUSE ===                                                            aplay: do_pause:1570: pause push error: File descriptor in bad state
                                                                          

#+                                                 | 00%
#+                                                 | 00%
#+                                                 | 00%
(1/50) pcm'HDMI2' cmd'aplay' id'1': Wait for 34 ms before pause

#+                                                 | 00%
#+                                                 | 00% 
#+                                                 | 00%
=== PAUSE ===                                                            
(1/50) pcm'HDMI1' cmd'aplay' id'0': Wait for 33 ms before pause
 
(1/50) pcm'HDMI1' cmd'aplay' id'0': Wait for 32 ms before resume

#+                                                 | 00%
=== PAUSE ===                                                            
(1/50) pcm'HDMI2' cmd'aplay' id'1': Wait for 48 ms before pause
 aplay: do_pause:1581: pause release error: Invalid argument
                                                                          
 
(1/50) pcm'HDMI2' cmd'aplay' id'1': Wait for 24 ms before resume
  aplay: do_pause:1581: pause release error: Invalid argument
```

**dmesg**
```
[ 3524.580444] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 4 state: 4
[ 3524.580448] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0xe070000|0x0: GLB_CHAIN_DMA
[ 3524.580732] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x2e000bca|0x0: GLB_CHAIN_DMA
[ 3524.580739] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 3018 - Unknown
[ 3524.580758] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0xe070000|0x0
[ 3524.580766] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 3524.580771] kernel:  HDMI1: ASoC: trigger FE cmd: 4 failed: -22
```

**mtrace**
```
[    0.054068] <inf> ll_schedule: num_tasks 2 total_num_tasks 2
[    0.094355] <inf> ipc: rx	: 0xe070000|0x0
[    0.094366] <inf> chain_dma: comp:128 0x800000 chain_task_start(), host_dma_id = 0x00000000
[    0.094375] <err> chain_dma: comp:128 0x800000 chain_task_start(), bad state transition
[    0.094381] <err> ipc: ipc4: FW_GEN_MSG failed with err 3018
```

kv2019i, 2023-01-20:
@mwasko @makarukp Assigning to you. This started with merging  https://github.com/thesofproject/sof/pull/6656

kv2019i, 2023-01-20:
Setting P1 as this impacts CI (multiple tests failing for all pull requests).

mwasko, 2023-01-20:
@kv2019i, @miRoox do we know if it is TGL only issue or does it reproduce on MTL configuration as well?

miRoox, 2023-01-20:
> do we know if it is TGL only issue or does it reproduce on MTL configuration as well?

It's TGL only.

makarukp, 2023-01-20:
Working on local reproduction in python. Probably it is problem with task state transitions but it is not yet confirmed.

kv2019i, 2023-01-24:
No longer seen on CI due to #6979 , but issue still likely to happen on all Intel IPC4 platforms where HDMI is enabled (or other user of Chain-DMA).

miRoox, 2023-02-14:
This isssue has been fixed by https://github.com/thesofproject/sof/pull/6983, tested with https://github.com/thesofproject/sof/pull/7073

closed 2023-02-14 by miRoox