thesofproject/sof #6976: [BUG] DSP is not suspended after playback is stopped 15s on TGLU_UP_HDA

miRoox, 2023-01-20:
**Describe the bug**
DSP is not suspended after playback is stopped 15s on TGLU_UP_HDA.

**To Reproduce**
```
TPLG=/lib/firmware/intel/avs-tplg/sof-hda-generic.tplg MODEL=TGLU_UP_HDA_IPC4ZPH ~/sof-test/test-case/check-runtime-pm-status.sh -l 1
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/cff3bce51a58
    * SOF: https://github.com/thesofproject/sof/commit/507ad53b4830
2) Name of the topology file
    * Topology: avs-tplg/sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_UP_HDA

**Screenshots or console output**

**console log**
```
2023-01-19 22:32:26 UTC [REMOTE_COMMAND] aplay -D hw:0,3 -r 48000 -c 2 -f S16_LE /dev/zero -q
2023-01-19 22:32:28 UTC [REMOTE_INFO] runtime status: active
2023-01-19 22:32:28 UTC [REMOTE_COMMAND] kill process: kill -9 1434
2023-01-19 22:32:28 UTC [REMOTE_INFO] aplay killed
2023-01-19 22:32:28 UTC [REMOTE_INFO] wait dsp power status to become suspended
2023-01-19 22:32:48 UTC [REMOTE_ERROR] dsp is not suspended after 15s, end test
```

[mtrace.txt](https://github.com/thesofproject/sof/files/10463250/mtrace.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/10463252/dmesg.txt)

kv2019i, 2023-01-20:
@mwasko @makarukp Assigning to you. This started with merging  https://github.com/thesofproject/sof/pull/6656

kv2019i, 2023-01-20:
Setting P1 as this impacts CI (multiple tests failing for all pull requests).

mwasko, 2023-01-20:
@kv2019i, @miRoox is this reproducing on TGL only or does it reproduce on MTL configuration as well?

miRoox, 2023-01-20:
> is this reproducing on TGL only or does it reproduce on MTL configuration as well?

It's TGL only.

fredoh9, 2023-01-23:
working fine after https://github.com/thesofproject/sof/pull/6979 merged

kv2019i, 2023-01-24:
Let me reopen this with lower priority. #6979 avoids this in default CI configuration, but if you reenable DMA-CHAIN for HDMI, you'll see this problem again.

makarukp, 2023-02-07:
It should work after #6983 . Could you retest with reenable DMA-CHAIN ?

lgirdwood, 2023-02-07:
@miRoox / @fredoh9 can someone retest. Thanks.

miRoox, 2023-02-14:
This isssue has been fixed by https://github.com/thesofproject/sof/pull/6983, tested with https://github.com/thesofproject/sof/pull/7073

closed 2023-02-14 by miRoox