thesofproject/sof #7015: [BUG][MTL]: Headset capture resuls in I/O error with mtl-003 firmware

RDharageswari, 2023-02-01:
Headset capture results in I/O error. There is no error logs in the dmesg.
I will try to get more logs. Filing the issue for earlier traction.
Please find the scenarios below:
**scenario1:**
------------------
start capture though headset/DMIC via command line, it results in I/O error.
localhost ~ # arecord -Dhw:0,0 try.wav -c2 -r48000 -fs32_le -vv
Recording WAVE 'try.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-rt5682' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 8192
  period_size  : 2048
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2048
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8192
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
arecord: pcm_read:2161: read error: Input/output error
localhost ~ #
**scenario 2:**
----------------
1)start playback through headset/speaker
2)start capture through headset/dmic(no i/o error)
3)stop pb
4)capture results in I/O error

**scenario 3**
----------------
1) start capture through UI (no issues)

wszypelt, 2023-02-03:
@RDharageswari can you share logs with ipc payload, and remote access to RVP for debugging

udaymb, 2023-02-10:
On the boards that has issue reproducible, observation 

- If IPC4 debug message are enabled - issue is not reproducible
- If sof_debug is enabled for mtrace -   issue is not reproducible

wszypelt, 2023-02-24:
fix has been created and is now being valdiated

abonislawski, 2023-03-03:
Fixed on 003:
https://github.com/thesofproject/sof/commit/98a20ce546ca09b36ef8639c058196e938b04a7a
https://github.com/thesofproject/zephyr/commit/a9975d6335431698257080027b4f7a1330ffed91

@RDharageswari please close if confirmed

wszypelt, 2023-03-10:
@RDharageswari can you confirm?

closed 2023-03-24 by lrudyX