thesofproject/sof #7026: [BUG][MTL][Chrome] MTL DSP init fail

udaymb, 2023-02-01:
**Describe the bug**
MTL DSP init fail is observed for every boot

**To Reproduce**
warm reboot/cold reboot

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.
Always reproducible

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA} - e823921711f544049ef1c82e569c4782  /lib/firmware/intel/sof-ipc4/mtl/community/sof-mtl.ri (mtl-003-hot-fix-1
2) Name of the topology file
    * Topology: ba43df44534345b6a54a9d49d83d25f5  /lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Chrome
A3-282 silicon on Chrome board - May be this is very specific to this silicon/board
Coreboot
Cselite RelFsp: [RelFsp_Image](https://ubit-artifactory-ba.intel.com/artifactory/chromefirmwarereleases-ba-local/MTL-P/REX/Rex-Proto-0/WW03_23/Static/RelFsp/image-rex0_ww03_static.serial.bin)
Cselite RelFsp NonSerial: [RelFsp_NonSerial_Image](https://ubit-artifactory-ba.intel.com/artifactory/chromefirmwarereleases-ba-local/MTL-P/REX/Rex-Proto-0/WW03_23/Static/NonSerial/image-rex0_ww03_static_nonserial.bin)

EC
[15308_CPFE_EC](https://ubit-artifactory-ba.intel.com/artifactory/chromefirmwarereleases-ba-local/MTL-P/REX/Rex-Proto-0/WW03_23/EC/15308_EC.bin)

OS
[on_demand_build_rex_R112-15330.0.0-d2023_01_27_124732-a1.bin.bz2](https://ubit-artifactory-or.intel.com/artifactory/chrome_os_prod-or-local/on_demand_build/13504/rex/3889/on_demand_build_rex_R112-15330.0.0-d2023_01_27_124732-a1.bin.bz2)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

[   46.316494] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   46.325976] sof-audio-pci-intel-mtl 0000:00:1f.3: MTL DSP init fail
[   46.332999] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[   46.341956] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[   46.351678] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x80000011, ROM debug error: 0xeeeb000d
[   46.362567] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[   46.370539] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   46.542893] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.0.0.1
[   46.573869] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[   46.583118] sof_rt5682 mtl_mx98357_rt5682: ASoC: Parent card not yet available, widget card binding deferred
[   46.661314] input: sof-rt5682 Headset Jack as /devices/pci0000:00/0000:00:1f.3/mtl_mx98357_rt5682/sound/card0/input12
[   46.673368] input: sof-rt5682 HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/mtl_mx98357_rt5682/sound/card0/input13
[   46.685381] input: sof-rt5682 HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/mtl_mx98357_rt5682/sound/card0/input14
[   46.697389] input: sof-rt5682 HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/mtl_mx98357_rt5682/sound/card0/input15
[   46.709399] input: sof-rt5682 HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/mtl_mx98357_rt5682/sound/card0/input16

lgirdwood, 2023-02-02:
@udaymb any update ? @mwasko fyi

mengdonglin, 2023-02-07:
We've not seen this issue in Linux daily testing.

lgirdwood, 2023-02-07:
> We've not seen this issue in Linux daily testing.

Ack - I've also not heard anyone reporting this except @udaymb - maybe specific to one device ?

wszypelt, 2023-02-17:
@udaymb any update?

udaymb, 2023-02-24:
@wszypelt - This issue was reproducible only on one device - will check and confirm if it's still reproducible

udaymb, 2023-02-28:
This issue cannot be reproducible now with the latest firmware, this issue can be closed.

closed 2023-03-02 by lgirdwood