thesofproject/sof #7036: [BUG] FW reported error: 164 - ipc error for msg 0x13000004|GLB_SET_PIPELINE_STATE

keqiaozhang, 2023-02-03:
**Describe the bug**
CI observed this issue on MTL-NOCODEC when testing multiple-pause-resume-50. But the reproduction rate is low, less than 20%.
Inner test ID:20490

**dmesg**
```
[ 7477.531321] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 7477.532140] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x330000a4|0x1: GLB_SET_PIPELINE_STATE
[ 7477.532171] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 164 - Unknown
[ 7477.532391] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000004|0x1
[ 7477.532433] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 7477.532456] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 7477.532476] kernel:  Port0: ASoC: trigger FE cmd: 4 failed: -22
[ 7477.532504] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[ 7477.532519] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 3 state: 3
[ 7477.532543] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
```

**mtrace**
```
[00:01:10.214,045] <err> dma_dw_common: xrun detected
[00:01:10.214,055] <err> dai_comp: comp:0 0x40000 dai_report_xrun(): overrun due to no space available
[00:01:10.214,063] <err> component: comp:0 0x40000 comp_overrun(): sink->free = 768, copy_bytes = 0
[00:01:10.214,070] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 6
[00:01:10.214,086] <err> pipe: pipe:0 0x0 pipeline_copy(): ret = -61, start->comp.id = 262144, dir = 0
[00:01:10.214,098] <inf> ipc: dai_dma_release: comp:0 0x40000 dai_config(): Component is in active state. Ignore resetting
[00:01:10.214,143] <err> component: comp:0 0x40000 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:10.214,153] <err> component: comp:0 0x40000 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:10.214,161] <err> pipe: pipe:0 0x0 pipeline_task(): xrun recovery failed! pipeline is stopped.
[00:01:10.214,168] <inf> ll_schedule: zephyr_ll_task_done: task complete 0xa00f5cc0 0x20180U
[00:01:10.214,175] <inf> ll_schedule: zephyr_ll_task_done: num_tasks 2 total_num_tasks 2
[00:01:10.214,188] <err> component: comp:1 0x40002 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:10.214,195] <inf> ll_schedule: zephyr_ll_task_done: task complete 0xa00f5200 0x20180U
[00:01:10.214,201] <inf> ll_schedule: zephyr_ll_task_done: num_tasks 1 total_num_tasks 1
[00:01:10.214,213] <inf> ll_schedule: zephyr_domain_unregister: zephyr_domain_unregister domain->type 1 domain->clk 3
[00:01:10.214,225] <wrn> dai_comp: comp:3 0x40005 dma was stopped earlier
[00:01:10.214,338] <inf> pipe: pipeline_trigger: pipe:2 0x0 pipe trigger cmd 0
[00:01:10.214,355] <err> copier: comp:2 0x40004 failed to find dai comp or sink pipeline not running.
[00:01:10.215,266] <inf> ipc: ipc_cmd: rx	: 0x46000002|0x3
[00:01:10.215,830] <inf> ipc: ipc_cmd: rx	: 0x12020000|0x0
[00:01:10.215,851] <inf> dma: dma_put: dma_put(), dma = 0x400daac8, sref = 0
[00:01:10.216,778] <inf> ipc: ipc_cmd: rx	: 0x12030000|0x0
[00:01:10.216,915] <inf> dma: dma_put: dma_put(), dma = 0x400daa38, sref = 0
[00:01:10.216,925] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 1
[00:01:10.216,931] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[00:01:10.769,931] <inf> ipc: ipc_cmd: rx	: 0x13000003|0x1
[00:01:10.770,523] <inf> ipc: ipc_cmd: rx	: 0x13000004|0x1
[00:01:10.770,536] <inf> copier: copier_prepare: comp:1 0x40003 copier_config_prepare(): Component is in active state.
[00:01:10.770,551] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 7
[00:01:10.770,565] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa00f5200 0x20180U priority 0 flags 0x0
[00:01:10.770,590] <inf> ll_schedule: zephyr_domain_register: zephyr_domain_register domain->type 1 domain->clk 3 domain->ticks_per_ms 38400 period 1000
[00:01:10.770,608] <inf> copier: copier_prepare: comp:0 0x40001 copier_config_prepare(): Component is in active state.
[00:01:10.770,625] <inf> dai_comp: dai_config_prepare: comp:0 0x40000 dai_config_prepare(): dma channel index 0 already configured
[00:01:10.770,633] <err> dai_comp: comp:0 0x40000 dai_prepare(): Missing dd->config.elem_array.elems.
[00:01:10.770,641] <err> pipe: pipe:0 0x0 pipeline_prepare(): ret = -22, dev->comp.id = 262145
[00:01:10.770,675] <err> component: comp:1 0x40003 comp_set_state(): wrong state = 5, COMP_TRIGGER_PRE_START
[00:01:10.770,686] <err> pipe: pipe:1 0x0 pipeline_trigger_run(): ret = -22, host->comp.id = 262147, cmd = 7
[00:01:10.770,695] <err> pipe: pipe:1 0x0 pipeline_task_cmd(): failed to trigger components: -22
[00:01:10.770,701] <inf> ll_schedule: zephyr_ll_task_done: task complete 0xa00f5200 0x20180U
[00:01:10.770,708] <inf> ll_schedule: zephyr_ll_task_done: num_tasks 1 total_num_tasks 1
[00:01:10.770,718] <inf> ll_schedule: zephyr_domain_unregister: zephyr_domain_unregister domain->type 1 domain->clk 3
[00:01:10.770,730] <err> ipc: ipc: pipe 0 comp 0 prepare failed -22
[00:01:10.770,740] <err> component: comp:0 0x40001 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:10.770,770] <err> component: comp:0 0x60000 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:10.770,805] <inf> component: comp_set_state: comp:0 0x40000 comp_set_state(), state already set to 1
[00:01:10.770,811] <err> ipc: failed to process msg 19 status 7
[00:01:10.770,818] <err> ipc: ipc4: FW_GEN_MSG failed with err 7
```

**To Reproduce**
~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
less than 20%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
   Kernel Branch: topic/sof-dev / 5ef24fe89beb
SOF Branch: main / 7f80908f28b3
2) Name of the topology file
    * Topology: sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP-RVP-NOCODEC

[mtrace.txt](https://github.com/thesofproject/sof/files/10574993/mtrace.txt)

[dmesg.txt](https://github.com/thesofproject/sof/files/10574995/dmesg.txt)

wszypelt, 2023-02-03:
@keqiaozhang can you share IPC payload?

wszypelt, 2023-02-03:
it is a duplicate issue: https://github.com/thesofproject/sof/issues/6734

closed 2023-02-03 by wszypelt