thesofproject/sof #7040: [BUG] 16-bit audio is distorted when ChainDMA is used

jsarha, 2023-02-03:
16-bit audio is distorted if chained dma is used for HDMI playback. 32-bit audio sounds Ok.

**To Reproduce**
If I revert "c3807ae4a topology2: avs-tplg: disable USE_CHAIN_DMA" from today's SOF head (and play 16-bit audio with my upx-i11 to the HDMI pcm (= hw:0,3), the sound on my TV is distorted. The sound is Ok if play 32-bit audio.

**Reproduction Rate**
1/1

**Expected behavior**
No distortion with supported sample-rates. 16-bit audio used to work before Chain DMA re-implementation.

**Impact**
We can't enable Chain DMA, and this in turn blocks HDMI multichannel audio.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {aa9bd450439db90f28c438641d159a3dd8c41297}
    * SOF: {2dd8a85fadaabb95939ae2a920c6fa6ba6743d1d} with c3807ae4a7617314b753ae7713200e3ae918e039 reverted.
2) Name of the topology file
    * Topology: {sof-hda-generic.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL} upx-i11

**Screenshots or console output**
[chain-dma-16-bit-linux-log.txt](https://github.com/thesofproject/sof/files/10580586/chain-dma-16-bit-linux-log.txt)
[chain-dma-16-bit-fw-log.txt](https://github.com/thesofproject/sof/files/10580587/chain-dma-16-bit-fw-log.txt)

jsarha, 2023-02-06:
> @jsarha Could you provide any waveform or distortion pattern ? What format are you testing exactly? (fs/channels/bitDepth) Is it standard PCM ? Accually we are testing 48000Hz, 2 chan, 16(16) in our internal validation and it passes.

Are yo able to hear sound being played in the validation? 

I try to provide you with some recording with my phone tomorrow.

makarukp, 2023-02-07:
> Are yo able to hear sound being played in the validation?

We don't have to hear it, becayse we are creating loopback on hda codec capture it check sound quality ald it looks clear.

jsarha, 2023-02-07:
Simplest way to reproduce the problem is by using speaker-test. There is a clear distortion when running this:

speaker-test -D hw:0,3 -c 2 -t sine -F S16_LE

This sounds Ok:

speaker-test -D hw:0,3 -c 2 -t sine -F S32_LE

jsarha, 2023-02-07:
> @jsarha Could you provide full IPC logs with payload for reproduction? 

The logs are attached here https://github.com/thesofproject/sof/files/10580586/chain-dma-16-bit-linux-log.txt
(The IPC messages are visible in the logs. Jump to the end to see the relevant parts)
and here are the FW side logs:
https://github.com/thesofproject/sof/files/10580587/chain-dma-16-bit-fw-log.txt

> Is it reprodicible with ACE firmware ?

No. If I change the to ACE FW, same Linux kernel, and same topology, the sound is Ok.

>Is it reproducible on MTL platform?

I do not have access  to MTL HW.

makarukp, 2023-02-08:
@jsarha 
I need complete linux logs with IPC payloads. 
For example in log you provided:
Feb 03 18:05:09 jhary kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
, where payload size ie 8, but I don't see that any data in logs.
Could you provide extended logs with payload ?

jsarha, 2023-02-08:
I added a print for the first 16 bytes of data (padded with zeros if  the payload is less than that). 
Here is the line you were specifically asking for:

Feb 08 16:33:58 jhary kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
Feb 08 16:33:58 jhary kernel: data ef 34 59 24 90 c9 ed ff 00 00 00 00 00 00 00 00

However I do not think that message is very relevant for the issue. Its some message exchange related to the the Linux boot. The actual chain dma playback start and stop happens at much later time. The full log with the payloads printed is attached bellow.

The rest is attached bellow:
[chain-dma-16-bit-linux-log-2.txt](https://github.com/thesofproject/sof/files/10686856/chain-dma-16-bit-linux-log-2.txt)

makarukp, 2023-02-09:
Reproduction encountered locally by requesting fifo size of 384 bytes for format 48k/2ch/16(16). Analysis in progress...

makarukp, 2023-02-09:
This PR will probably fix that issue. Please retest with #7068 .

lgirdwood, 2023-02-10:
> This PR will probably fix that issue. Please retest with #7068 .

@jsarha good to close as PR is merged ?

jsarha, 2023-02-10:
> > This PR will probably fix that issue. Please retest with #7068 .
> 
> @jsarha good to close as PR is merged ?

Yes, can and should be closed. The issue in question is fixed.

closed 2023-02-10 by jsarha