thesofproject/sof #7045: No sound on speakers and analog headphone jack on Asus N7601ZM laptop

ErikEngerd, 2023-02-06:
With default settings, I have no sound with default settings on this laptop. There is also no sound with the legacy HD audio driver. The digital microphone works though. 

```
options snd-intel-dspcfg dsp_driver=1
```

However, I followed these instructions from [here](https://asus-linux.org/blog/sound-2021-01-11/) to create a firmware file. With this firmware file, the legacy HD audio driver works but of course I have no dmic support. 

The firmware patch file is as follows: 
```
# cat /lib/firmware/alc294-sound-patch.fw 

[codec] 
0x10ec0294 0x104312a3 0

[pincfg]
0x19 0x03A11050
0x1a 0x03A11C30
0x21 0x03211420

[verb]
0x20 0x500 0x62
0x20 0x400 0xa007
0x20 0x500 0x10
0x20 0x400 0x8420
0x20 0x500 0x0f
0x20 0x400 0x7774
```
The firmware file is essential, if I leave out the verbs for instance it doesn't work anymore.

ErikEngerd, 2023-02-06:
The alsa info for the case where legacy audio is enabled (dsp_driver=1) and without dmic is as follows. 
[alsa-info.legacy.txt](https://github.com/thesofproject/sof/files/10668178/alsa-info.legacy.txt)

ErikEngerd, 2023-02-06:
With legacy HD audio disabled:
```
options snd-hda-intel patch=alc294-sound-patch.fw
#options snd-intel-dspcfg dsp_driver=1
#options snd-hda-intel model=laptop-dmic 
```

then the alsa-info is a follows: 

[alsa-info.sof.txt](https://github.com/thesofproject/sof/files/10668218/alsa-info.sof.txt)

ErikEngerd, 2023-02-06:
It seems like I am really close to getting a fully working sound setup. Is there a way to get a permanent fix for this in sof, or is there perhaps another workaround I can use to get audio fully working?

plbossart, 2023-02-06:
@ErikEngerd what is the problem if you enable SOF and keep the "options snd-hda-intel patch=alc294-sound-patch.fw"

In theory there's no correlation between SOF and snd-hda-intel, we use the snd-hda-intel driver but don't control its internal behavior.