thesofproject/sof #7046: [BUG] FW reported error: 12| ipc error for msg 0x12020000 on MTLP-NOCODEC

keqiaozhang, 2023-02-07:
**Describe the bug**
This is a new IPC error in CI test, it happened on MTLP-NOCODEC platform when testing simultaneous-playback-capture. The reproduction is about 20%

**To Reproduce**
~/sof-test/test-case/simultaneous-playback-capture.sh -l 25

**Reproduction Rate**
20%

**dmesg**
```
[ 5774.632456] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12020000|0x0: GLB_DELETE_PIPELINE
[ 5774.633059] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x3200000c|0x0: GLB_DELETE_PIPELINE
[ 5774.633087] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 5774.633170] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12020000|0x0
[ 5774.633245] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.8
[ 5774.633316] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules copier.module.7.2:0 -> gain.7.1:0
[ 5774.633336] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: widget copier.module.7.2 freed
[ 5774.633349] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules gain.7.1:0 -> copier.host.7.1:0
[ 5774.633363] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: widget gain.7.1 freed
[ 5774.633380] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12030000|0x0: GLB_DELETE_PIPELINE
[ 5774.634014] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x3200000c|0x0: GLB_DELETE_PIPELINE
[ 5774.634021] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 5774.634220] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12030000|0x0
[ 5774.634309] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.7
[ 5774.634379] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[ 5774.634452] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
```

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel Branch: topic/sof-dev [54494c4753c8]
    * SOF Branch: main [df5eaf07cdd8]

2) Name of the topology file
    * Topology: sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC

[dmesg.txt](https://github.com/thesofproject/sof/files/10671116/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/10671117/mtrace.txt)

kv2019i, 2023-03-06:
Not assigned and priority lowered so moving to v2.6

keqiaozhang, 2023-04-03:
No reproductions in the CI within two months. Closing this bug.

closed 2023-04-03 by keqiaozhang