thesofproject/sof #7066: SOF 2.5 Release Checklist

kv2019i, 2023-02-09:
Pre RC1 Steps:
- [] tag any of the issues and feature lists with milestone v2.X  https://github.com/thesofproject/sof/issues?q=is%3Aopen+is%3Aissue+milestone%3Av2.5
- [] Choose a SOF commit on main branch as candidate for branch out stable-v2.X
- [] Testing & sign off this commit
- [] branch out stable-v2.5 , target WW10

RC1 Steps: 
- [] Cherry pick critical bug fixings and features to stable-v2.X for RC1
- [] Perform FW security scan and fix opens
- [] Testing & sign off
- [] Tag v2.5-rc1
- [] Make rc1 release on https://github.com/thesofproject/sof/releases
- [] signing of the FW, upload FW and topology binaries to internal release folder
- [] Update binary release script
- [] Make rc1 release on https://github.com/thesofproject/sof-bin

Final release steps:
- [ ] Cherry pick critical bug fixings to stable-v2.X
- [ ] Testing & sign off
- [ ] Tag v2.X
- [ ] Make release on https://github.com/thesofproject/sof/releases

Vendor specific steps / Intel
- [ ] signing of the FW, upload FW and topology binaries to internal release folder
- [ ] Make release on https://github.com/thesofproject/sof-bin

Final steps:
- [ ] Declare the release on https://www.sofproject.org/blog/

kv2019i, 2023-03-13:
SOF2.5 stable branch created https://github.com/thesofproject/sof/tree/stable-v2.5 
Target to tag rc1 next week.

marc-hb, 2023-03-13:
> stable-v2.5 branch WW10

Lucky us, Intel WW and standard WW numbers seem to match in 2023 :-)

https://www.intel.com/content/www/us/en/supplier/resources/misc/documents/workweek-calendar.html

dbaluta, 2023-03-14:
@kv2019i let's not forget to actually add a 'tag' for this branch so that version shown in dmesg is correct. I think we forgot that for stable-v2.4

kv2019i, 2023-03-14:
@dbaluta Thanks, now fixed. I also added it to the checklist so the next person doing releases remembers to do this as well. I do hate this mechanism to set version numbers from git tags. ;)

marc-hb, 2023-03-15:
>  I do hate this mechanism to set version numbers from git tags. ;)

- #6952
> I closed this PR draft - I will prepare a new solution based on a manually changed *.h file containing the current version of FW. New solution will be prepared in the new PR.

kv2019i, 2023-03-31:
SOF 2.5 is out  https://github.com/thesofproject/sof/releases/tag/v2.5 ! All issues tagged to v2.5 are either done or rescheduled to a later release, so this checklist can be considered complete and closing the item.

closed 2023-03-31 by kv2019i