thesofproject/sof #7072: [BUG]Regression. MixInMixOut. Silent in captured files

wszypelt, 2023-02-10:
**Describe the bug**
[image]

TEST DESCRIPTION:
        Transfer data with Mix In -> maximum number of MixOuts test as host-host loopback.
        MixIn -> multiple MixOuts is used in Linux topology.

        Tested Topology:
        +----+  +--------------------+   +--------------------+
        |    |  | +------+   +-----+ |   | +------+  +------+ |       1
        |    |--->|Copier|-->|MixIn|---+-->|MixOut|->|Copier|--------->+
        |    |  | +------+   +-----+ | | | +------+  +------+ |        |
        |    |  +--------------------+ | +--------------------+        |
        |    |                         |                               |
        |    |                         | ...                           |
        |    |                         |                               |
        |    |                         | +--------------------+        |
        |    |                         | | +------+  +------+ |   N    |
        |    |                         +-->|MixOut|->|Copier|----->|   |
        |    |                           | +------+  +------+ |    |   |
        |HOST|                           +--------------------+    |   |
        |    |                                                     |   |
        |    |  N                                                  |   |
        |    |<--------------------------------------------------- |   |
        |    |                                                         |
        |    |...                                                      |
        |    |                                                         |
        |    | 1                                                       |
        |    |<-------------------------------------------------------+|
        |    |
        |    |
        +----+

        _____________________
        TEST STEPS:
        1. Load FW
        2. Generate playback file
        3. Build topology
        4. Run all pipelines concurrently
        5. Wait for playback stream to finish
        6. Stop all pipelines concurrently
        8. Verify recorded files
        _____________________
        ACCEPTANCE CRITERIA:
        ·       No errors during test
        ·       Successful output files verification

**To Reproduce**
pytest -s "tests/avs/fw_01_base_fw_modules/test_18_mixers_topologies.py::TestMixinMixout::test_01_18_mixers_max_outputs[channel_count_8-mixout_count_3]"
**Reproduction Rate**
Reproducible

**Impact**
Regression since:
[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: [SOF_FW_MASTER_01.01.00.4059](http://popychaczka-ace.igk.intel.com/products/compo/download/fw/26268)
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL RVP 1

**Screenshots or console output**

wszypelt, 2023-02-10:
@ranj063 Can you look at this bug? I think it could be your changes

mengdonglin, 2023-03-13:
@wszypelt Can you please clarify which commit introduced this regression?
There are updates on module adapter and Zephyr: 
Last Passed: https://quickbuild.igk.intel.com/build/11180522/log
First Failed: https://quickbuild.igk.intel.com/build/11184576/log
[image]

wszypelt, 2023-03-13:
@mengdonglin Okay, this will take some time(~24h), but I'll try to clarify which commit introduced this regression

wszypelt, 2023-03-14:
@abonislawski  already found the exact commit that causes this: thesofproject/sof@fba35b1

lgirdwood, 2023-03-14:
@ranj063 can you check, looks like we are not copying data for capture.

wszypelt, 2023-04-21:
@ranj063 any update?

lgirdwood, 2023-04-28:
@ranj063 @singalsu @andrula-song can someone comment here and take a look

paulstelian97, 2023-04-28:
This sounds eerily similar to the bug we fixed on IPC3 with https://github.com/thesofproject/sof/pull/7420

lgirdwood, 2023-05-04:
> This sounds eerily similar to the bug we fixed on IPC3 with #7420 + #7357 + #7264

@wszypelt does this resolve the issue for us ?

ranj063, 2023-05-04:
@wszypelt @mwasko I have tested a mixin-mixout in the capture path and I able to to capture the played audio file with SSP 1 loopback. The topology I tested is as below: As you can see On pipeline 10 there's a mixin-mixout in the path:
[image]

fredoh9, 2023-05-04:
I tried with yesterday daily build (SOF SHA1 is 1236f9d8f18a) and above Ranjani's tplg. I can see Ranjani's tplg has MixIn/MixOut in Pipeline 10, daily build doesn't have them.

I tested loopback test with SSP0 and SSP1. I have very decent loopback waveform for BOTH with daily build and with Ranjani's TPLG. Is this fixed after merged with https://github.com/thesofproject/sof/pull/7420 + https://github.com/thesofproject/sof/pull/7357 + https://github.com/thesofproject/sof/pull/7264 ?

ranj063, 2023-05-04:
@wszypelt Can you please share the SHA ID of the firmware commit you are testing? We can't seem to reproduce this with main. Or alternatively would you be able to retest with main?

wszypelt, 2023-05-05:
@ranj063 @lgirdwood @fredoh9 
Main SHA: 43b08c62b523f1d17cfb318a0c649a7ebafde913
I checked now on the latest main (SHA above) Unfortunately the error still reproduces

mszleszy, 2023-05-08:
@ranj063 @fredoh9 
Just a small clarification from my side. We see the problem only when using multiple sinks on same mixIn module (don't mistake with multi mixin modules)
like in here:
[image]

In topologies mentioned before there are no pipelines using miltiple sinks on same mixin module, so no reproduction will be possible 

We don't see any problem when using only first sink, but when we bind anything to second sink, we receive only zero samples on all mixIn sinks

lgirdwood, 2023-05-11:
@ranj063 @fredoh9 any update ?

ranj063, 2023-05-25:
@mszleszy Thanks for the update. I will look into it today

ranj063, 2023-05-25:
@mszleszy I created a topology with 2 mixouts connected to the mix in the capture path as below and capture still works OK on Linux. I cannot create a host_copier->mix->mixout->host_copier on Linux as it just wont work.

[image]

mszleszy, 2023-05-30:
@ranj063 

Unfortunately still having repro of the issue . I'm adding more details below
I took fw binary directly from 7693 PR.
When MixIn sinks end up on same MixOut module audio worked pretty well. 
But couldn't get same results on MixIn into two MixOut->hostCopier

For final confirmation I used additional i2s loop:
[image]

Still when using one sink I have proper data, after binding module to MixIn second sink, I have only zeros on all output pins
[ipc.log](https://github.com/thesofproject/sof/files/11599192/ipc.log)

mszleszy, 2023-06-06:
another update:
After https://github.com/thesofproject/sof/pull/7746  I'm able to run all possible MixIn counts but if MultiPipelineSetState has payload order as described in last comment. 
When running MixIn pipeline first and have 2 or more MixIn sink, I'm recording only zero samples.
I can change tests but I'm a bit afraid that I'll hide possible bug so need some architecture consultation

pblaszko, 2023-06-12:
> another update: After #7746 I'm able to run all possible MixIn counts but if MultiPipelineSetState has payload order as described in last comment. When running MixIn pipeline first and have 2 or [...]

Order of starting pipelines should not cause permanent problem with signal. Driver may start pipelines in various order.

lrudyX, 2023-10-03:
No longer reproduces 
[image]

closed 2023-10-03 by lrudyX