thesofproject/sof #7074: [BUG]: [Multiple Capture - No-codec mode]: Observed dragging and metallic noise from data captured from cascaded (second level) copier

jairaj-arava, 2023-02-12:
With the PR #6783 verified the cascaded mode with no-code mode and the data captured from the PCM's of copier.host.18.1 and copier.host.20.1 has dragging and mettalic noise. 
[image]

jairaj-arava, 2023-02-12:
@sathya-nujella @RanderWang @RDharageswari

bardliao, 2023-02-13:
@lyakh

mengdonglin, 2023-02-14:
@jairaj-arava Can you please share the FW version in your test? And are you using Chrome kernel?
We don't observe this issue with sof-dev and SOF main in daily testing on MTL. So hope to have a cross check with you.

lgirdwood, 2023-02-14:
@mwasko fyi.

jairaj-arava, 2023-02-15:
> @jairaj-arava Can you please share the FW version in your test? And are you using Chrome kernel? We don't observe this issue with sof-dev and SOF main in daily testing on MTL. So hope to have a cross check with you.

@mengdonglin I tried with both  origin/mtl-003-drop-stable (top commit smart_amp_test: fix data size check)
and sof/main (top commit smart_amp: eliminat pops when switch device between speaker and headphone). 

Regarding kernel, we see reboot issue with sof-dev kernel on our boards, hence @bardliao has posted changes on top of our internal kernel 6.1 https://github.com/bardliao/os.chrome.internal.kernel-bleeding-edge/commits/mtl_6_1_upstream along with two patches from https://patchwork.kernel.org/project/alsa-devel/patch/20230209142123.17193-3-peter.ujfalusi@linux.intel.com/ that supports the mutliple catpure feature.

But still we see the issue.

lgirdwood, 2023-02-15:
@lyakh fyi

RanderWang, 2023-02-22:
I can reproduce it on MTL RVP

RanderWang, 2023-02-22:
@jairaj-arava  can you try this one https://github.com/thesofproject/sof/pull/7145 ?  It fixes this bug

jairaj-arava, 2023-02-23:
> @jairaj-arava can you try this one #7145 ? It fixes this bug

@RanderWang , I tried the #7145  fix but still I see the same issue with both no-codec and codec mode.

plbossart, 2023-02-23:
@jairaj-arava is this happening only with the DMIC input? 
Wondering if the cascaded copiers on SSP capture also show the same problem (should be easy to test in nocodec mode with SSP lbm bit set).

jairaj-arava, 2023-02-23:
> @jairaj-arava is this happening only with the DMIC input? Wondering if the cascaded copiers on SSP capture also show the same problem (should be easy to test in nocodec mode with SSP lbm bit set).

@plbossart checked only with DMIC

RanderWang, 2023-02-24:
@jairaj-arava  can you help attach the wav file ?   I used audacity to check the my dumped wav file and didn't find problem. Thanks!

jairaj-arava, 2023-02-24:
> @jairaj-arava can you help to attach the wav file in github? I used audacity to check the my dumped wav file and didn't find problem. Thanks!

@RanderWang  I used Audacity too. PFA captured with the fix
[Drag_Noise_with_Fix.zip](https://github.com/thesofproject/sof/files/10819824/Drag_Noise_with_Fix.zip)

RanderWang, 2023-02-24:
thanks, did you playback with simple sine wav ?  I can't get any hint from your wav file.

jairaj-arava, 2023-02-24:
> thanks, did you playback with simple sine wav ? I can't get any hint from your wav file.

I captured the data in codec mode (speaker codec exits) with Titanium music file. If you listen to the dmic99.wav, it is clear since it is captured from the endpoint of first level copier, however dmic97.wav and dmic98.wav those files are captured from endpoints of second level copier.

lgirdwood, 2023-03-01:
@jairaj-arava is it possible to capture a sine wave as it helps to visualize the audio distortion. i.e. it can show any gaps or sample duplication in the data that is helpful to pinpoint the root cause.

singalsu, 2023-03-01:
I was able to repeat this with UPX and sof-tgl-nocodec.tplg, for PCMs after copier.module.19.1. I can record some sine wave too.

singalsu, 2023-03-01:
Here's a screenshot of distorted chirp capture part from hw:0,28. There's discontinuity every 4 ms in the waveform. The wavs are quite big, but please drop me a note and I'll share them separately if you need.

[image]

lgirdwood, 2023-03-03:
> Here's a screenshot of distorted chirp capture part from hw:0,28. There's discontinuity every 4 ms in the waveform. The wavs are quite big, but please drop me a note and I'll share them separately if you need.

ok, r/w pointers are getting mixed up here. @lyakh fyi

kv2019i, 2023-03-07:
Discussed with @mengdonglin @singalsu @lyakh -> moving to v2.6

btian1, 2023-05-04:
@jairaj-arava , could you check this issue again with latest build( include chrome version) and mtl-005 board and update here?

jairaj-arava, 2023-05-04:
> @jairaj-arava , could you check this issue again with latest build( include chrome version) and mtl-005 board and update here?

@btian1, 
Sure, I will check and update here.

lgirdwood, 2023-05-10:
@jairaj-arava any results yet ?

alex-cri, 2023-05-25:
@jairaj-arava, if you can share latest results that will be great. Moving to v2.7 release but would like to close if its no longer an issue.

jairaj-arava, 2023-05-25:
Sorry to catch up late here. 
I checked with the latest MTL-005 drop (both fw and tplg. Tip of the branch: dc2ef666872 (origin/mtl-005-drop-stable) ipc4: fix multi pipeline set state if only one secondary core used)
I used internal mtl OS image that is >chorme 6.1 kernel which has all the latest backport kernel series.
**I still see the noise in the recordings with copier.host.18.1 and copier.host.20.1 PCM's** Attaching the same
[dmic28_main_new.zip](https://github.com/thesofproject/sof/files/11568619/dmic28_main_new.zip)

btian1, 2023-05-26:
will have a check next week.

jairaj-arava, 2023-06-12:
@btian1 any update from your end?

lgirdwood, 2023-07-05:
@btian1 @jairaj-arava ping - any update here ?

jairaj-arava, 2023-07-05:
@lgirdwood waiting to hear from @btian1.

btian1, 2023-07-06:
let me have a check on MTL board and update

btian1, 2023-07-07:
1. this issue should only be reproducible on nocodec board with dmic board.
2. today, I tried with tgl nocodec and cmd shared from jairaj:
    1. play a sine tone: 
        speaker-test -c 2 -t sine -Dhw:1,0
    2. capture 10, 27, 28 simultaneously in another window.
    arecord -d 60 -c 4 -f S32_LE -r 48000 -Dhw:0,27 32bit_27.wav & arecord -d 40 -c 4 -f S32_LE -r 48000 -Dhw:0,28 32bit_28.wav & arecord -d 20 -c 4 -f S32_LE -r 48000 -Dhw:0,28 32bit_10.wav
    4. check the captured file
  
TGL board is fine with latest config, due to our current mtlp board are all configed with HDA, change to nocodec may cause CI issue.

This means, at least, there is no issue in copier for module copy part, since both of them share same code, next suspect area is tplg, will compare the tplg between Jairaj and mine later.

Next Monday， I will move one mtlp board to nocodec mode and have a check also.

jairaj-arava, 2023-07-18:
> 1. this issue should only be reproducible on nocodec with dmic board.
> 2. today, I tried with tgl nocodec and cmd shared from jairaj:
>    
>    1. play a sine tone:
>       speaker-test -c 2 -t sine -Dhw:1,0
>    2. capture 10, 27, 28 simultaneously in another window.
>       arecord -d 60 -c 4 -f S32_LE -r 48000 -Dhw:0,27 32bit_27.wav & arecord -d 40 -c 4 -f S32_LE -r 48000 -Dhw:0,28 32bit_28.wav & arecord -d 20 -c 4 -f S32_LE -r 48000 -Dhw:0,28 32bit_10.wav
>    3. check the captured file
> 
> TGL board is fine with latest config, due to our current mtlp board are all configed with HDA, change to nocodec may cause CI issue.
> 
> This means, at least, there is no issue in copier for module copy part, since both of them share same code, next suspect area is tplg, will compare the tplg between Jairaj and mine later.
> 
> Next Monday， I will move one mtlp board to nocodec mode and have a check also.

@btian1  did you get a chance check on this? 
Btw you see the issue only when you start the recording first with DMIC 10 PCM (keep it running) and immediately with 27/28 PCM.  If you test it in the reverse i.e. recording with 27/28 PCM first and later with PCM 10, you will not see the issue.

btian1, 2023-07-19:
Hi， Jairaj

I met some issues on our local mtlp nocodec machine during deploy envionment, our CI is helping check on that, as a reference, could you help check this tplg: 
[image]

I will check again after my holiday.

Thanks
Tim

jairaj-arava, 2023-07-19:
> Hi， Jairaj
> 
> I met some issues on our local mtlp nocodec machine during deploy envionment, our CI is helping check on that, as a reference, could you help check this tplg:
> 
> I will check again after my h [sof-mtl-nocodec.tplg.txt](https://github.com/thesofproject/sof/files/12091644/sof-mtl-nocodec.tplg.txt) oliday.
> 
> Thanks Tim

@btian1 
The topology you shared is not loading on my MTL board. The one I shared to you was built from sof-main.

**[   13.990600] sof-audio-pci-intel-mtl 0000:00:1f.3: error: source module-copier.21.2 not found**
[   14.005586] acpi device:06: GPIO: looking up reset-gpios
[   14.005600] acpi device:06: GPIO: _DSD returned device:06 1 0 0
[   14.005748] gpio gpiochip0: Persistence not supported for GPIO 359
[   14.005903] sof-nocodec sof-nocodec: ASoC: topology: add_route failed: -22
[   14.013659] sof-nocodec sof-nocodec: ASoC: topology: could not load header: -22
[   14.037368] sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22

btian1, 2023-07-27:
yesterday, I tried with the one(only one) mtlp nocodec board, however, this board does not have sound card, so, can't play sinetone hence no dmic can record, then consult with Rander, maybe we can try with a non-nocodec board with nocodec kernel, will try today or tomorrow.

lgirdwood, 2023-07-31:
@jairaj-arava @btian1 ok - that looks like channel swapping in the copier, there are some copier PRs in flow for speed up and simplification, can we retest after merging these PRs as the issue may be easier to resolve (with the simplifications merged).
@mwasko fyi.

jairaj-arava, 2023-07-31:
> @jairaj-arava @btian1 ok - that looks like channel swapping in the copier, there are some copier PRs in flow for speed up and simplification, can we retest after merging these PRs as the issue may be easier to resolve (with the simplifications merged). @mwasko fyi.

@lgirdwood  sure, pls help mention them here if possible to keep track of them in case if they are few.

btian1, 2023-08-01:
@jairaj-arava , sorry, we don't have a mtlp nocodec with sound card and dmic injector device, HW rework is also not easy due to there is an conflicts between mtlp DMIC and ssp, on tglu, the reproduce rate is not 100%, only sometimes on first boot.
if possible, please share your remote and then I can have a check on your device.

@lgirdwood , @mwasko I don't have rootcause for this issue yet, if you have draft PRs, could you paste link here?

lgirdwood, 2023-08-01:
> @jairaj-arava , sorry, we don't have a mtlp nocodec with sound card and dmic injector device, HW rework is also not easy due to there is an conflicts between mtlp DMIC and ssp, on tglu, the [...]

@btian1 The draft PRs I mean are your and @ranj063 copier PRs that simplify/optimize. 
From the channel swapping screenshot it does look like we have a miconfigured format and in/out mappings between our copiers as the channel swap is constant size and repeating synchronously.

jairaj-arava, 2023-08-01:
> @jairaj-arava , sorry, we don't have a mtlp nocodec with sound card and dmic injector device, HW rework is also not easy due to there is an conflicts between mtlp DMIC and ssp, on tglu, the [...]

Sure, I will make the full setup ready for you today to access remotely.

btian1, 2023-08-02:
I tried the setup shared by Jairaj, however, there is no input, only background noise, @jairaj-arava , possible to use an dmic injector to a sound card, then the dmic input will be the playback sinetone, it will be much better for judgement.

@lgirdwood , issue reported in Feb,  nothing merged at that time, should not related with the optimization patches.

jairaj-arava, 2023-08-03:
> I tried the setup shared by Jairaj, however, there is no input, only background noise, @jairaj-arava , possible to use an dmic injector to a sound card, then the dmic input will be the playback [...]

Yesterday tried with @btian1 on the call, could repro it on the setup I shared.

jairaj-arava, 2023-08-08:
Hi @bardliao  and @btian1,
Today I tried on other board and could easily repro the issue on this board too. I am using the Chrome kernel (different from SOF-DEV) and also we didn't disable any SSP1 from the coreboot (I don't see any reason to disable it). Let us sync up again on this today.

jairaj-arava, 2023-08-09:
> @btian1 did you get a chance check on this? Btw you see the issue only when you start the recording first with DMIC 10 PCM (keep it running) and immediately with 27/28 PCM. If you test it in the [...]

@bardliao
As discussed, pls start with PCM10 first and then PCM 27/28 then you will see the issue. If you start 27/28 and then PCM10, you will not see the issue.

btian1, 2023-08-11:
@jairaj-arava , issue is rootcaused and double confirmed by you and me, and it was fixed by: https://github.com/thesofproject/sof/pull/7890/commits/6bcaf899cde497db312603408877401d24f35768

However, this PR#7890 is used to convert copier module interface, merged in July 12th, #7890 depend on #7740.

There are two solutions:

1. close this issue and mark it as mtl_006 will fix this issue.
2. if you want to port it, please port #7740 and #7890 together to mtl_005 branch.

Thanks Bard for the help on this.

jairaj-arava, 2023-08-14:
Hi @btian1,
Seems porting to MTL-005 there is other patches dependency apart from the #7740. We are aiming for MTL-006 and will discuss with the fw team on this.

Since, the PR #7890  fixes the issue, let's close this and will confirm it once we have the MTL-006 release.

closed 2023-08-14 by jairaj-arava

btian1, 2023-08-15:
yes, before #7740, there are copier optimization patches, maybe > 50, so it is big task to port it from main to mtl-005 branch, we can just wait for mtl-006 release and test again, if still have issues in mtl-006, we will check again.