thesofproject/sof #7093: [BUG] topology2: cavs-nocodec: fix capture device names

plbossart, 2023-02-14:
The device names are inconsistent

````
**** List of CAPTURE Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 2: Port2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 10: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 12: ssp-capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 27: DMIC0 Raw (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 28: DMIC0 Raw 2 (*) []
````

No idea what 'ssp-capture' means here, not what the difference between DMIC, DMIC0 Raw and DMIC0 Raw 2 is.

This should be a bit more self-explanatory.

I don't have time to fix this myself so filing as a bug so that someone else can cover this.
@RanderWang @ranj063 @bardliao please chime in.

lgirdwood, 2023-02-14:
@jsarha fyi.

bardliao, 2023-02-15:
`ssp-capture` is the second capture pcm for SSP0. It is for SSP multi stream capture. DMIC Raws are for DMIC multi stream capture.

jsarha, 2023-02-15:
Its easy enough to set what ever names we want for these PCMs [1]. I am happy to do it, if someone can tell me what we want 😅.

[1] https://github.com/thesofproject/sof/blob/70f9faf870daeb835c8f4e07467ce75a05887bf9/tools/topology/topology2/cavs-nocodec.conf#L604
https://github.com/thesofproject/sof/blob/70f9faf870daeb835c8f4e07467ce75a05887bf9/tools/topology/topology2/cavs-nocodec.conf#L658
https://github.com/thesofproject/sof/blob/70f9faf870daeb835c8f4e07467ce75a05887bf9/tools/topology/topology2/cavs-nocodec.conf#L673

plbossart, 2023-02-16:
Suggested names:

"Port0 Raw"
"DMIC"
"DMIC Raw"
"DMIC MFX" (MFX is the Mode FX in Windows, describes an intermediate processing)

jsarha, 2023-02-28:
> Suggested names:
> 
> "Port0 Raw" "DMIC" "DMIC Raw" "DMIC MFX" (MFX is the Mode FX in Windows, describes an intermediate processing)

So "ssp-capture" -> "Port0 Raw"?
"DMIC" remains "DMIC"?
 What should I do to "DMIC Raw 2", and where should I put "DMIC MFX" ?

kv2019i, 2023-03-06:
Bumping to P2. 2.5 will branch this week, let's try to get this in.

plbossart, 2023-03-06:
What I was suggesting is 
DMIC: all processing included
DMIC MFX: some processing included
DMIC RAW: no processing included

jsarha, 2023-03-08:
> What I was suggesting is
> DMIC: all processing included 
> DMIC MFX: some processing included 
> DMIC RAW: no processing included

@bardliao can you help me with above suggestion. What are there devices that match the above criteria? Could you point them to me so I can rename them in the topology?

bardliao, 2023-03-09:
@jsarha I am afraid that is not just rename. I expect the three DMIC devices will be land on three levels, a gain module is added to each level. And we will replace the gain modules to real process modules someday.
Like
[image]
DMIC RAW: copier.host.13.1
DMIC MFX: copier.host.18.1
DMIC: copier.20.1

cc @RanderWang

kv2019i, 2023-03-10:
Seems like more work is needed, bumping to v2.6.

lgirdwood, 2023-05-11:
@jsarha @plbossart @ranj063 are we good now ? Can we close ?

jsarha, 2023-05-11:
No changes have been made. I have hard time pulling together what @plbossart wants, and what is there according to @bardliao .

I can easily change the PCM names in the topology, but I do not have the knowledge of what all these bells and whistles in the nocodec topology are for and in what way they are connected to the HW (nor do I know the HW too well). 

Somebody really needs to tell me the names put in each already existing PCM, for me to be able to fix this.

Hope this picture helps:
[image]

ranj063, 2023-05-12:
> No changes have been made. I have hard time pulling together what @plbossart wants, and what is there according to @bardliao .

@jsarha this comment tells you what to modify right? https://github.com/thesofproject/sof/issues/7093#issuecomment-1432276826

ssp-capture -> Port0 Raw
DMIC0 Raw -> DMIC Raw
DMIC0 Raw 2 -> DMIC MFX

Please correct me is I am wrong @plbossart

jsarha, 2023-05-12:
> > No changes have been made. I have hard time pulling together what @plbossart wants, and what is there according to @bardliao .
> 
> @jsarha this comment tells you what to modify right? [#7093 (comment)](https://github.com/thesofproject/sof/issues/7093#issuecomment-1432276826)
> 
> ssp-capture -> Port0 Raw DMIC0 Raw -> DMIC Raw DMIC0 Raw 2 -> DMIC MFX
> 
> Please correct me is I am wrong @plbossart

I do not follow. Why is "Gain Capture 18"-stream PCM called "DMIC0 Raw", but  "Gain Capture 20" is called "DMIC0 MFX"? At least from the picture it looks like the both are symmetrically multiplexed from the same "Gain Capture 19"?

But, sure, I can do the renaming, if the above is correct. I would just be nice to know what is the logic behind.

jsarha, 2023-05-24:
Ping @ranj063 , @bardliao , and @plbossart . If you all agree the above suggestion what we want please just say so and I do it. 

The "ssp-capture -> Port0 Raw" looks Ok to me, but I do not follow why "DMIC0 Raw" should be "Raw" as there is less "processing" in "DMIC", or why the other symmetric multiplexed path should be  "DMIC0 MFX".  @bardliao suggested earlier that the gains are supposed to "simulate" processing, so both "DMIC0 Raw" and "DMIC0 Raw 2" look "processed paths to me. Should they both be indexed "MFX" PCMs? Or one "DMIC MFX" and other  just "DMIC"?  And current "DMIC" should be "DMIC Raw"?

plbossart, 2023-05-24:
The ask was simple @jsarha 

"
No idea what 'ssp-capture' means here, not what the difference between DMIC, DMIC0 Raw and DMIC0 Raw 2 is.
"

I don't know why we veered into this discussion on gains.

eqiir.14.1 models EFX

-> gain13.1 should not exist, the copier.host.13.1 should be connected to "DMIC RAW"

gain.19.1 models MFX

Gain.18.1 and gain.20.1 model two types of SFZ

-> copier.host.18.1 should be connected to DMIC SFX1
-> copier.host.20.1 should be connected to DMIC SFX2

does this answer to your question @jsarha 

The idea is really a raw stream, then some processing and additional processing on top.

jsarha, 2023-05-25:
> No idea what 'ssp-capture' means here, not what the difference between DMIC, DMIC0 Raw and DMIC0 Raw 2 is.

Yes, I think I understand the problem quite well, as I have the same problems myself. That is also why I need to bug people to gather the implicit knowledge needed to fix the issue. 

In general I think we should have  much more comment in the topology source files, so that the intention of topology implementation would be clear also to latecomer like me.

> does this answer to your question @jsarha

I think so. Thanks!

jsarha, 2023-05-25:
@plbossart , two more  questions. In addition to rename ssp-capture and DMIC0 Raw PCMs,  to fix this issue should I:

Drop gain.13.1? 
It comes from dmic-generic.conf and the change would affect all these topologies:
cavs-nocodec-bt.conf, cavs-nocodec.conf, cavs-rt5682.conf, cavs-sdw.conf, sof-hda-generic.conf

And add yet another DMIC PCM called "DMIC MFX" to nocodec.conf connected to module-copier.19.1 ?
I think that requires yet another copier layer, and probably takes some time to get it to work, and I would need a device with DMIC to test it.