thesofproject/sof #7094: [BUG] sof-firmware prevents suspension [hda_dsp_dais_suspend]

davix3f, 2023-02-14:
**Describe the bug**
_A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?_

 I'm running Arch Linux on a HP Omen gaming laptop. Everything's been fine until I updated a couple of days ago and noticed that the laptop won't suspend anymore [_Did I try rolling back the sof-firmware package?_ Yes I did, the issue persisted]. Looking at dmesg, the cause of this seems to be sof-firmware / snd_hda_intel / sof-audio-pci-intel-cnl. The issue does not happen when I remove the package sof-firmware, but then I have no working audio.

DMESG output right after boot:

```
[   16.979469] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979473]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979474]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979518] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979519]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979519]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979552] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979553]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979554]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979583] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979584]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979585]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979623] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979624]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979624]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979728] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979730]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979730]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979764] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979765]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979765]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979801] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979802]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979802]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979832] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979832]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979833]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.979871] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.979872]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.979872]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980005] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980006]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980007]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980056] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980057]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980057]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980101] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980102]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980102]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980151] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980152]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980152]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980245] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980246]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980247]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980434] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980435]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980436]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980496] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980497]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980498]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980560] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980560]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980561]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980620] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980621]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980621]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980679] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[   16.980679]  iDisp1: ASoC: error at __soc_pcm_hw_params on iDisp1: -22
[   16.980680]  HDMI1: ASoC: error at dpcm_fe_dai_hw_params on HDMI1: -22
[   16.980800] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp2 Pin: -22
[... 173 lines cut ...]
**To Reproduce**
Suspend using `systemctl suspend` or from the graphical interface dedicated command.

**Reproduction Rate**
Always

**Expected behavior**
The laptop should suspend just fine, without being resumed immediately because of soundcard failing to suspend

**Impact**
I can't suspend my laptop in any way. It's not terrible, but it's pretty annoying.

**Environment**
Arch Linux with systemd
KDE Plasma X11
Kernel: 6.1.11-arch1-1
Platform: x86_64

**Output of**  `inxi -A`:
+ Device-1: Intel Comet Lake PCH cAVS driver: sof-audio-pci-intel-cnl
+ Device-2: NVIDIA TU104 HD Audio driver: snd_hda_intel
+ Sound API: ALSA v: k6.1.11-arch1-1 running: yes
+ Sound Server-1: PipeWire v: 0.3.65 running: yes

LMK if I missed some informations that can help to fix this

lgirdwood, 2023-02-15:
@davix3f can you provide the dmesg of the sound driver loading after boot. This will provide FW version, topology used and any details from the audio device initialization.
```
[   16.983070] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[   16.983071]  iDisp3: ASoC: error at __soc_pcm_hw_params on iDisp3: -22
[   16.983071]  HDMI3: ASoC: error at dpcm_fe_dai_hw_params on HDMI3: -22
```
This looks like the sound server is trying to open the PCM on iDisp3 which may not exist ?
@ujfalusi @plbossart fyi.

ujfalusi, 2023-02-15:
@davix3f, it might be a complicated thing with NVIDIA binary drivers, but can you try to revert to an older kernel version?

I only have one machine with NVIDIA gfx, running Artix (Arch based w/o systemd):
```
# inxi -A
Audio:
  Device-1: Intel Tiger Lake-H HD Audio driver: sof-audio-pci-intel-tgl
  Device-2: NVIDIA GA106 High Definition Audio driver: snd_hda_intel
  Sound API: ALSA v: k6.1.10-artix1-1 running: yes
  Sound Server-1: PipeWire v: 0.3.65 running: yes
```
```
# pacman -Q sof-firmware 
sof-firmware 2.2.4-1
```
I'm yet to get the 6.1.11 kernel update...

ujfalusi, 2023-02-15:
Fetched, compiled and installed stable-6.1.11, everything looks normal on my device (I'm also using KDE Plasma X11)
 
@davix3f, you were saying that this issue started after the latest update, right? Can you pin-point what else got updated in between? pipewire, wireplumber, alsa-lib, alsa-utils, alsa-ucm-conf, alsa-card-profiles, NVIDIA binary (or are you using nouveau?) comes to mind.

davix3f, 2023-02-15:
> @davix3f can you provide the dmesg of the sound driver loading after boot. This will provide FW version, topology used and any details from the audio device initialization.

Sure. [I will attach a dmesg.log file right after boot](https://github.com/thesofproject/sof/files/10743611/dmesgboot.log), I think it's better than of copypasting it.

> @davix3f, you were saying that this issue started after the latest update, right? Can you pin-point what else got updated in between? pipewire, wireplumber, alsa-lib, alsa-utils, alsa-ucm-conf, alsa-card-profiles, NVIDIA binary (or are you using nouveau?) comes to mind.

Pipewire, alsa-stuff, and nvidia binary have been upgraded, since january 2023, but not from one major version to the other. Wireplumber does not appear in the pacman logs in the period when I discovered this issue, which is from Jan 23rd to Feb 10th (I did not upgrade my system in between those two dates because I did not use my arch partition - I had to do a 3gb upgrade from pacman on Feb 10th).

davix3f, 2023-02-15:
> @davix3f, it might be a complicated thing with NVIDIA binary drivers, but can you try to revert to an older kernel version?

I will try to figure out, maybe I should switch to nvidia-dkms instead of nvidia binary in order to do that. I will report back

ujfalusi, 2023-02-15:
@davix3f, you are using pipewire-media-session as session manager for pipewire? I suppose you have pipewire-pulse and the others installed and no PA around, right?

davix3f, 2023-02-15:
> @davix3f, you are using pipewire-media-session as session manager for pipewire? I suppose you have pipewire-pulse and the others installed and no PA around, right?

No, I don't have pipewire-media-session. 

**pacman -Q | grep -E .\*wire**
```
kpipewire 5.26.5-1
lib32-pipewire 1:0.3.65-2
pipewire 1:0.3.65-6
pipewire-alsa 1:0.3.65-6
pipewire-audio 1:0.3.65-6
pipewire-pulse 1:0.3.65-6
wireplumber 0.4.13-1
libwireplumber 0.4.13-1
```

**pacman -Q | grep alsa**
```
alsa-card-profiles 1:0.3.65-6
alsa-lib 1.2.8-1
alsa-plugins 1:1.2.7.1-1
alsa-topology-conf 1.2.5.1-1
alsa-ucm-conf 1.2.8-1
alsa-utils 1.2.8-1
lib32-alsa-lib 1.2.8-1
lib32-alsa-plugins 1.2.7.1-1
pipewire-alsa 1:0.3.65-6
qemu-audio-alsa 7.2.0-3

```

**pacman -Q | grep pulse**
```
lib32-libpulse 16.1-1
libpulse 16.1-3
pipewire-pulse 1:0.3.65-6
```

plbossart, 2023-02-15:
@davix3f Looks like the same issue as https://github.com/thesofproject/linux/issues/4151 to me. PipeWire is trying to use HDMI/DP devices that are not functional and there's a refcount issue in the SOF driver, fixed with https://github.com/thesofproject/linux/pull/4155

@ujfalusi probably something to send upstream and CC: stable?

davix3f, 2023-02-15:
> @davix3f Looks like the same issue as [thesofproject/linux#4151](https://github.com/thesofproject/linux/issues/4151) to me. PipeWire is trying to use HDMI/DP devices that are not functional and there's a DMA stream_tag leak in the SOF driver, fixed with [thesofproject/linux#4155](https://github.com/thesofproject/linux/pull/4155)

Is there anything I can do on my system to patch this, or I simply wait for the next release?

plbossart, 2023-02-15:
@davix3f if you have the ability to recompile your kernel, sure you can apply the commit in thesofproject/linux#4155. it's just moving one block of code around.

ujfalusi, 2023-02-15:
> @davix3f Looks like the same issue as [thesofproject/linux#4151](https://github.com/thesofproject/linux/issues/4151) to me. PipeWire is trying to use HDMI/DP devices that are not functional and there's a DMA stream_tag leak in the SOF driver, fixed with [thesofproject/linux#4155](https://github.com/thesofproject/linux/pull/4155)
> 
> @ujfalusi probably something to send upstream and CC: stable?

That does not apply at all on upstream it is depending on the HDA DAI abstraction (and might be introduced by it?), not going to send it within 5 minutes...

plbossart, 2023-02-15:
@ujfalusi it's always been there, the issue is that PipeWire started using devices that are not functional, so that impacts all kernels since probably 5.7

ujfalusi, 2023-02-15:
> @ujfalusi it's always been there, the issue is that PipeWire started using devices that are not functional, so that impacts all kernels since probably 5.7

Right, for me PW is not doing that for some reason, but I did saw it on other device where it did.
I'll send a patch for upstream tomorrow, we will deal with it when it comes back to us.

davix3f, 2023-02-15:
> @davix3f if you have the ability to recompile your kernel, sure you can apply the commit in [thesofproject/linux#4155](https://github.com/thesofproject/linux/pull/4155). it's just moving one block of code around.

I'm using the linux package provided by arch, kernel compiling really is not for me. I will wait for an update. Thank you all for your time and assistance, keep up the great work :cool:

ujfalusi, 2023-02-16:
First baby step towards 6.1 stable: https://lore.kernel.org/alsa-devel/20230216144054.26203-1-peter.ujfalusi@linux.intel.com/
As soon this is in mainline, I will send a backport patch for stable kernels.

It will take some time, but it will show up soon.

ujfalusi, 2023-02-17:
@davix3f, meanwhile can you try to roll back pipewire to an older version? This started to happen for you after an update, there must be something which can mitigate it while the fix is on it's way.

davix3f, 2023-02-18:
> @davix3f, meanwhile can you try to roll back pipewire to an older version? This started to happen for you after an update, there must be something which can mitigate it while the fix is on it's way.

I tried to rollback pipewire, but a bunch of libraries and programs require the last version, and trying to roll those back will yield other libraries and utilities and so on. I will just bear with it until this gets patched with some update in arch packages, ssd+linux means a boot fast enought that it's not really an issue

davix3f, 2023-02-27:
Updated from linux version 6.1.14 to 6.2.1 today, and the problem seems to be fixed. I think the issue can be closed. Thanks everyone!

closed 2023-02-27 by davix3f

SKProCH, 2023-06-24:
Also got this problem but on `6.3.1-arch2-1` kernel. 
In my case seems like this is proprietary nvidia drivers problem. When my laptop trying go to sleep/hybernate its instantly wakes up. Solution for me is to activate some drivers: 
```
sudo systemctl enable nvidia-suspend
sudo systemctl enable nvidia-hibernate
sudo systemctl enable nvidia-resume
```
This solution described [here](https://mihaly4.ru/arch-linux-nvidia-suspend-hibernation) (in russian).  
Hope this helps someone.