thesofproject/sof #7098: [BUG] Parent card not yet available

RobusTetus, 2023-02-15:
**Describe the bug**
This warning just shows up in my dmesg every boot, but the audio works fine.

**To Reproduce**
Just boot the laptop.

**Reproduction Rate**
Every boot.

**Expected behavior**
Clear dmesg.

**Impact**
It is just annoyance in my dmesg. I don't know what this is. The audio works fine.

**Environment**
    * Platform: Fedora Linux 37
    * Latest version from the repos.

**Screenshots or console output**
```
[    6.532236] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
[    6.601163] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    6.601174] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    6.601445] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[    6.601455] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[    6.601462] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[    6.601469] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Codec Output Pin1 overwritten
[    6.601476] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Codec Input Pin1 overwritten
[    6.601487] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
[    6.601498] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
[    6.601507] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
[    6.601519] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
[    6.601528] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
[    6.601537] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[    6.601555] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
```

lgirdwood, 2023-02-15:
@ujfalusi @plbossart fyi

RobusTetus, 2023-02-18:
Today I noticed some actual sound artifacts from my microphone, also that may as well be a problem with the external sound card for the microphone. It was using the Analog Input in my sound settings, so I've switched to Digital. Again no idea if my mic external sound card is faulty or the kernel driver.

But to clear the confusion I may cause with this. The warning messages do occur when nothing is plugged to the laptop as well.

lgirdwood, 2023-02-21:
These messages are information only and should not be fatal. They indicate to driver engineers if something is overridden from the base topology in the case when audio does not work. i.e. no sound from speakers.

RobusTetus, 2023-02-21:
@lgirdwood Wouldn't it be better to change the loglevel of those messages to INFO, in that case? Currently it shows as WARN in dmesg. (bold text when dmesg -l warn,err is run)

lgirdwood, 2023-02-21:
> @lgirdwood Wouldn't it be better to change the loglevel of those messages to INFO, in that case? Currently it shows as WARN in dmesg. (bold text when dmesg -l warn,err is run)

It may well be, that's why I've added @plbossart and @ujfalusi to provide feedback. IIRC, this used to indicate potential issues in the past, but my understanding is that these old issues are all resolved now ?

RobusTetus, 2023-08-05:
Just checking in again. The issue is still present on latest 6.4.7-200.fc38.x86_64 fedora kernel.

lgirdwood, 2023-08-07:
@ujfalusi @plbossart is there a solution for this or is this still needed for 1st level triage ?