thesofproject/sof #7108: [BUG] [multi-core] FW panic when stream is tested on core 1 on MTL RVP

RanderWang, 2023-02-16:
When Core 1 is enabled by topology for stream 1, test the following cmd:

root@sh-mtlp-rvp-nocodec-04:~# aplay 48K_Let_It_Go.wav -Dhw:0,1 -d 3;sleep 1;aplay 48K_Let_It_Go.wav -Dhw:0,1 -d 3
Playing WAVE '48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing WAVE '48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2127: write error: Input/output error

No issue on TGL RVP

Kernel log:

[   98.517121] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[   98.517996] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x67000000|0x0: MOD_SET_DX
[   98.518220] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[   98.518229] Message payload: 00000000: 00000002 00000002
[   98.518233] snd_sof:snd_sof_dsp_core_get: sof-audio-pci-intel-mtl 0000:00:1f.3: Core 1 powered up
[   98.518237] snd_sof:snd_sof_dsp_core_get: sof-audio-pci-intel-mtl 0000:00:1f.3: Core ref 1 
[   98.518243] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x11000005|0x100000: GLB_CREATE_PIPELINE
[   99.025987] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x11000005|0x100000
[   99.026000] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[   99.026015] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x91000005|0x100000|0x0, target: 0x1b060000|0x0|0x0, ctl: 0x3
[   99.026021] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[   99.026024] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   99.026027] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[   99.026030] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[   99.026041] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[   99.026044] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[   99.026052] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[   99.026054] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   99.026059] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to create module pipeline.3
[   99.026069] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[   99.528976] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[   99.528990] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[   99.528998] Message payload: 00000000: 00000002 00000000
[   99.529004] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets

FW log:
[00:00:03.565,590] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 1
[00:00:03.565,595] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[00:00:03.566,001] <inf> ipc: ipc_cmd: rx	: 0x47000000|0x0
[00:00:04.583,588] <inf> ipc: ipc_cmd: rx	: 0x47000000|0x0
[00:00:04.584,705] <inf> ipc: ipc_cmd: rx	: 0x11000005|0x100000
[00:00:04.584,728] <err> os:  ** FATAL EXCEPTION
[00:00:04.584,735] <err> os:  ** CPU 1 EXCCAUSE 9 (load/store alignment)
[00:00:04.584,740] <err> os:  **  PC 0xa006c3ac VADDR 0x60313
[00:00:04.584,745] <err> os:  **  PS 0x50f20
[00:00:04.584,760] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:15 CALLINC:1)
[00:00:04.584,768] <err> os:  **  A0 0xa006c3d8  SP 0xa00e7820  A2 0x60313  A3 0x60323
[00:00:04.584,775] <err> os:  **  A4 0x6006c4fe  A5 0xa00ec540  A6 0xa00e781c  A7 0x400f14fc
[00:00:04.584,781] <err> os:  **  A8 0x6006c510  A9 0xa00ec520 A10 0xa00e781c A11 0x10
[00:00:04.584,788] <err> os:  ** A12 (nil) A13 0x400de6a0 A14 0xfffffffe A15 0x1
[00:00:04.584,793] <err> os:  ** LBEG 0xa0037126 LEND 0xa0037137 LCOUNT 0xa00683e8
[00:00:04.584,798] <err> os:  ** SAR 0x1

Backtrace:0xa006c3a9:0xa00e7820 0xa006c3d5:0xa00e7850 0xa006790c:0xa00e7880 0xa0067ced:0xa00e78a0 0xa0045594:0xa00e78d0 0xa0034e16:0xa00e7900 0xa003311b:0xa00e7930 

[00:00:04.584,853] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
[00:00:04.584,858] <err> os: Current thread: 0x400f0b88 (unknown)

RanderWang, 2023-02-16:
Found a hint in kernel driver, try it first

RanderWang, 2023-02-16:
SW control core 0 only, other core is controlled by fw, so it is not related to driver.

RanderWang, 2023-02-17:
a simple debug found that the second core was enabled but exception happen when it process idc interrupt.

mszleszy, 2023-02-17:
Confirmed in firmware-automated-testing.fat framework.
I'm passing logs and repro instruction to dev team

RanderWang, 2023-03-20:
https://github.com/thesofproject/sof/pull/7174
https://github.com/zephyrproject-rtos/zephyr/pull/55182

marcinszkudlinski, 2023-03-22:
https://github.com/thesofproject/sof/pull/7322

wszypelt, 2023-04-14:
@RanderWang the potential fix is ready does this issue still reproduces?

RanderWang, 2023-04-18:
@wszypelt  which branch ?  I pull latest main branch and still got FW panic.  test case stream 0,0 on core 0, stream 0,1 on core 1: aplay 48K_Let_It_Go.wav -d 5 -Dhw:0,0 ;aplay 48K_Let_It_Go.wav -d 5 -Dhw:0,1;aplay 48K_Let_It_Go.wav -d 5 -Dhw:0,0 ;aplay 48K_Let_It_Go.wav -d 5 -Dhw:0,1;

wszypelt, 2023-04-24:
@RanderWang Please correct me if I'm wrong, but wasn't the first problem only with the secondary core? and the above example is related to the transition from main core to secondary?

RanderWang, 2023-04-25:
> @RanderWang Please correct me if I'm wrong, but wasn't the first problem only with the secondary core? and the above example is related to the changing from main core to secondary?

yes, you are right. @wszypelt  do you suggest create another issue for core switch ?

wszypelt, 2023-04-25:
@RanderWang yes, in this case I would like to ask you to create a new issue if possible

RanderWang, 2023-04-25:
@wszypelt  I tested the following cmd on core 1 :  speaker-test -c 2 -Dhw:0,1 -d 5;speaker-test -c 2 -Dhw:0,1 -d 5;  The FW panic can happen easily. The issue still valid

wszypelt, 2023-04-28:
Reproduction also occurs in out tests:
pytest -s "tests/conv/fw_000_basic/test_01_multicore_host.py::TestMulticoreHostTopology::test_000_01_pipeline_on_secondary_core_stress[16000Hz_16in16bit_2ch-core_1]"

RanderWang, 2023-05-12:
MTL-005 is failed with core 0, core 1 switch

RanderWang, 2023-05-16:
My test script.  It is failed in a very few loops in most time
````
root@sh-mtlp-rvp-nocodec-04:~# ./sleep.sh 
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0

````

````c
#!/bin/bash

for i in $(seq 1 100)
do
	aplay ~/48K_Let_It_Go.wav -d 1;
	if [ $? != 0 ]; then
		break;
	fi

	aplay ~/48K_Let_It_Go.wav -d 1 -Dhw:0,1;
	if [ $? != 0 ]; then
		break;
	fi

	aplay ~/48K_Let_It_Go.wav -d 1 -Dhw:0,2;
	if [ $? != 0 ]; then
		break;
	fi
done
````

RanderWang, 2023-05-16:
Error always happens with SET_DX for core 1 or core 2

[  540.062143] snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.SSP.4.1, IPC size is 188
[  540.062152] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  540.563706] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[  540.563782] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  540.563842] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x47000000|0x0|0x0, target: 0x1b060000|0x0|0x0, ctl: 0x3
[  540.563941] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  540.563985] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  540.564032] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  540.564064] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  540.564116] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  540.564395] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  540.564499] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  540.564544] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  540.564619] Message payload: 00000000: 00000002 00000002
[  540.564626] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to enable target core for widget pipeline.3
[  540.564683] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
[  540.564743] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 1 dir 0
[  540.564800] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[  540.564926]  Port1: ASoC: error at __soc_pcm_hw_params on Port1: -110
[  540.564989]  Port1: ASoC: error at dpcm_fe_dai_hw_params on Port1: -110

RanderWang, 2023-05-16:
If the core2 is not involved in the test, sometimes the test can pass, sometimes it is failed after 30+ cycles
````c
31
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
32
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
33
Playing WAVE '/root/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
````

RanderWang, 2023-05-19:
The new updated mtl-005 makes thing even worse that multicore feature completely can't work now.  The original 005 can't pass stress test but now it is failed at the first try

RanderWang, 2023-05-19:
I see the latest update for set_pipeline_state, but now topology setting make sure that all the pipelines for one stream are on single core, and different streams uses different core. So the change is no useful now. Now the error happens with SET_DX

mengdonglin, 2023-05-19:
> I see the latest update for set_pipeline_state, but now topology setting make sure that all the pipelines for one stream are on single core, and different streams uses different core. So the change is no useful now. Now the error happens with SET_DX

@RanderWang Is the problem you observe on mtl-005 (https://github.com/thesofproject/sof/commit/f2fec7476d1f625c11c9383f921d5a1e6bf06c75) same as https://github.com/thesofproject/sof/issues/7649?

RanderWang, 2023-05-19:
Make great progress, with this change the 005 branch can pass the stress test (the change is based on windows FW). Can any one to check it and improve it ?
````c
diff --git a/soc/xtensa/intel_adsp/ace/multiprocessing.c b/soc/xtensa/intel_adsp/ace/multiprocessing.c
index 0d26fc6139..b966f40236 100644
--- a/soc/xtensa/intel_adsp/ace/multiprocessing.c
+++ b/soc/xtensa/intel_adsp/ace/multiprocessing.c
@@ -124,6 +124,8 @@ void soc_start_core(int cpu_num)
	/* Setting the Power Active bit to the off state before powering up the core. This step is
	 * required by the HW if we are starting core for a second time. Without this sequence, the
	 * core will not power on properly when doing transition D0->D3->D0.
	 */
	DSPCS.capctl[cpu_num].ctl &= ~DSPCS_CTL_SPA;

	/* Checking current power status of the core. */
	while (((DSPCS.capctl[cpu_num].ctl & DSPCS_CTL_CPA) == DSPCS_CTL_CPA)) {
		k_busy_wait(HW_STATE_CHECK_DELAY);
	}

+	DSPCS.bootctl[cpu_num].bctl |= DSPBR_BCTL_WAITIPCG ;

	DSPCS.capctl[cpu_num].ctl |= DSPCS_CTL_SPA;

	/* Waiting for power up */
	while (((DSPCS.capctl[cpu_num].ctl & DSPCS_CTL_CPA) != DSPCS_CTL_CPA) &&
	       (retry > 0)) {
		k_busy_wait(HW_STATE_CHECK_DELAY);
		retry--;
	}

````

RanderWang, 2023-05-19:
> @RanderWang on 005 multiple set state will now work on all configurations like primary+2x secondary or only single secondary or only several secondary cores

@abonislawski can you check my comments ? thanks!

RanderWang, 2023-05-19:
No such issue on TGL, as we know both cavs & ace share most code, only difference in hw register setting.

RanderWang, 2023-05-19:
@abonislawski  I got such warning with latest 005-branch

````
-- Generated new /home/rander/zephyr/build-mtl/zephyr/include/generated/sof_versions.h
[77/105] Performing build step for 'smex_ep'
ninja: no work to do.
[80/105] Building C object modules/sof/CMakeFiles/modules_sof.dir/home/rander/zephyr/sof/src/idc/idc.c.obj
/home/rander/zephyr/sof/src/idc/idc.c:290:8: warning: implicit declaration of function 'set_pipeline_state' is invalid in C99 [-Wimplicit-function-declaration]
        ret = set_pipeline_state(ppl_icd, cmd, &delayed);
              ^
1 warning generated.
````

abonislawski, 2023-05-19:
> @abonislawski can you check my comments ? thanks!

On 005 we enabled clock gating by default (WAITIPCG), looks like it should be disabled in multicore scenario? @tmleman

abonislawski, 2023-05-22:
@RanderWang I looked at this and looks like DSPBR_BCTL_WAITIPCG  is already enabled for secondary core in soc_mp_startup(), so the difference is when it is enabled, before or after power up. Potentially this could be our issue.
I moved this registry setup out of soc_mp_startup() to earliest moment in soc_start_core() (like in your test change), please verify this test branch, we can push this to 005 if it will pass
https://github.com/abonislawski/sof/tree/mtl-005-drop-stable
@keqiaozhang please also verify #7649 on this branch

RanderWang, 2023-05-23:
@abonislawski  With your branch, the stress test will be failed after device reboot, but pass stress test after reloading driver.

abonislawski, 2023-05-23:
@RanderWang clock gating is disabled on main branch, you can revert this from 005 and it will work too

abonislawski, 2023-05-23:
Verified today 2 working solutions:
1. Rander proposal with DSPBR_BCTL_WAITIPCG without removing WAITIPCG soc_mp_startup()
2. Set DSPBR_BCTL_WAITIPCG for core 0 when more than one core used

For now pushed Rander proposal to my 005 branch but on main we need to develop correct clock gating management.
https://github.com/abonislawski/sof/tree/mtl-005-drop-stable

RanderWang, 2023-05-24:
@abonislawski  Thanks!  I tested for a few hours. 005branch can pass my test script.  But I found a new way to reproduce the bug: power off the mtl rvp and boot up it after a few seconds, then I can reproduce it at the first test. I will set up another bug for this behavior. We can close this bug for current state.

closed 2023-05-26 by wszypelt