thesofproject/sof #7116: [BUG][MTL][HDA]  A volume drop wave at the beginning of headset recording

keqiaozhang, 2023-02-17:
**Describe the bug**
when checking headset capture with zero data, we observed there's a volume drop wave at the beginning. Not sure if it's a power up pop noise or a real gain issue. Need help from developer to clarify.

[image]

[image]

**To Reproduce**
arecord -Dhw:0,0 -f dat -c 2 test.wav -vvv

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: main branch
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP-RVP-HDA

lgirdwood, 2023-02-20:
@singalsu @andrula-song any comments ?

singalsu, 2023-02-20:
It's a DC level settling in headset microphone. Should add the high-pass IIR to headset capture pipeline, 0 dB gain 40 Hz cut-off as we have in tplg1 in sof-hda-generic. I'm working on adding it to DMIC and I can address this too.

singalsu, 2023-02-20:
@keqiaozhang Can you please try #7128 if it helps.

keqiaozhang, 2023-02-22:
> Can you please try #7128 if it helps.

Verified on MTL-HDA, no volume drop at the beginning of recording after applying https://github.com/thesofproject/sof/pull/7128
[image]

andrula-song, 2023-02-22:
Thanks Seppo! But in fact, this should be something like the noise of hardware codec power-on, we do can use IIR to filter it out, but can we ask hardware experts to help solve him from the root cause?

singalsu, 2023-02-22:
@andrula-song The reason for the DC in the beginning is the settling of the single ended microphone bias circuit. It would be avoided with a differential microphone connection but there's not enough pins for such in four pole 3.5 mm stereo plug (gnd, mic, l, r). There's a DC level jump always when the mic BIAS is enabled or when headset button is pressed (pulls down the mic line).

If a codec has a similar digital high-pass filter that could be enabled it would fix this similarly. I don't know if such exists but at least such does not seem to be enabled by Linux drivers. I don't see such ALSA control in alsamixer of e.g. ALC888.

This also improves sound quality of headset in wind outdoors and such so finding out codec's capability to do this would be good. 

Meanwhile the fix should be merged since we have the same fix in tplg1 for all hda-generic platforms. It's easy to remove the filter from topology if codec gets the same feature.

PR #7128 (2023-03-02): Tools: Topology2: Add high-pass IIR to HDA analog capture
    This patch includes to sof-hda-generic.conf the new
    highpass-capture-be.conf as passthrough-capture-be.conf replacement
    that adds to DAI copier pipeline the IIR high-pass filter. The
    response is hard coded to 0 dB amplification and 40 Hz cut-off.
    
    The analog microphone headsets creates a strong DC input when the bias
    voltage settles. It can be even full-scale and sound very loud. The
    high-pass filter suppresses such input signal.
    
    Fixes #7116
commit 7564e5fc (2023-03-02): Tools: Topology2: Add high-pass IIR to HDA headset capture

This patch includes to sof-hda-generic.conf the new
highpass-capture-be.conf as passthrough-capture-be.conf replacement
that adds to DAI copier pipeline the IIR high-pass filter. The
response is hard coded to 0 dB amplification and 40 Hz cut-off.

The analog microphone headsets creates a strong DC input when the bias
voltage settles. It can be even full-scale and sound very loud. The
high-pass filter suppresses such input signal.

Fixes #7116
closed 2023-03-02 by lgirdwood