thesofproject/sof #7123: [BUG][ADL][IPC4] FW reported error: 3 - Not enough space in the IPC reply buffer to complete the request

keqiaozhang, 2023-02-20:
**Describe the bug**
This is a new issue after fixing https://github.com/thesofproject/sof/issues/6864 on TGL-IPC4 platform.

**demsg**
```
[ 6354.530291] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: pipeline: 8 memory pages: 6
[ 6354.530294] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: Create widget pipeline.8 instance 0 - pipe 8 - core 0
[ 6354.530297] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x11000006|0x0: GLB_CREATE_PIPELINE
[ 6354.530852] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x31000003|0x0: GLB_CREATE_PIPELINE
[ 6354.530859] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 3 - Not enough space in the IPC reply buffer to complete the request
[ 6354.530966] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x11000006|0x0
[ 6354.530974] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to create module pipeline.8
[ 6354.530979] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set up connected widgets
[ 6354.530984] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 12 dir 1
[ 6354.530987] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[ 6354.531000] kernel:  ssp-capture: ASoC: error at __soc_pcm_hw_params on ssp-capture: -22
[ 6354.531007] kernel:  ssp-capture: ASoC: error at dpcm_fe_dai_hw_params on ssp-capture: -22
[ 6354.531013] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 12 dir 1
[ 6354.531194] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 12 dir 1
```
**mtrace**
```
[   44.418096] <inf> smart_amp_test: comp:1 0x90000 smart_amp_free()
[   44.418166] <inf> dma: dma_put(), dma = 0x9e080de0, sref = 0
[   44.418176] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power index 0
[   44.418183] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[   45.070001] <inf> ipc: rx	: 0x11000006|0x0
[   45.070018] <inf> pipe: pipeline new pipe_id 0 priority 0
[   45.070025] <err> pipe: pipeline_new(): Out of Memory
[   45.070031] <err> ipc: ipc: pipeline_new() failed
[   45.070038] <err> ipc: ipc4: FW_GEN_MSG failed with err 3
[   45.076708] <inf> ipc: rx	: 0x11000006|0x0
[   45.076726] <inf> pipe: pipeline new pipe_id 0 priority 0
[   45.076733] <err> pipe: pipeline_new(): Out of Memory
[   45.076740] <err> ipc: ipc: pipeline_new() failed
[   45.076746] <err> ipc: ipc4: FW_GEN_MSG failed with err 3
```

**To Reproduce**
$ ~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
TBD

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel Branch: topic/sof-dev
Kernel Commit: 
[80d7c2e0fe7c](https://github.com/thesofproject/linux/commit/80d7c2e0fe7c)
SOF Branch: main
SOF Commit: 
[64fe2ea7d289](https://github.com/thesofproject/sof/commit/64fe2ea7d289)
Zephyr Commit: 
[zephyr-v3.2.0-3377-gd9c4ec31fc49](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.2.0-3377-gd9c4ec31fc49)
2) Name of the topology file
    * Topology: {sof-adl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {ADLP_RVP_NOCODEC_IPC4ZPH}

[dmesg.txt](https://github.com/thesofproject/sof/files/10778888/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/10778893/mtrace.txt)

lgirdwood, 2023-02-20:
@mwasko fyi - may be leaking memory somewhere ?

mwasko, 2023-02-21:
@softwarecki FYI, to check if it is correlated with pause resume issue #6734 that you are currently debugging.

keqiaozhang, 2023-04-03:
No reproductions in CI for nearly 2 months, closing this bug.

closed 2023-04-03 by keqiaozhang