thesofproject/sof #7126: [BUG] SOF build fails with Zephyr mainline

kv2019i, 2023-02-20:
**Describe the bug**
Build failure with latest Zephyr main line

https://github.com/thesofproject/sof/actions/runs/4222153768/jobs/7330313865

** logs **
```
/zep_workspace/zephyr/drivers/watchdog/wdt_dw.c:181:13: error: 'dw_wdt_isr' defined but not used [-Werror=unused-function]
  181 | static void dw_wdt_isr(const struct device *dev)
      |             ^~~~~~~~~~
```
**Impact**
Showstopper, prevents Zephyr baseline updates to SOF main

kv2019i, 2023-02-20:
Seems to relate to "drivers: watchdog: Add a new driver for Synopsys DesignWare watchdog" that was merged today. Assigning to @softwarecki

kv2019i, 2023-02-20:
FYI @juimonen

softwarecki, 2023-02-20:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/54993

kv2019i, 2023-02-20:
Thanks @softwarecki !  @juimonen @abonislawski This means we can't handle the dai_config change or update to SSP-blob until 54993 is merged and we update directly to that (the intermediate versions won't work). And when we do that update, then we need Jaska's dai_config SOF PR to go in at the same time.

lgirdwood, 2023-02-20:
@kv2019i https://github.com/zephyrproject-rtos/zephyr/pull/54993 now merged.

closed 2023-02-21 by softwarecki