thesofproject/sof #7140: [BUG] ipc tx error, dpcm_fe_dai_hw_params failed (-22)

01mf02, 2023-02-21:
Since a few days, sound has stopped working completely with spurious messages in `dmesg` appearing, namely a repetition of the following pattern:

~~~
[    9.293136] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[    9.293145] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[    9.293146] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[    9.293149]  Speakers: ASoC: __soc_pcm_hw_params() failed (-22)
[    9.293371]  Speakers: ASoC: dpcm_fe_dai_hw_params failed (-22)
[    9.294350] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[    9.294352] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[    9.294354] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[    9.294356]  Speakers: ASoC: __soc_pcm_hw_params() failed (-22)
[    9.294578]  Speakers: ASoC: dpcm_fe_dai_hw_params failed (-22)
[    9.295535] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
...
~~~

Before, the sound was working fine.
I am not aware of having installed or updated any packages since the last time the sound has worked.
The problem remains even after rebooting.

Is it possible that this is a hardware issue? I remember having hit the laptop a bit harder on the side around the time the issue appeared, but I see only a superficial scratch where the laptop was impacted. Still, it was quite near the left speaker ...

**Environment**

I am on a Ubuntu 22.04.2 system, using kernel 5.19.0-32-generic and firmware-sof-signed 2.0-1ubuntu4.1 installed from the Ubuntu default sources. Sorry I cannot provide the commit hashes of kernel and sof, I do not know how to obtain these.
I can also provide the firmware log, if I obtain instructions how to do so.

[alsa-info.txt](https://github.com/thesofproject/sof/files/10795068/alsa-info.txt)

lgirdwood, 2023-02-21:
@plbossart @ujfalusi could this be the iDisp backend failing hw_params ?

plbossart, 2023-02-27:
Can you please add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/10845127/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the dmesg log.

My money is the dual pulseaudio/pipewire setup, in the past we've had similar issues that went away when only pipewise was used.

01mf02, 2023-03-04:
I found that using an older kernel (5.15.0-60-generic), the sound works.

paulmenzel, 2023-12-02:
Has the regression been fixed. Is it working again with current Linux kernel versions?

lgirdwood, 2023-12-04:
@ujfalusi @plbossart IIRC, the above issue was solved some time ago. We can close ?

01mf02, 2023-12-04:
I just tried it with Linux 6.2.0-37 (Ubuntu), and sound does not work with it.
@lgirdwood, how was the above issue solved?

lgirdwood, 2023-12-04:
> I just tried it with Linux 6.2.0-37 (Ubuntu), and sound does not work with it. @lgirdwood, how was the above issue solved?

@01mf02 can you provide the dmesg with Linux-6.2 (what Ubuntu version ?) 

Fwiw, a good number of the SOF CI machines are running on Ubuntu so this may well be an integration issue with your Tigerlake Chromebook running non Chrome kernels.
```
Manufacturer:      Google
Product Name:      Lillipup
Product Version:   rev3
Firmware Version:  MrChromebox-4.19
System SKU:        sku524289
Board Vendor:      Google
Board Name:        Lillipup
```

Wondercool, 2024-07-22:
Hi, I am on Arch on an Acer Spin 713 Chromebook, flashed with MrChromeBox firmware and have exactly the same error:

[jeroen@spin ~]$ inxi -SMA
System:
  Host: spin Kernel: 6.9.10-arch1-1 arch: x86_64 bits: 64
  Desktop: KDE Plasma v: 6.1.3 Distro: EndeavourOS
Machine:
  Type: Desktop System: Google product: Voxel v: rev3
    serial: <superuser required>
  Mobo: Google model: Voxel v: rev3 serial: <superuser required>
    UEFI: coreboot v: MrChromebox-2405.0 date: 06/15/2024
Audio:
  Device-1: Intel Tiger Lake-LP Smart Sound Audio
    driver: sof-audio-pci-intel-tgl
  API: ALSA v: k6.9.10-arch1-1 status: kernel-api
  Server-1: PipeWire v: 1.2.1 status: active

from dmesg;
[    5.303498] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.308192] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    5.310578] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    5.328982] sof-audio-pci-intel-tgl 0000:00:1f.3: NHLT table not found
[    5.329058] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[    5.331396] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
[    5.331399] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/community/sof-tgl.ri
[    5.331401] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-tgl-max98373-rt5682.tplg
[    5.332025] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    5.332027] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    5.332031] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    5.443024] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    5.443029] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    5.470429] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[    5.470584] sof_rt5682 tgl_rt5682_def: ASoC: Parent card not yet available, widget card binding deferred
...
[   14.567665] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   14.567677] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   14.567680] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   14.568436] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   14.568442] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   14.568445] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   14.569190] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   14.569197] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   14.569201] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   14.570032] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   14.570044] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   14.570047] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
..

The errors return in dmesg when I access for instance my microphone in a Skype test.

Playback is working but my microphone isn't. I haven't tried an old kernel, I will try to boot an old Live CD.

plbossart, 2024-07-22:
it's because Chromebooks have their own setup for DMICs that doesn't rely on ACPI - which is required for all other Linux distros.

You will have to force the kernel to use multiple 

try using

````
options snd_sof_intel_hda_common dmic_num=4
````

Not that this isn't guaranteed to work, you also have to use the firmware from Chrome along with the topology from Chrome. If you mix and match and use sof-bin or packages from Linux distros, then it's almost guaranteed to fail.

Wondercool, 2024-07-22:
Thanks for the fast reply, Pierre-Louis. Sorry, where do I add:

options snd_sof_intel_hda_common dmic_num=4

plbossart, 2024-07-22:
you can add the option to any .conf file in /etc/modprobe.d/

Just name it sof_intel.conf :-)

Wondercool, 2024-07-22:
Unfortunately that didn't work. Sorry if these seem to be stupid questions but: if I reinstall ChromeOS and copy the drivers from ChromeOS, it still wouldn't work? Because the drivers in ChromeOS don't use ACPI?
No way forward so? Seems extra usb mic is the only way now?

01mf02, 2024-11-17:
In the meantime, I have installed Ubuntu 24.04, and since then, both sound input / output are working! The only thing that does not work yet is line-out: When plugging in headphones, the system acts as if nothing was plugged in, i.e. sound continues to play on the loudspeakers.

@Wondercool, so I can confirm: In my configuration, the microphone works. The used kernel has version 6.8.0.

Wondercool, 2024-11-17:
Thank you very much for telling your setup and confirming it works.
I would love to try it but I can't at the moment.

Disaster has struck twice regarding the Acer Spin 713 Chromebook for me and I have abandonned it.
At the risk of looking stupid: I tried to replace the SSD with a bigger one and some ribbon connector lost its clip and subsequently a connector.
I bought a new one and something similar happened with the battery connector in a different way :( 

I know, I should have been more careful... I will try to bring it to a repairshop as I know exactly what's wrong but I have learned one thing: if the price for the hardware looks too good, it probably is (250 USD for a laptop that has hardware that would easily cost 600 USD in Europe...)
In the mean time, I bought a Thinkbook G6 14", also with user replaceable RAM and 2 SSD slots and EVERYTHING works, including the finger print reader and the face recognition (Howdy) uinder Linux.

I will look back on it as a very expensive lesson and will only buy hardware where the internet consensus is that it will run Linux well and is good stuff.

01mf02, 2024-11-19:
@Wondercool, oh no, that sounds horrible! I hope you'll get more out of your ThinkBook then. :) (I also have a ThinkPad as secondary laptop, it works well, even if it is not that fast anymore.)

All in all, I'm quite content with the Chromebook --- I suppose that the low price results from the idea that most users are running ChromeOS and thus pay with their data.

closed 2024-11-19 by lgirdwood