thesofproject/sof #7148: [BUG][MTL] volume change doesn't take effect during playback

keqiaozhang, 2023-02-22:
**Describe the bug**
There're something wrong when updating gain.  The volume will not change with the adjustment of gain. The ramp duration should be wrong. 
This issue only happens on all MTL platforms(I2S, HDA and SDW), no such issue on TGL-IPC4 platforms.

**To Reproduce**
1. Terminal A: aplay -Dhw:0,0 -f s16_le -c 2 -r 48000 test.wav
2. Terminal B: change the gain in alsamixer.

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: main
2) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

andrula-song, 2023-02-22:
I draft a PR https://github.com/thesofproject/sof/pull/7152  to fix it

lgirdwood, 2023-02-22:
@keqiaozhang can we close ?