thesofproject/sof #7161: [BUG] Discord link is invalid

mbana, 2023-02-22:
The channel linked to in the `README.md` is invalid:

```markdown
[#sof channel on discord.com](https://discord.com/channels/720317445772017664/930855494472589362)
```

lgirdwood, 2023-02-27:
@marc-hb fyi

marc-hb, 2023-03-02:
Is the link asking you to login?

DaveLak, 2024-11-27:
FWIW it's still a broken link. When I click it on a browser where I'm already logged in I just get an empty Discord page (see Screenshot in the foldout below.) I think you need a proper invite link if you want to allow users to joing. The format is something like: `https://discord.gg/<$INVITE_CODE>`

<details><summary>Screenshot</summary>

[image]

</details>

marc-hb, 2024-11-28:
Thanks @DaveLak . Can you submit a pull request replacing this link with a link to https://github.com/zephyrproject-rtos and a short sentence? This `#sof` channel is part of the Zephyr area (another gap in the README.md), so no point duplicating the INVITE process/code/documentation here.

DaveLak, 2024-11-28:
> Thanks @DaveLak . Can you submit a pull request replacing this link with a link to https://github.com/zephyrproject-rtos and a short sentence? This `#sof` channel is part of the Zephyr area (another gap in the README.md), so no point duplicating the INVITE process/code/documentation here.

Yes, I'd be happy to. Thanks for the context!

DaveLak, 2025-01-03:
@marc-hb PR with the README changes as requested: https://github.com/thesofproject/sof/pull/9762

closed 2025-01-13 by lgirdwood