thesofproject/sof #7162: Speakers not working on Jasperlake chromebook with rt1015 amp.

WeirdTreeThing, 2023-02-22:


cujomalainey, 2023-02-22:
Thanks, reassigning this issue to follow up on the topology bug

WeirdTreeThing, 2023-02-22:
I'll try modifying the UCM to get pipewire to work without any workarounds.

WeirdTreeThing, 2023-02-22:
I was able to fix the UCM issues.

lgirdwood, 2023-02-27:
@WeirdTreeThing can we close ?

WeirdTreeThing, 2023-02-27:
Using the ChromeOS topology as a workaround works for now but the upstream topology is still broken.

cujomalainey, 2023-02-27:
I have an email chat with @stolx going, will make sure that is fixed

stolx, 2023-03-09:
jsl topologies are not in main branch anymore.
Please suggest which branch still needs to be fixed if any.

WeirdTreeThing, 2023-03-09:
I see that jsl was removed in b637889efb5e99e9f87f0ed1b081826f6ff992f0, can I ask why it was removed?

cujomalainey, 2023-03-09:
@lgirdwood @mwasko can answer why it was removed

It needs to be fixed in stable-v2.2

lgirdwood, 2023-03-10:
> @lgirdwood @mwasko can answer why it was removed
> 
> It needs to be fixed in stable-v2.2

Ack - stable v2.2 - 
@WeirdTreeThing The Intel team removed the older Intel platforms due to the effort involved in bringing all platforms onto Zephyr and IPC4. It is better to focus effort on the more recent and newer Intel platforms (with more capabilities) to maximize the audio features we could bring on the newer platforms.

WeirdTreeThing, 2023-03-11:
Well if jsl support was dropped there is nothing to fix so the issue can be closed.

cujomalainey, 2023-03-12:
@WeirdTreeThing the devices are getting long term support on the stable v2.2 branch till the device EOLs

WeirdTreeThing, 2023-03-12:
Oh sorry, I didn't realize that.

stolx, 2023-03-14:
@WeirdTreeThing @cujomalainey, got the fix here #7266 
unfortunately, I have no device to try it on.
but .conf files generated out of .m4 files look correct now.

cujomalainey, 2023-03-14:
if "Speaker" is showing up in .conf then that is good enough for me

WeirdTreeThing, 2023-03-14:
I tested on the jsl device and it works perfectly.

stolx, 2023-03-15:
thank you @WeirdTreeThing , @cujomalainey

closed 2023-03-15 by cujomalainey