thesofproject/sof #7163: [BUG] Waves MaxxAudio causing DSP panic if the pipeline id is not set to 2

coolstar, 2023-02-23:
**Describe the bug**
Waves MaxxAudio causes DSP crash if the pipeline ID is not set to 2. It works fine when set to 2, but crashes otherwise.

**To Reproduce**
Create the following topology and attempt to play audio:

Pipe id: Anything other than 2. Pipe id 2 works fine.
Component IDs don't appear to matter, and waves works fine so long as the pipe id is 2.
DAI doesn't matter, works on SSP0 and SSP1 so long as pipe id is 2.

Core: Only using core 0

Topology is configured as following:

    host --> buffer (size 0x300) --> waves --> buffer (size 0x300) --> volume --> buffer (size 0x300) --> DAI (SSP)

Waves parameters is extracted from Chrome OS

**Reproduction Rate**

All the time with the pipeline id set to anything other than 2. Pipeline ID of 2 works fine and plays audio.

Happens with all waves param binaries from the brya image (doesn't matter the boardname or whether the param is for speaker or headphone). Works with pipe 2, crashes with any other pipe number.

**Expected behavior**

Audio plays back regardless of the pipeline ID being set

**Impact**

Effectively means waves maxxaudio can only work on 1 endpoint (e.g. headphone only) and does not let it work on more than 1 at a time.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Custom non-Linux kernel driver
    * SOF: 2:0:0-eb119 (extracted from Chrome OS brya image from /lib/firmware/intel/sof/taeko)
2) Name of the topology file
    * Topology: Roughly based on pipe-waves-codec-playback.m4 (All other endpoints disabled for testing)
3) Name of the platform(s) on which the bug is observed.
    * Platform: brya

coolstar, 2023-02-23:
I was also able to reproduce the same issue with the firmware + waves parameters from a volteer image (lillipup).

[Though that one is an older ABI (seems based on 1.7) and uses a different UUID for waves maxxaudio, not the current one for module_adapter]

coolstar, 2023-02-23:
cc @cujomalainey

cujomalainey, 2023-02-23:
To be honest this doesn't surprise me given how old that branch is and how many problems we have had with the codec adapter and how much it has changed from its launch branch. Just wondering why it matters though as the topology should not be changing and that firmware it ~3 years old.

lgirdwood, 2023-02-23:
I cant think of any runtime logic where we change flow if the pipeline ID is != a specific number. Connected pipelines where we have multiple connected pipelines do check ID to determine when one pipe ends and another begins but this example is a single pipe. Really not sure why 2 is special here.

cujomalainey, 2023-03-06:
@stolx FYI

closed 2025-07-24 by github-actions[bot]