thesofproject/sof #7176: [BUG] mtl-004-drop-stable: nocodec playback can't work

RanderWang, 2023-02-27:
FW error log:

[00:00:00.163,918] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties dai_index 2
[00:00:00.163,965] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties fifo 172048
[00:00:00.164,016] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties handshake 6
[00:00:00.164,066] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:00:00.164,186] <inf> dai_comp: dai_playback_params: comp:1 0x40001 dai_playback_params() dest_dev = 6 stream_id = 0 src_width = 4 dest_width = 4
[00:00:00.164,245] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties dai_index 2
[00:00:00.164,293] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties fifo 172048
[00:00:00.164,343] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties handshake 6
[00:00:00.164,393] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:00:00.165,100] <err> dai_comp: comp:1 0x40001 dai_config_prepare(): dma_request_channel() failed
[00:00:00.165,190] <err> pipe: pipe:1 0x0 pipeline_prepare(): ret = -5, dev->comp.id = 196608
[00:00:00.165,255] <err> ipc: ipc: pipe 1 comp 0 prepare failed -5
[00:00:00.165,971] <inf> component: comp_set_state: comp:1 0x40001 comp_set_state(), state already set to 1

RanderWang, 2023-02-27:
@mengdonglin @keqiaozhang

RanderWang, 2023-02-27:
DMIC can work with 004 branch

keqiaozhang, 2023-03-01:
I tried latest mtl-004 branch(https://github.com/thesofproject/sof/commit/d7698202f086112509da770ed18be94e1a8a89e8), but the result is bad. playback failed at the first round with `ipc timed out 0x13000004|0x1 GLB_SET_PIPELINE_STATE`. 
https://sof-ci.ostc.intel.com/#/result/planresultdetail/21514

@abonislawski , please help to double confirm it on your device.

mengdonglin, 2023-03-01:
> I tried latest mtl-004 branch([d769820](https://github.com/thesofproject/sof/commit/d7698202f086112509da770ed18be94e1a8a89e8)), but the result is bad. playback failed at the first round with `ipc timed out 0x13000004|0x1 GLB_SET_PIPELINE_STATE`. https://sof-ci.ostc.intel.com/#/result/planresultdetail/21514
> 
> @abonislawski , please help to double confirm it on your device.

**On MTL with SDW:**
dmesg:
'''
[  786.353442] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 1 state: 4
[  786.353481] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  786.354952] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[  786.354977] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  786.354984] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  786.766684] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x40000
[  786.767241] kernel: snd_soc_rt711_sdca:rt711_sdca_interrupt_callback: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_interrupt_callback control_port_stat=4, sdca_cascade=1
[  786.768347] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x20000
[  786.798785] kernel: snd_soc_rt711_sdca:rt711_sdca_headset_detect: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_headset_detect, detected_mode=0x3
[  786.798794] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, jack_type=0x1
[  786.798802] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, btn_type=0x0
[  786.798809] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, scp_sdca_stat1=0x1, scp_sdca_stat2=0x0
[  786.861399] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[  786.861430] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  786.861435] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  786.861465] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x33000000|0x1|0x0, ctl: 0x3
[  786.861492] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  786.861511] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  786.861530] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  786.861544] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  786.861568] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  786.861586] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  786.861608] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  786.861624] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  786.861659] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[  786.861682] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  786.861709] kernel:  Jack out: ASoC: trigger FE cmd: 1 failed: -110
'''
mtrace:
'''
[00:00:00.243,761] <inf> init: print_version_banner: FW ABI 0x301a000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-2260-gd7698202f086 zephyr:v3.1.0-rc3-10534-g9c22efe46882 src hash 0x33132836 (ref hash 0x33132836)
[00:00:00.244,011] <inf> clock: clock_set_freq: clock 0 set freq 10000000Hz freq_idx 0
[00:00:00.244,148] <inf> clock: clock_set_freq: clock 1 set freq 10000000Hz freq_idx 0
[00:00:00.244,278] <inf> clock: clock_set_freq: clock 2 set freq 10000000Hz freq_idx 0
*** Booting Zephyr OS build v3.1.0-rc3-10534-g9c22efe46882 ***
[00:00:00.245,733] <inf> main: main: SOF on intel_adsp_ace15_mtpm
[00:00:00.245,833] <inf> main: main: SOF initialized
[00:00:00.246,258] <inf> ipc: ipc_cmd: rx	: 0x44000000|0x31400008
[00:00:00.250,393] <inf> ipc: ipc_cmd: rx	: 0x44000000|0x3060004c
[00:00:56.508,730] <inf> ipc: ipc_cmd: rx	: 0x11000005|0x0
[00:00:56.508,923] <inf> pipe: pipeline_new: pipeline new pipe_id 0 priority 0
[00:00:56.513,125] <inf> ipc: ipc_cmd: rx	: 0x40000004|0x15
[00:00:56.513,460] <inf> ipc: buffer_new: buffer new size 0x180 id 0.0 flags 0x0
[00:00:56.513,860] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
Terminated
'''

abonislawski, 2023-03-03:
@mengdonglin @keqiaozhang can we rerun and double confirm? It works fine on my linux platform and Dhara also reported echo ref is working fine on 004

keqiaozhang, 2023-03-07:
This issue is related to the debug build, if we enable the debug option `-d`  when building FW, we will run to this error on all MTL platforms.

kv2019i, 2023-03-07:
@abonislawski  I suspect this an assert. This branch has "app: zephyr: enable CONFIG_ASSERT in debug overlay" commit, but I suspect either SOF or the Zephyr branch used here, is missing some assert fix, and now you hit an assert when you build with "-d" (which enables asserts now).

abonislawski, 2023-03-07:
@keqiaozhang @kv2019i I assume debug build works fine on main branch?

keqiaozhang, 2023-03-08:
@abonislawski debug build works fine on main branch.

abonislawski, 2023-03-09:
I found root cause for crashing with -d option, normal FW (without -d) is affected too but still handle this.

@keqiaozhang I need assistance from kernel team to check why FW receives 0 CPC for gain module, all other modules are fine.

> Analyzed today the problem with -d option, we are crashing because of wrong clock selected for this load and this results in timer overruns. Without -d option DSP can still handle it and crashing only with asserts enabled.
> 
> However, the clock selection mechanism and manifest data are correct. **The true root cause is incorrect data in IPC for gain module.** For gain FW receives 0 CPC instead of 10183 from manifest and this results in wrong calculations for clock selection. All other modules receives normal CPC.

https://github.com/thesofproject/sof/issues/7228

abonislawski, 2023-03-10:
@RanderWang thanks!

So looks like manifest data is ignored anyway, for example for MIXIN in topology I can see CPC 910 but measured minimum is 2572 (added to rimage manifest here https://github.com/thesofproject/rimage/commit/0bcef6fea4d9592d7d9ee4b5847c1960a97d6052 )

Why this works like that?

RanderWang, 2023-03-10:
@abonislawski  First I wanted to talk about windows driver. The cpc value is zero in extended module config and windows driver calculates it based on CPS, then sends to FW. So Linux driver should do it like this, but now we get cpc from topology, not from extended module config. I don't think we need to set cpc value in rimage for above reason

RanderWang, 2023-03-10:
topology change: https://github.com/thesofproject/sof/pull/7242

abonislawski, 2023-03-10:
@RanderWang its not really a big difference cps/cpc, it can be calculated from one to another. At end we need to get CPC in FW, its fine if you calculate it from cps (but from manifest in current windows solution).
However please note that windows solution have limitations (for example lets think about copier types but cps/cpc is based only on format) and it is already outdated and we could think about something better where we could differentiate normal copier from i2s copier, etc. so after all topology solution is not that bad but at least needs to be extended

RanderWang, 2023-03-10:
> @RanderWang its not really a big difference cps/cpc, it can be calculated from one to another. At the end we need to get CPC in FW, its fine if you calculate it from cps (but from manifest as in current windows solution). However please note that windows solution have limitations (for example lets think about copier types but cps/cpc is based only on format) and it is already outdated and we could think about something better where we could differentiate normal copier from i2s copier, etc. so after all topology solution is not that bad but at least needs to be extended

@abonislawski  Thanks for your info. We need to figure out a general method to make cSOF compatible with both windows and Linux.

RanderWang, 2023-03-20:
fixed

closed 2023-03-20 by RanderWang