thesofproject/sof #7180: [BUG] Waveform discontunity in chain dma transfer in multiple run/pause scenario

makarukp, 2023-02-27:
**Describe the bug**
Internal fw tests detected discontinuity during capturing signal from run/pause scenario. 
However, it is not clear if it is FW issue, because current methodology could be affected by codec imperfections. 
To investigate issue it is worth to improve test methodology.

Test methodology is currenlty being improved by @prolkowski by switching from hda codec loopback to soundwire asynnchronous codec loopback. 

**To Reproduce**
Run FW internal validation test: 
tests.conv.fw_006_hda.test_01_multiple_hda_chain.TestDmaChain.test_006_01_multiple_simultaneous_playback_chain_2_capture_chain_hda_codec_loopback

**Reproduction Rate**
100% after first pause

**Expected behavior**
Continuous waveform

**Impact**
Could affect [#65936](https://devops.igk.intel.com/IPG%20FW/Audio%20FW_SW/_workitems/edit/65936)

**Environment**
Main branch, MTL platform

**Screenshots or console output**
[image]

plbossart, 2023-02-27:
not sure what can be improved with SoundWire loopback, on MTL this isn't the same DMA type for host and link, so there's no concept of chain DMA at all.

makarukp, 2023-02-28:
This is potential internal test issue. I close that issue.

closed 2023-02-28 by makarukp