thesofproject/sof #7184: [ADL/TGL NOCODEC] Frequency response is different between left and right channel

fredoh9, 2023-02-28:
Found alsabat failure with ADLP_RVP_NOCODEC_IPC4ZPH.
https://sof-ci.01.org/linuxpr/PR4212/build3371/devicetest/index.html?model=ADLP_RVP_NOCODEC_IPC4ZPH&testcase=check-alsabat-headset-playback

- Kernel PR: 4212
- SOF Branch: main
- SOF Commit: a74ad932700c
-  Zephyr Commit: zephyr-v3.3.0-200-g9af2789cad17
- tplg: sof-adl-nocodec.tplg

[image]

alsabat test only complains about right channel.

```
BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 821.00 Hz
Amplitude: 31128.9; Percentage: [95]
Detected peak at 821.04 Hz of 41.56 dB
 Total 42.4 dB from 818.85 to 823.24 Hz
 PASS: Peak detected at target frequency
Detected at least 1 signal(s) in total

Channel 2 - Checking for target frequency 821.00 Hz
Amplitude: 30990.5; Percentage: [94]
Detected peak at 815.92 Hz of 20.72 dB
 Total 20.7 dB from 815.92 to 815.92 Hz
Detected peak at 821.04 Hz of 38.11 dB
 Total 39.5 dB from 817.38 to 822.51 Hz
 PASS: Peak detected at target frequency
Detected peak at 823.97 Hz of 20.52 dB
 Total 39.5 dB from 823.97 to 823.97 Hz
 PASS: Peak detected at target frequency
Detected at least 3 signal(s) in total

Return value is -1003
```

Attached: [ADLP_RVP_NOCODEC_alsabat_wav_captured.zip](https://github.com/thesofproject/linux/files/10820349/ADLP_RVP_NOCODEC_alsabat_wav_captured.zip)

- bat.wav.9LAB5b ; alsabat test output
- left.wav ; only left channel
- right.wav ; only right channel- 

Left and Right channels are separated. Did frequency analysis.
[image]
[image]

One caveat is I couldn't reproduce this manually with same device.

juimonen, 2023-02-28:
tried to reproduce with tgl, but just could not....

singalsu, 2023-02-28:
There's a glitch in the second channel at about 0.466s time:

[image]

singalsu, 2023-02-28:
One sample is from 1ms ago:

[image]

[image]

juimonen, 2023-02-28:
yeah, so there is the phase difference from smart amp between channels, but also some "gaps/peaks" in the channels. I think I actually also spotted them also in the left channel, but much more rarely than right channel. So I suspect something is fishy with buffer copy, somewhere.

ranj063, 2023-02-28:
@aiChaoSONG can you please look into this?

lgirdwood, 2023-03-02:
Unlikely DMA as glitch would happen on both channels at the same time. Is SSP loopback mode used here ? SSP reads one word in FIFO at a time.

juimonen, 2023-03-08:
I tried with mtl device and ssp1 and ssp2 are clean with the alsabat test. This has to be something to do with the complex pipeline with smart amp feedback.

commit 72d4854e (2023-04-04): smart_amp_test: process same frames from all inputs

The smart amp test module will first process feedback if
presents, and then process main input to its output.

However, minimum available frames from main input and
feedback is used in processing feedback, while available
frames from main input is used in processing main input.
The different frames of audio data could be processed from
main input buffer and feedback buffer.

This patch make sure the same number of frames are processed
from main input and feedback buffers.

Fixes: #7184
closed 2023-04-04 by kv2019i