thesofproject/sof #7185: [BUG]: IPC timeout observed for topology with mux components

maruthi-machani, 2023-02-28:
with PR: https://github.com/thesofproject/sof/pull/7099 observing IPC timeout with TDM mode (Mux component) on AMD platforms.
Attaching the error snippet.

[image]

maruthi-machani, 2023-03-02:
[image]

lgirdwood, 2023-03-06:
@ujfalusi @lyakh any inputs here ?
@maruthi-machani do you have FW logs prior to timeout ? i.e. any logs from shared memory on your platform.

ujfalusi, 2023-03-06:
@maruthi-machani, a description on how this failure can be triggered would help

maruthi-machani, 2023-03-07:
Issue observed during initial topology loading. by reverting the changes in ipc-common.c, the topology load is successful.

lgirdwood, 2023-03-08:
@maruthi-machani not seeing this on Intel HW atm, are able to add some debug, it looks like an exception (I'm not sure why the kernel would say the otherwise). I can only think we are trying to deref a mux buffer that does not or no longer exists ?

kv2019i, 2023-03-13:
@maruthi-machani Do you have time to debug this for v2.5? We can't really see this on any Intel platform, so we'd need help on the debug. We are targetting for a v2.5-rc1 end of this week.

maruthi-machani, 2023-03-14:
@kv2019i we need sometime to debug this. seems this is specific to AMD platform.

maruthi-machani, 2023-03-19:
@kv2019i we are observing for pipeline_29 buffer_get_comp() is waiting to acquire the buffer (buffer_acquire()).
Is this happen due to improper termination of pipeline_29 (ACPHS_VIRTUAL Dai). Can you suggest any reference topology for proper termination of mux component having two PCM playbacks.

maruthi-machani, 2023-03-20:
@kv2019i with the echo-ref dai topology, the issue of waiting to acquire the buffer got resolved. Also IPC timeout not observed.

maruthi-machani, 2023-03-23:
with AMD topology specific changes the issue is resolved, no blocker.

lgirdwood, 2023-03-24:
> with AMD topology specific changes the issue is resolved, no blocker.

@maruthi-machani Have these updates been merged, can we close ?

maruthi-machani, 2023-03-27:
updates merged https://github.com/thesofproject/sof/pull/7313, please close the issue.

maruthi-machani, 2023-03-28:
updates merged https://github.com/thesofproject/sof/pull/7313,  closing the issue.

closed 2023-03-28 by maruthi-machani