thesofproject/sof #7191: [BUG] ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0

keqiaozhang, 2023-03-01:
**Describe the bug**
Observed this issue in CI daily test. From the kernel message, there's a IPC timed out `0x13020004`, but mtrace recorded a fatal exception and caused kernel panic on CPU0. 

**DMESG**
```
[ 3556.753931] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13020004|0x0: GLB_SET_PIPELINE_STATE
[ 3556.788076] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: open stream 0 dir 1
[ 3556.788082] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-tgl 0000:00:1f.3: period min 192 max 16384 bytes
[ 3556.788084] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-tgl 0000:00:1f.3: period count 2 max 16
[ 3556.788086] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-tgl 0000:00:1f.3: buffer max 65536 bytes
[ 3556.788185] kernel: snd_sof_intel_hda_common:hda_link_dma_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: format_val=0x11, rate=48000, ch=2, format=2
[ 3556.788191] kernel: snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: hw params stream 0 dir 1
[ 3556.788198] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 3556.788212] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 3556.788216] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x4000
[ 3556.788217] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: periods:4
[ 3556.788228] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: copier copier.HDA.4.1, type 28
[ 3556.788230] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: matched audio format index for 48000Hz, 32bit, 2 channels: 0
[ 3556.788232] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: Init input audio formats for copier.HDA.4.1
[ 3556.788234] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002) buffer size 384
[ 3556.788237] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: Output audio format for copier.HDA.4.1
[ 3556.788238] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002) buffer size 384
[ 3556.788240] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: copier copier.HDA.4.1, IPC size is 84
[ 3556.788242] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: copier copier.host.3.1, type 24
[ 3556.788245] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: matched audio format index for 48000Hz, 16bit, 2 channels: 0
[ 3556.788247] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: Init input audio formats for copier.host.3.1
[ 3556.788248] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002) buffer size 384
[ 3556.788250] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: Output audio format for copier.host.3.1
[ 3556.788251] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3: Pin index #0: 48000Hz, 16bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1002) buffer size 192
[ 3556.788253] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: copier copier.host.3.1, IPC size is 84
[ 3556.788256] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: pipeline: 4 memory pages: 2
[ 3556.788258] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: Create widget pipeline.4 instance 3 - pipe 4 - core 0
[ 3557.257000] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13020004|0x0
[ 3557.257071] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3557.257076] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3557.257135] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000f80 rirb 00
[ 3557.257208] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3557.257275] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93020004|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 3557.257350] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3557.257402] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3557.257453] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3557.257489] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 3557.257544] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 3557.257621] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 3557.257680] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3557.257755] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
```

**MTRACE**
```
[   17.176621] <inf> ipc: rx	: 0x45010002|0x1000003
[   17.176640] <inf> ipc: buffer new size 0x300 id 2.1 flags 0x0
[   17.176683] <inf> pipe: comp:2 0x20001 connect buffer 1 as sink
[   17.176695] <inf> pipe: comp:1 0x30000 connect buffer 1 as source
[   17.177976] <inf> ipc: rx	: 0x13020003|0x0
[   17.178016] <wrn> ipc: ipc_pipeline_complete(): no scheduling component specified, use comp 131073
[   17.178288] <inf> ipc: rx	: 0x13020004|0x0
ASSERTION FAIL [aligned_addr == addr] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:163
	Buffer must be 128 byte aligned
[   17.178553] <err> os:  ** FATAL EXCEPTION
[   17.178578] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[   17.178586] <err> os:  **  PC 0xbe05227a VADDR (nil)
[   17.178595] <err> os:  **  PS 0x60f20
[   17.178608] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:15 CALLINC:2)
[   17.178618] <err> os:  **  A0 0xbe014f58  SP 0xbe0a2f40  A2 0x4  A3 0xbe0a2f50
[   17.178628] <err> os:  **  A4 0xbe0a2f30  A5 0x4  A6 0x30  A7 (nil)
[   17.178638] <err> os:  **  A8 0xbe013461  A9 0xbe0a2eb0 A10 0xbe074028 A11 0xbe0a2f50
[   17.178648] <err> os:  ** A12 0xbe0a2f30 A13 0x4 A14 0x1 A15 0x1
[   17.178658] <err> os:  ** LBEG 0xbe016339 LEND 0xbe016348 LCOUNT 0xbe016bc5
[   17.178665] <err> os:  ** SAR 0x1c

Backtrace:0xbe052277:0xbe0a2f40 0xbe014f55:0xbe0a2f50 0xbe01de0c:0xbe0a2f70 0xbe01dfd8:0xbe0a2f90 0xbe02b077:0xbe0a2fb0 0xbe043bf5:0xbe0a3030 0xbe02b660:0xbe0a30c0 0xbe02b715:0xbe0a3100 0xbe02b72f:0xbe0a31d0 0xbe02b9a5:0xbe0a3200 0xbe02bbd9:0xbe0a3240 0xbe02bdf2:0xbe0a3290 0xbe02c65f:0xbe0a32c0 0xbe01fb86:0xbe0a32f0 0xbe021edc:0xbe0a3310 0xbe028ab9:0xbe0a3330 0xbe04c38c:0xbe0a3360 0xbe013a82:0xbe0a33b0 

[   17.178855] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[   17.178863] <err> os: Current thread: 0x9e0a8d80 (unknown)
[   17.181323] <err> os: Halting system
Terminated
```

**To Reproduce**
~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 25

**Reproduction Rate**
Hard to reproduce, no reproductions so far.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev [04408e12e9c4](https://github.com/thesofproject/linux/commit/04408e12e9c4)
    * SOF: main/ [b2073f1f2676](https://github.com/thesofproject/sof/commit/b2073f1f2676)
2) Name of the topology file
    * Topology: avs-tplg/sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_UP_HDA_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/10857398/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/10857399/mtrace.txt)

lgirdwood, 2023-03-02:
@abonislawski @mwasko looks like an assert in the hda logic
```
ASSERTION FAIL [aligned_addr == addr] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:163
```

keqiaozhang, 2023-03-07:
This issue happened again in CI. Test ID:https://sof-ci.ostc.intel.com/#/result/planresultdetail/21700

abonislawski, 2023-03-07:
> looks like an assert in the hda logic

interesting because we have correct alignment in dtsi: dma-buf-addr-alignment = <128>
which is used correctly in SOF in host/dai_zephyr.c and chain_dma.c:

```
ret = dma_get_attribute(cd->dma_host->z_dev, DMA_ATTR_BUFFER_ADDRESS_ALIGNMENT,
				&addr_align);
```

I found we are using wrong alignment for buffer size in chain_dma.c (the same alignment used for addr and size instead of separate alignments) but it shouldnt be a problem (at least in this case)

```
fifo_size = ALIGN_UP_INTERNAL(fifo_size, addr_align);
cd->dma_buffer = buffer_alloc(fifo_size, SOF_MEM_CAPS_DMA, addr_align);
```

lgirdwood, 2023-03-08:
@abonislawski will you send a FW fix or will this be a topology fix (or both) ?

abonislawski, 2023-03-09:
@lgirdwood Im not working on this issue, just checked if we are using alignment correctly

kv2019i, 2023-03-14:
Assigning to myself as this is affecting SOF2.5 release.

kv2019i, 2023-03-15:
Somewhat hard to hit this issue today, but it seems we have a problem somewhere in host_zephyr_params(). I can see from the oops that the offending address is not valid at all (e.g. buffer address passed to DMA driver is 0x12c0). This is not aligned with 128 bytes, but alignment is not the worst problem here. FYI @abonislawski @juimonen Continuing the debug...

kv2019i, 2023-03-15:
Very low reproduction rate slowing down debug.

kv2019i, 2023-03-22:
Tested with https://github.com/thesofproject/sof/pull/7323/ -> kernel panic still occurs

kv2019i, 2023-04-03:
With today's mainline ("362a0781f20b6709cf8c609219798f6079a57511 (origin/main) topology2: Remove dma_buffer_size attribute"), the bug no longer shows up as Zephyr panic, but instead there's a plain IPC timeout and nothing in the FW log. But the original test sequence with "~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 25" is still failing.

kv2019i, 2023-04-17:
Identical panic signature now seen in different sof-test case in daily testing: ~/sof-test/test-case/multiple-pause-resume.sh -r 50

kv2019i, 2023-04-18:
I believe this is now fixed with #7450  -- @keqiaozhang let's check after daily results.

fredoh9, 2023-04-20:
For two days, I don't see same problem. We can check another daily build.

fredoh9, 2023-04-21:
https://github.com/thesofproject/sof/pull/7450 merged, no issue have found for 3 days.

keqiaozhang, 2023-04-26:
This issue happened again on TGLU_UP_HDA_IPC4ZPH in daily test:https://sof-ci.ostc.intel.com/#/result/planresultdetail/24325.

**mtrace**
```
[   33.992626] <inf> ipc: rx	: 0x13000004|0x0
ASSERTION FAIL [aligned_addr == addr] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:163
	Buffer must be 128 byte aligned
[   33.992863] <err> os:  ** FATAL EXCEPTION
[   33.992873] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[   33.992881] <err> os:  **  PC 0xbe051447 VADDR (nil)
[   33.992888] <err> os:  **  PS 0x60820
[   33.992901] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:8 CALLINC:2)
[   33.992913] <err> os:  **  A0 0xbe014fa8  SP 0xbe0a32c0  A2 0x4  A3 0xbe0a32d0
[   33.992923] <err> os:  **  A4 0xbe0a32b0  A5 0x4  A6 0x30  A7 (nil)
[   33.992933] <err> os:  **  A8 0xbe0134b1  A9 0xbe0a3230 A10 0xbe074158 A11 0xbe0a32d0
[   33.992943] <err> os:  ** A12 0xbe0a32b0 A13 0x4 A14 0x1 A15 0x1
[   33.992951] <err> os:  ** LBEG 0xbe016519 LEND 0xbe016528 LCOUNT 0xbe016da5
[   33.992960] <err> os:  ** SAR (nil)

Backtrace:0xbe051444:0xbe0a32c0 0xbe014fa5:0xbe0a32d0 0xbe01e3d4:0xbe0a32f0 0xbe01e5a0:0xbe0a3310 0xbe02aa72:0xbe0a3330 0xbe043945:0xbe0a33b0 0xbe02b1e8:0xbe0a3440 0xbe02b29d:0xbe0a3480 0xbe02b2b7:0xbe0a3550 0xbe02b52d:0xbe0a3580 0xbe02b761:0xbe0a35c0 0xbe02b9e2:0xbe0a3610 0xbe02c247:0xbe0a3640 0xbe0201f6:0xbe0a3670 0xbe021844:0xbe0a3690 0xbe0284d1:0xbe0a36b0 0xbe04b71c:0xbe0a36e0 0xbe013ad2:0xbe0a3730 

[   33.993150] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[   33.993160] <err> os: Current thread: 0x9e0a9df0 (unknown)
[   33.995655] <err> os: Halting systemTerminated
```

**dmesg**
```
[ 3044.935795] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[ 3044.937427] kernel: snd_sof_intel_hda_common:hda_dai_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: cmd=1 dai iDisp3 Pin direction 0
[ 3044.937436] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 5 dir 0 cmd 1
[ 3044.937438] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 1 state: 4
[ 3045.440348] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000004|0x0
[ 3045.440368] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 3045.440375] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3045.440397] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000180 rirb 00
[ 3045.440404] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3045.440433] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 3045.440441] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3045.440445] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3045.440449] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3045.440453] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 3045.440465] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 3045.440516] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 3045.440520] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3045.440558] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 3045.440565] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3045.440576] kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
```

@kv2019i , please help to confirm if they are the same. 

Re-opening this bug.

kv2019i, 2023-04-26:
@keqiaozhang I think this is the original bug again. 3 weeks ago this bug got hidden by a failure with higher repro rate, I commented here 3 weeks ago "With today's mainline ("362a0781f20b6709cf8c609219798f6079a57511 (origin/main) topology2: Remove dma_buffer_size attribute"), the bug no longer shows up as Zephyr panic, but instead there's a plain IPC timeout and nothing in the FW log. But the original test sequence with "~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 25" is still failing." 

With #7450 , we are back to the original bug, so I guess we can keep this open. Ideally we'd have opened a new bug back then, but alas we didn't so let's continue with this one as the title matches the panic. And the remaining buf I believe is connected with https://github.com/thesofproject/sof/issues/7361

ranj063, 2023-05-04:
one more instance today: https://sof-ci.ostc.intel.com/#/result/planresultdetail/24820?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-5

fredoh9, 2023-05-15:
Found similar but a bit different mtrace, possible double free case?

Internal daily test result: planresultdetail/25644?model=ADLP_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-50

```
[   64.715496] <inf> dai_intel_ssp: dai_ssp_update_bits base 77400, reg 30, mask 100, value 0
[   64.715505] <inf> dai_intel_ssp: dai_ssp_stopTX stop
ASSERTION FAIL [chunk_used(h, c)] @ /srv/home/jenkins/workspace/sof_generic_build@2/zephyr/lib/os/heap.c:176
	unexpected heap state (double-free?) for memory at 0x9e0b8840
[   64.715583] <err> os:  ** FATAL EXCEPTION
[   64.715593] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[   64.715601] <err> os:  **  PC 0xbe051c67 VADDR (nil)
[   64.715610] <err> os:  **  PS 0x60320
[   64.715623] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:3 CALLINC:2)
[   64.715633] <err> os:  **  A0 0xbe014fa8  SP 0xbe0a3520  A2 0x4  A3 0xbe0a3530
[   64.715643] <err> os:  **  A4 0xbe0a3510  A5 0x4  A6 0x9e0ab100  A7 0x1
[   64.715653] <err> os:  **  A8 0xbe0134b1  A9 0xbe0a3490 A10 0xbe0719b4 A11 0xbe0a3530
[   64.715663] <err> os:  ** A12 0xbe0a3510 A13 0x4 A14 (nil) A15 0x9e0b9358
[   64.715673] <err> os:  ** LBEG 0xbe016229 LEND 0xbe01623d LCOUNT (nil)
[   64.715681] <err> os:  ** SAR 0x1b

Backtrace:0xbe051c64:0xbe0a3520 0xbe014fa5:0xbe0a3530 0xbe013e36:0xbe0a3550 0xbe028aa4:0xbe0a3570 0xbe028cda:0xbe0a3590 0xbe03ec48:0xbe0a35b0 0xbe0260e8:0xbe0a35e0 0xbe02627f:0xbe0a3660 0xbe02b772:0xbe0a36c0 0xbe02b961:0xbe0a3700 0xbe02bbe2:0xbe0a3750 0xbe02c447:0xbe0a3780 0xbe0202a6:0xbe0a37b0 0xbe0218f4:0xbe0a37d0 0xbe028731:0xbe0a37f0 0xbe04bef4:0xbe0a3820 0xbe013ad2:0xbe0a3870 

[   64.715861] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[   64.715870] <err> os: Current thread: 0x9e0a9df0 (unknown)
[   64.718321] <err> os: Halting system
Terminated
```

keqiaozhang, 2023-05-16:
The reproduce rate is increased recently,  we have observed this issue 3 times this week in daily tests. Changing the priority to P1.

fredoh9, 2023-05-17:
Heap double free exception again.

```
[    9.541380] <inf> pipe: pipe:3 0x0 pipe trigger cmd 0
[    9.541446] <inf> dai_intel_ssp: dai_ssp_update_bits base 77400, reg 30, mask 100, value 0
[    9.541455] <inf> dai_intel_ssp: dai_ssp_stopTX stop
ASSERTION FAIL [chunk_used(h, c)] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/lib/os/heap.c:176
	unexpected heap state (double-free?) for memory at 0x9e0b8840
[    9.541535] <err> os:  ** FATAL EXCEPTION
[    9.541546] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[    9.541555] <err> os:  **  PC 0xbe051c77 VADDR (nil)
```

Internal daily test result: planresultdetail/25859?model=ADLP_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-50

marc-hb, 2023-05-17:
According to @kv2019i this is another occurence: https://sof-ci.01.org/sofpr/PR7546/build7928/devicetest/index.html?model=ADLP_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-5

And maybe this one too: https://sof-ci.01.org/sofpr/PR7595/build7636/devicetest/index.html?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-5

fredoh9, 2023-05-18:
MTLP_RVP_NOCODEC: has ZEPHYR FATAL ERROR in multiple-pause-resume-50.sh
Looks same or variation of  [FW#7191](https://github.com/thesofproject/sof/issues/7191)

mtrace
```
[00:00:29.404,360] <inf> ipc: ipc_cmd: rx	: 0x13000002|0x1
[00:00:29.404,378] <inf> pipe: pipeline_trigger: pipe:4 0x0 pipe trigger cmd 0
ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c] @ ZEPHYR_BASE/lib/os/heap.c:185
	corrupted heap bounds (buffer overflow?) for memory at 0x40119340
[00:00:29.404,475] <err> os:  ** FATAL EXCEPTION
[00:00:29.404,485] <err> os:  ** CPU 0 EXTerminated
```

Internal daily test: planresultdetail/25969?model=MTLP_RVP_NOCODEC&testcase=multiple-pause-resume-50

kv2019i, 2023-05-19:
I could hit the os/heap.c:185 locally. The failing heap free is called from module_adapter_free() and specifically this free

```
»       rfree(mod->stream_params);
```

This isn't the faulty call, but should help to isolate the problem. Will continue to debug this.

kv2019i, 2023-05-19:
Found a potential culprit with heap poisoning -- it seems we freeing heap buffer while DMA is still active.

marc-hb, 2023-05-24:
Small variation?
https://sof-ci.01.org/sofpr/PR7633/build8377/devicetest/index.html?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pause-resume-5
```
[    5.916803] <inf> ipc: rx	: 0x13000002|0x1
[    5.916841] <inf> pipe: pipe:4 0x0 pipe trigger cmd 0
[    5.916908] <inf> dai_intel_ssp: dai_ssp_update_bits base 77400, reg 30, mask 100, value 0
[    5.916916] <inf> dai_intel_ssp: dai_ssp_stopTX stop
ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/lib/os/heap.c:185
	corrupted heap bounds (buffer overflow?) for memory at 0x9e0b9ec0
[    5.916996] <err> os:  ** FATAL EXCEPTION
[    5.917006] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[    5.917015] <err> os:  **  PC 0xbe051ecf VADDR (nil)
[    5.917023] <err> os:  **  PS 0x60020
[    5.917036] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[    5.917046] <err> os:  **  A0 0xbe014fa8  SP 0xbe0a35a0  A2 0x4  A3 0xbe0a35b0
[    5.917056] <err> os:  **  A4 0xbe0a3590  A5 0x4  A6 0x9e0ab100  A7 0x1
[    5.917066] <err> os:  **  A8 0xbe0134b1  A9 0xbe0a3510 A10 0xbe071a1c A11 0xbe0a35b0
[    5.917076] <err> os:  ** A12 0xbe0a3590 A13 0x4 A14 (nil) A15 0x9e0ba9d8
[    5.917086] <err> os:  ** LBEG 0xbe016229 LEND 0xbe01623d LCOUNT (nil)
[    5.917093] <err> os:  ** SAR 0x1b

Backtrace:0xbe051ecc:0xbe0a35a0 0xbe014fa5:0xbe0a35b0 0xbe013e0a:0xbe0a35d0 0xbe028cbc:0xbe0a35f0 0xbe028ef2:0xbe0a3610 0xbe03f508:0xbe0a3630 0xbe026298:0xbe0a3660 0xbe02642f:0xbe0a36e0 0xbe02b982:0xbe0a3740 0xbe02bb71:0xbe0a3780 0xbe02bdf2:0xbe0a37d0 0xbe02c67f:0xbe0a3800 0xbe0202ee:0xbe0a3830 0xbe021aa4:0xbe0a3850 0xbe028949:0xbe0a3870 0xbe04c164:0xbe0a38a0 0xbe013ad2:0xbe0a38f0 

```

Interestingly, two "panic while handling a panic" follow:
```
[    5.917275] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[    5.917285] <err> os: Current thread: 0x9e0a9df0 (unknown)
[    5.919761] <err> zephyr: Halting system
ASSERTION FAIL [reason != K_ERR_KERNEL_PANIC] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/kernel/fatal.c:148
	Attempted to recover from a kernel panic condition
```

keqiaozhang, 2023-05-25:
Observed this issue again on 3 CAVS-IPC4 platforms in today's daily test, it seems that the reproduce rate has been increased.
Test ID:26355

kv2019i, 2023-05-25:
Potential fix in https://github.com/thesofproject/sof/pull/7660 . Please report if you see ayny cases of this error with this PR merged.

kv2019i, 2023-05-29:
Based on a few daily test campaigns, it seems the higher frequency "ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c" case is fixed, but the unfortunately the original "aligned_addr == addr" still seems to be occuring. Latter is lower reproduction rate, but it still does occur, so we'll have to keep the bug open.

In future, if we have new failure mods, let's file a separate bug to track.

plbossart, 2023-05-30:
same error seen @kv2019i 

result/planresultdetail/26668?model=TGLU_UP_HDA_IPC4ZPH&testcase=multiple-pipeline-capture-50

ASSERTION FAIL [aligned_addr == addr] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:163
	Buffer must be 128 byte aligned

lgirdwood, 2023-06-01:
@kv2019i will this have impact for Linux topologies ? if not, lets resolve for v2.7

kv2019i, 2023-06-02:
@lgirdwood We do have cases that affect TGL HDA tplg on multiple-pipeline-playback-50 tests. We did have this bug open on SOF2.5 already, so this is not a regression, but let's keep the door open for a last minute fix.

I now have suspect under investigation.

kv2019i, 2023-06-02:
It seems chain-dma is somehow related to this. I've noted in past week this has only occured on systems where cha-dmain is enabled in topology.

I did a custom test run in Intel CI and without chain-dma, issue is not seen (26927). Same test with current main and chain-dma enabled, issue happens (26930).

fredoh9, 2023-06-02:
Today we have non alignment exception in TGL UP Xtreme, we have seen this before though.

```
[   46.704056] <inf> dai_intel_dmic: dmic_start(), cic 0x0000c001
[   46.704068] <err> os:  ** FATAL EXCEPTION
[   46.704078] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[   46.704086] <err> os:  **  PC 0xbe01cddc VADDR 0x4
[   46.704095] <err> os:  **  PS 0x60620
[   46.704108] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:6 CALLINC:2)
[   46.704118] <err> os:  **  A0 0xbe019793  SP 0xbe0a7110  A2 0xbe084154  A3 (nil)
[   46.704128] <err> os:  **  A4 0xf  A5 0x9e095138  A6 0x9e0aab1c  A7 (nil)
[   46.704138] <err> os:  **  A8 0x1  A9 0x1 A10 0x9e0aab00 A11 0x18c0
[   46.704148] <err> os:  ** A12 (nil) A13 (nil) A14 0xbe099310 A15 0xbe099320
[   46.704158] <err> os:  ** LBEG 0xbe01d2b1 LEND 0xbe01d2c5 LCOUNT 0xbe01d61c
[   46.704166] <err> os:  ** SAR 0x18
```

Intel internal daily test: planresultdetail/26938?model=TGLU_UP_HDA_IPC4ZPH&testcase=multiple-pipeline-capture-50

kv2019i, 2023-06-02:
@fredoh9 wrote:
> [   46.704078] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
> [   46.704086] <err> os:  **  PC 0xbe01cddc VADDR 0x4

Let's track this as a separate error (do we have a bug for this?). The PC points to Zephyr dmic interrupt handler and the value of
dmic->reg_base is invalid, leading to errror when running

```
»       val0 = dai_dmic_read(dmic, OUTSTAT0);
»       val1 = dai_dmic_read(dmic, OUTSTAT1);
```

The PIF exception occurs on the OUTSTAT0. I'd need to the matching ELF image to be sure, but I think A8=0x1 is value from dmic->reg_base (which explains the unaligned access exception).

@singalsu @juimonen @abonislawski @softwarecki any ideas? It seems we get an interrupt with foobar dmic context.

Anyways, until we have concrete data this is related to this bug, I'd track this in a separate bug and focus this #7191 on the aligned_addr assertion.

juimonen, 2023-06-05:
@kv2019i at this point I can only thing that something is writing over the dmic regbase (aka mem corruption), it is read from DT in compilation phase into dmic driver struct and I think nothing in dmic code should write there.

keqiaozhang, 2023-06-07:
In today's CI test(ID:27130), this issue happened when testing check-playback-100times on MTL-NOCODEC, the reproduce rate is almost 100%. It should be a regression. 

 I did a quick bisect, it points to https://github.com/thesofproject/sof/commit/07ed14b6fd1a95d2448cf6bab9f357e188f266b7

**mtrace**
```
[00:00:12.705,005] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:00:12.705,016] <inf> dai_intel_ssp: dai_ssp_pre_start: dai_ssp_pre_start
[00:00:12.705,051] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_power_on: intel_adsp_gpdma_power_on: dma dma@7c000 initialized
[00:00:12.705,083] <inf> dai_intel_ssp: dai_ssp_start: dai_ssp_start
[00:00:12.705,095] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 30, mask 100, value 100
ASSERTION FAIL [aligned_addr == addr] @ ZEPHYR_BASE/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:163
	Buffer musTerminated
```

@kv2019i , please help to verify it, I can open a new bug if needed.

kv2019i, 2023-06-07:
@keqiaozhang Trying to reproduce on one of the CI machines, but so far no luck, so I'm not seeing that 100% repro rate. But let's keep this in this bug as the assert is the same. https://github.com/thesofproject/sof/commit/07ed14b6fd1a95d2448cf6bab9f357e188f266b7 has a Zephyr baseline update (to 3.4.0-rc2) so it could bring in some change.

marc-hb, 2023-06-07:
Reproduction will be affected by this?
- #7726

keqiaozhang, 2023-06-08:
@kv2019i , this issue happened again in today's daily test, I also can easily reproduce it. Please make sure you use the latest FW.

kv2019i, 2023-06-08:
This didn't make the v2.6 cut, moving to v2.7.

marc-hb, 2023-06-10:
After a lot of MTL test failures across all PRs for a couple days(with empty mtraces since #7726), today's https://sof-ci.01.org/softestpr/PR1054/build419/devicetest/index.html is mostly green. Go figure.

kv2019i, 2023-06-12:
@marc-hb This is starting to look like a cache coherency issue and reproduction rate is very sensitive to any change to code. E.g. the rimage change merged on Friday seems to make this bug disappear again https://github.com/thesofproject/sof/pull/7756 .  The rimage change was not intended, so I sent a fixup PR and ta-daa, we have 7191 triggering agani with high rate -> https://github.com/thesofproject/sof/pull/7782 .

marc-hb, 2023-06-12:
Indeed  MTLP_RVP_NOCODEC is mostly red again in https://sof-ci.01.org/sofpr/PR7782/build9265/devicetest/index.html 

Other MTL configurations are green. It seems to be MTLP_RVP_NOCODEC affected every time.

kv2019i, 2023-06-14:
Status update:
 - extensive testing to check for use-after-free and out-of-bounds access to heap objects, but no indication of any unexpected access to the buffer object that gets corrupted
 - debug consistently points to sequence where we can observe correct data via uncached read, then we have some cached accesses, and then we see unexpected data via uncached read
 - above two points both hint the issue is not stray DMA writes overwriting parts of heap (DMA does not change DSP cache state) 
 - primary main suspect is buffer_alloc() and its use of incoherent.h -- the failure is always seen in this function

kv2019i, 2023-06-14:
Debugging with @lyakh , we finally have a hypothesis that explains what happens and why e.g. #7786 helps:

- at time t1: write to memory X via uncached alias
- at time t2: cache invalidate of buffer X
- at time t3: prefetch of data cache kicked off 
- at time t4: write to memory X via uncached alias
- at time t5: reads and writes to memory X via cached access
- at time t6: memory X written back to memory, cached use ends (e.g. at end of buffer_alloc())

At step t5, the writes to t4 are not seen in the prefetched data and end up being discarded.

In case of this bug, buffer->stream.addr is written two values, NULL at t1 and allocated heap address at t4. The t4 write gets lost due to prefetch and we end up asserting due to invalid buffer address.

kv2019i, 2023-06-15:
Further experiments with prefetcher configuration today, confirming the analysis and hypothesis of https://github.com/thesofproject/sof/issues/7191#issuecomment-1591608474 . 

Let's wait for some time to see how the results look, but expectation is now that this bug no longer occurs after #7786 . Based on these findings, we've surveyed the rest of the codebase and e.g. https://github.com/thesofproject/sof/pull/7808 was submitted.

keqiaozhang, 2023-06-19:
Confirmed that this issue no longer exists in CI. Closing this bug.

closed 2023-06-19 by keqiaozhang