thesofproject/sof #7193: [BUG] First IPC sent before FW gets confirmation for FW_READY message

dbaluta, 2023-03-01:
**Describe the bug**

In a stress test with suspend/resume we noticed a strange behavior for IPC interaction at FW boot. This was only noticed on i.MX8QM
which has 6 ARM cores.  We couldn't reproduce this on boards with 2 or 4 cores. 

```
                                       Linux                                                                                                 SOF
core #0                                                            core #1                                                             DSP core                    
sof_resume           
snd_sof_run_firwmare    
wait_event(boot_wait)                                                                                                                       
                                                                                                                                  platform_boot_complete
                                                            <---- DSP sends notification msg #0
                                                                   handle_irq_event
                                                                   sof_ipc3_rx_msg
                                                                   wake_up()
set_up_all_pipelines
send first IPC
                                                           ---> Linux sends notifications msg #1
                                                                                                                                            ipc_isr()
                                                           ---> Linux sends notification for reply msg #0
                                                                                                                                             SOF misses second interrupt!
                                                                                                                          
                                                           <--- DSP sends notification for reply msg #1
```

I wonder if SOF IPC can support this case. Basically there are two messages initiated (without waiting for confirmation reply) from both SOF FW and Linux.

**Reproduction Rate**

Always after stress testing with suspend / resume sometimes after 1 minute / sometimes after 1hr.

I notice that this happens more after this patch:

```
commit 74ad8ed6512186134527fc82440f62007a98ff48
Author: Peter Ujfalusi <peter.ujfalusi@linux.intel.com>
Date:   Tue Apr 5 10:27:00 2022 -0700

    ASoC: SOF: ipc3: Implement rx_msg IPC ops
    
    Add the implementation for the rx_msg callback to handle message reception
    for IPC3.
    
    The implementation is equivalent to the currently used code in ipc.c
```

Because this introduces some processing after FW ready msg is received on the Linux side.

dbaluta, 2023-03-01:
Cc: @lgirdwood @ranj063 @ujfalusi @iuliana-prodan @paulstelian97

lgirdwood, 2023-03-02:
@ujfalusi @juimonen are you guys seeing the same issue today with IPC4 ? Any clues ?

ujfalusi, 2023-03-02:
@dbaluta, that `74ad8ed65121 ("ASoC: SOF: ipc3: Implement rx_msg IPC ops")` is just re-implementing what was already implemented at that time in ipc.c under ipc3.
`ce6b49afa52a ("ASoC: SOF: ipc: Switch over to use the rx_msg ops")` would switch to use the IPC specific rx handling and if you take a look, the removed and the IPC3 code is identical.

We have not seen this with IPC3 or IPC4 on Intel platforms, but by the look of things it should be equally applicable since:
we receive the FW ready in sof/intel/ code
we call `snd_sof_ipc_msgs_rx()`
that will handle the FW_READY and then `wake_up(&sdev->boot_wait);` to proceed on the firmware booting thread
when the `sof_ipc3_rx_msg()` returns, we will ack the message to the DSP.

The only reason can be that the execution of the `snd_sof_ipc_msgs_rx()` is suspended (or migrated to another core?) while the other thread (which got woken up) are running and the order changes.

Interesting.

ujfalusi, 2023-03-02:
@dbaluta, so, yes, we also have some variation it seams:
```
[  536.712601] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[  536.712604] snd_sof:ipc3_fw_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP is ready 0x70000000 offset 0x81000
[  536.712606] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 6
[  536.712612] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[  536.712614] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 6 -> 7
[  536.712618] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  536.712621] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[  536.712625] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[  536.712627] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[  536.712634] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  536.712648] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  536.712648] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
[  536.712649] [JEEE] cnl_ipc_irq_thread: acking
[  536.712651] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x0
[  536.712652] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: periods:1
```
and
```
[  595.347984] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[  595.347987] snd_sof:ipc3_fw_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP is ready 0x70000000 offset 0x81000
[  595.347989] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 6
[  595.348033] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
[  595.348035] [JEEE] cnl_ipc_irq_thread: acking
[  595.348045] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[  595.348046] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 6 -> 7
[  595.348050] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  595.348053] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[  595.348057] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[  595.348060] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[  595.348077] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  595.348092] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  595.348095] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x0
[  595.348097] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-tgl 0000:00:1f.3: periods:1
```
this is on a TigerLake platform, suspend/resume while playing audio. The `[JEEE]` is printed in the IRQ handler before acking the rx.

We have few things done in post run, but this should be serialized out for sure.

iuliana-prodan, 2023-03-02:
For imx this is solved by https://github.com/thesofproject/sof/pull/7188.
The problem is mx_mu_xsr_rmw() which doesn't work correctly for w1c bits because it reads the register and writes it back fully.

ujfalusi, 2023-03-03:
@iuliana-prodan, I see, so the fix is on the fw side for iMX, let me check it again on our side in full how this has not affected us, apart from the small variation in execution. But I cannot recall a single case where the first IPC message would have been sent before acking the FW_READY which makes me wonder if it is just "luck".

For IPC3 we had similar robustness in firmware to not miss messages. IPC4 needed a bit different logic and safety net to achieve the same level. This might explain why we don't encounter similar issue, but from the flow pow, this could be possible, but the acking of the notification is in (soft) irq thread. It should not be put on ice for long.

Thank you for the information!

dbaluta, 2023-03-03:
@ujfalusi on our side it only happens on a board with 2 x A75 + 4 x A53. It never happens on "smaller" SoCs. So you need to have a high concurrency env.

dbaluta, 2023-03-03:
Problem fixed on FW side with https://github.com/thesofproject/sof/pull/7188

closed 2023-03-03 by dbaluta