thesofproject/sof #7204: [BUG] missing IIR EQ module config in rimage for MTL

keqiaozhang, 2023-03-03:
**Describe the bug**
IIR high-pass with gain has been added to MTL tplgs(https://github.com/thesofproject/sof/pull/7141 and https://github.com/thesofproject/sof/pull/7128), but there's no IIR module config in https://github.com/thesofproject/rimage/blob/3863e94fa5d361e434f404655dac17541dbf20c6/config/mtl.toml.

This caused FW failed to load on MTL platforms:
```
[   66.983415] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-mtl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2004) buffer size 768
[   66.983477] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Direct firmware load for intel/sof-ipc4-lib/mtl/5150C0E6-27F9-4EC8-8351-C705B642D12F.bin failed with error -2
[   66.983482] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Library file 'intel/sof-ipc4-lib/mtl/5150C0E6-27F9-4EC8-8351-C705B642D12F.bin' is missing
[   66.983538] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to find module info for widget eqiir.12.1 with UUID 5150C0E6-27F9-4EC8-8351-C705B642D12F
[   66.983600] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed updating IPC struct for eqiir.12.1
[   66.983938] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22
[   66.983983] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to load DSP topology -22
[   66.984019] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[   66.984106] kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[   66.984466] kernel: skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
```

RanderWang, 2023-03-03:
please check tgl-cavs.toml

singalsu, 2023-03-06:
> @singalsu please check tgl-cavs.toml

Would need the documentation... Could someone familiar with the format write to SOF Docs?

keqiaozhang, 2023-03-09:
Fixed by https://github.com/thesofproject/sof/pull/7220. Closing this bug.

closed 2023-03-09 by keqiaozhang