thesofproject/sof #7218: [BUG] unable to enable probes logging for MTL

kv2019i, 2023-03-06:
**Describe the bug**
Build fails to

work/sof/src/probe/probe.c:170: undefined reference to `dma_copy_set_stream_tag'

**To Reproduce**
Add to app/prj.conf

CONFIG_PROBE=y
CONFIG_LOG_BACKEND_SOF_PROBE=y
CONFIG_LOG_BACKEND_SOF_PROBE_OUTPUT_DICTIONARY=y

Rebuild SOF for MTL.

**Reproduction Rate**
100%

**Expected behavior**
Build to succeed.

**Impact**
Unable to use probes to get FW logs.

**Environment**
**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs

lgirdwood, 2023-03-06:
@mwasko fyi

kv2019i, 2023-03-07:
@abonislawski  A quick smoketest was succesful at least! As for the ifdefs, that's tricky as this is generic functionality and not really limited to Intel platforms. Maybe the IPC4 path can be limited to Zephyr and native drivers only.

lrudyX, 2023-03-24:
Change priority  to P2. Optional debug feature