thesofproject/sof #7225: [BUG][Regression] Failed to compile topologies

dbaluta, 2023-03-07:
**Describe the bug**

Compiling the topologies I get the following error:

```
$  ./scripts/build-tools.sh 

541/562] Generating sof-tgl-rt711-rt1316-rt714.tplg
Unable to process include file 
Unable to process include file 
Failed to process includes
Failed to process conditional includes in input config
FAILED: topology/topology2/avs-tplg/sof-tgl-rt711-rt1316-rt714.tplg /home/nxf25322/work/sof_dir/sof/tools/build_tools/topology/topology2/avs-tplg/sof-tgl-rt711-rt1316-rt714.tplg 
cd /home/nxf25322/work/sof_dir/sof/tools/build_tools/topology/topology2/avs-tplg && cat /home/nxf25322/work/sof_dir/sof/tools/build_tools/topology/topology2/avs-tplg/../abi.conf /home/nxf25322/work/sof_dir/sof/tools/topology/topology2/avs-tplg/../cavs-sdw.conf > sof-tgl-rt711-rt1316-rt714.conf && ALSA_CONFIG_DIR=/home/nxf25322/work/sof_dir/sof/tools/topology/topology2 alsatplg ${VERBOSE:+-v 1} -I /home/nxf25322/work/sof_dir/sof/tools/topology/topology2/avs-tplg/../ -D 'NUM_SDW_AMPS=2,SDW_DMIC=1,PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-sof-tgl-rt711-rt1316-rt714.bin' -p -c sof-tgl-rt711-rt1316-rt714.conf -o sof-tgl-rt711-rt1316-rt714.tplg
ninja: build stopped: subcommand failed.
```

The bug first appeared with commit:

```
commit 594e3f7da920b9308896bab84ee20894476ca98a (refs/bisect/bad)
Author: Yong Zhi <yong.zhi@intel.com>
Date:   Wed Jan 25 17:12:13 2023 -0600

    topology2: cavs-rt5682: add reference capture for MAX98357A speaker amp
    
    Use lbm_mode to loopback playback data for PCM27.
    Add copier module in DAI BE for demux function.
    
    Signed-off-by: Yong Zhi <yong.zhi@intel.com>
```

**To Reproduce**

$  ./scripts/build-tools.sh 

**Reproduction Rate**
Always

**Expected behavior**

no compilation error

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).

    * SOF: b637889ef  platform: remove support for cAVS 2.0 platforms

dbaluta, 2023-03-07:
[error_log.txt](https://github.com/thesofproject/sof/files/10910381/error_log.txt)

plbossart, 2023-03-07:
@dbaluta what happens if you use the docker image? the topology stuff is very dependent on alsa-lib/util versions.

dbaluta, 2023-03-07:
@plbossart I've upgrader the alsa-utils and it works now.

closed 2023-03-07 by dbaluta