thesofproject/sof #7228: [BUG][MTL-Chromebook][MTL-004] IPC timeout when set pipeline state to running

aiChaoSONG, 2023-03-08:
**Describe the bug**
The DMIC capture pipeline on MTL chromebook cannot work, we see ipc timeout for setting pipeline state to running IPC.

**To Reproduce**
do recording through DMIC pipeline.

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {8d5df0f5a from chrome kernel}
    * SOF: {MTL-004 release, tag: releases/mtl/v4.0}
2) Name of the topology file
    * Topology: {sof-mtl-max98360a-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL Chromebook}

```
[   89.493528] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[   89.998215] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000004|0x0
[   90.006716] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[   90.016161] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x0|0x0, target: 0x33000000|0x0|0x0, ctl: 0x3
[   90.028894] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[   90.038132] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   90.047562] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[   90.053983] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[   90.062642] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[   90.071002] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[   90.081211] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[   90.089184] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   90.098422] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set state 4 for pipeline 14
[   90.107366] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[   90.118354]  DMIC: ASoC: trigger FE cmd: 1 failed: -110
[   90.124197] sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 99 dir 1 cmd 0

```

keqiaozhang, 2023-03-08:
This issue also can be reproduced on MTL-RVP. I filed a similar issue https://github.com/thesofproject/sof/issues/7231

jairaj-arava, 2023-03-08:
@aiChaoSONG I see the same issue on the device at my end with MTL-004 release with both DMIC and Headset MIC.

**DMIC:** 
[  187.862623] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13020003|0x0
[  187.871121] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  187.880602] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x13020003|0x0|0x0, target: 0x33000000|0x0|0x0, ctl: 0x3
[  187.893337] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  187.902574] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  187.912003] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  187.918424] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  187.927272] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  187.935665] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  187.946010] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  187.954014] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  187.963253] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to pause pipeline 14
[  187.971227] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  187.982218]  DMIC: ASoC: trigger FE cmd: 1 failed: -110

**Headset:**
[   74.643167] init: ml-service (mojo_service) post-start process (5057) terminated with status 2
[   77.403116] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13020003|0x0
[   77.411612] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[   77.421065] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x13020003|0x0|0x0, target: 0x33000000|0x0|0x0, ctl: 0x3
[   77.433823] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[   77.443060] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   77.452491] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[   77.458911] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[   77.467569] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[   77.475933] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[   77.486217] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[   77.494192] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   77.503429] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to pause pipeline 6
[   77.511318] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[   77.522375]  Headset: ASoC: trigger FE cmd: 1 failed: -110

abonislawski, 2023-03-09:
Analyzed today the problem with -d option, we are crashing because of wrong clock selected for this load and this results in timer overruns. Without -d option DSP can still handle it and crashing only with asserts enabled.

However, the clock selection mechanism and manifest data are correct.
**The true root cause is incorrect data in IPC for gain module.** For gain FW receives 0 CPC instead of 10183 from manifest and this results in wrong calculations for clock selection.
All other modules receives normal CPC.

mengdonglin, 2023-03-22:
Closing as not reproduced on Chrome Rex.

aiChaoSONG, 2023-03-22:
can not be reproduced on Intel RVP