thesofproject/sof #7235: [BUG] Microphone not working on Fujitsu LIFEBOOK U9312

dvandok, 2023-03-08:
Installation of Debian 12 (bookworm, testing) on laptop with manually adding latest SOF firmware results in a working audio set up with the exception of the microphone.

I've followed guidance on https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#digital-mic-issues to replace the topology file and use PDM1 and that worked.

Files were taken from sof-v2.24.

Not sure if this needs to be tracked for every type of device or if another solution is in the works. In any case, the hardware
report is here:

https://linux-hardware.org/?probe=19a72f502b

Here is a snippet from syslog, let me know if it helps to add more detail.
```
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware: direct-loading firmware intel/sof/sof-adl.ri
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware: direct-loading firmware intel/sof-tplg/sof-hda-generic-2ch-pdm1.tplg
mrt 08 10:01:02 bobo kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:26:0 Kernel ABI 3:23:0
```

dvandok, 2023-03-08:
would add the DMIC-PDM1 label but I cannot

plbossart, 2023-03-08:
@dvandok thanks for reporting this and confirming that the pdm1 workaround works. Unfortunately we have no means to detect this automagically at the kernel level.

what we could do though is add a set of udev rules to select this topology option by default on specific systems. This is something we've discussed in the past but never acted on, adding @lgirdwood and @kv2019i on this topic.

dvandok, 2023-03-09:
I fear this is a major usability issue for laptops and handheld devices as they are commonly used for voice and video calls. Is this topology tied to the specific sound chip, or could that even be different between different vendors? The issues page linked above mentions the way Chromebooks seem to handle this (always expose 4 channels, drop the unpopulated channels) and that it is not possible (yet) in upstream UCM/Linux. Would that be the way to solve it, though? I think this might be worth a bump in priority.

plbossart, 2023-03-09:
@dvandok i understand the frustration but we have no descriptors or indicators from PCI, ACPI or DMI telling us which PDM interface is used. The Chrome case is different, they always use 4 channels but have platform-specific UCM configurations that extract the correct channels.

We have zero information for plain Linux, so no amount of priority escalation will help. This problem affects specific OEMs more than others, my guess is that 95% of platforms rely on PDM0 and 5% on PDM1, if that.

closed 2025-07-24 by github-actions[bot]

calestyo, 2025-09-30:
Having the same on my Fujitsu U7512 ... the alternate file fixes it.

@plbossart anything new with that udev quirk rules? Would at least fix the issue for all systems where it's known and verified :-)

Could of course provide any identifying information should you ever do something like that. Would be quite neat, cause it actually took me quite a while to find the solution (not accounting the even longer time it took me to reach my laziness threshold 😇)

oh any btw:<br>
https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#digital-mic-issues
says: `options snd-sof-pci` but modinfo for that says that's deprecated and `snd_sof` shall be used.<br>
Perhaps the website should be aliged.