thesofproject/sof #7238: [FEATURE] RFC New SOF_SCHEDULE_COMM scheduler - IPC & IDC scheduling refactor

aborisovich, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
Feature is related to the IPC4 refactoring. Currently IDC for zephyr platforms uses zephyr [p4wq](https://docs.zephyrproject.org/apidoc/latest/structk__p4wq.html) abstraction to schedule IDC messages, while IPC4 for zephyr platforms uses SOF `list_item` primitive supported by [k__work__delayable](https://docs.zephyrproject.org/apidoc/latest/structk__work__delayable.html) zephyr abstraction.
Pointing out further issues:
* IPC synchronization is scattered among zephyr ipc driver and sof code
* IPC operates on `sof_task` abstraction, while IDC uses only [k_p4wq_work](https://docs.zephyrproject.org/apidoc/latest/structk__p4wq__work.html) as single unit of work (single message handling).
* IPC operates on EDF scheduler that had never been truly finished (since it lacks any deadlines calculation and does not provide any preemption).

Moreover, IPC4 code for Zephyr platforms is very scattered so understanding the implementation takes ages.

**Describe the solution you'd like**
Introduce new type of SOF scheduler - "comm_scheduler".
It would be responsible for scheduling IPC and IDC messages using zephyr treads, priorities, preemption and deadlines.
It would be a wrapper for  [p4wq](https://docs.zephyrproject.org/apidoc/latest/structk__p4wq.html) abstraction.
Each core would run its own "comm_scheduler" instance.
"comm_scheduler" would operate on `sof_task` abstraction while mapping those to underlying [k_p4wq_work](https://docs.zephyrproject.org/apidoc/latest/structk__p4wq__work.html) items managed by the  [k_p4wq_work](https://docs.zephyrproject.org/apidoc/latest/structk__p4wq__work.html).
New scheduler would work on the single [Zephyr preemptive thread](https://docs.zephyrproject.org/3.2.0/kernel/services/threads/index.html#thread-priorities) of the `priority index = 0`.
This would place it in the whole system as "middle priority tasks" scheduler (since the LL schedulers work on the [Zephyr cooperative threads](https://docs.zephyrproject.org/3.2.0/kernel/services/threads/index.html#thread-priorities) that have `priority index < 0` and the new [DP scheduler](https://github.com/thesofproject/sof/pull/7089) has `priority index > 0`.
Also this change would cause IDC messages to be assigned to `sof_task` abstraction so it may be managed by the generic scheduler interface.

**Model prepared**
Prepared a model of the proposed changes.
Model consists of the "class diagram like" static description of data structures along with comments on their runtime uses/behavior.
It describes the existing IPC scheduling mechanism and the new scheduler model.
There are also two "sequence diagram like" graphs describing a flows on the comm_scheduler instances.
One diagram shows scenario when the IPC message is received from Host by interrupt, IPC target is Core0 and the response send.
Second diagram shows a scenario when IPC message is received from Host by interrupt, IPC target is Core1, so the message is redirected to Core1 using IDC messaging. It also shows new improved synchronization mechanisms (synchronization of IPC messages with each other, and synchronization of IPC messages with IDC completion).
[IPC & IDC scheduling and execution refactor rev1.0.pdf](https://github.com/thesofproject/sof/files/10937619/IPC.IDC.scheduling.and.execution.refactor.rev1.0.pdf)

Please share your feedback on the implementation idea. In this issue I would like to discuss all possible matters and doubts before starting works on this feature. Moreover, I would like to create a reasonable roadmap of works to be done (how to split this on PRs).

marcinszkudlinski, 2023-03-10:
existing EDF scheduler may be removed when COMM is introduced, 
 - it is basically the same just based on zephyr WQ instead of P4_WQ. 
 - misleading name - does not really make EDF scheduling

mmaka1, 2023-03-10:
If both, the scheduling and the execution of tasks is run within the same execution context, "comm_thread" name of a participant might be better than "comm_scheduler". 

In the SOF case, (1) the "medium" group does not run till completion, the time slice is limited (it is called a "budget"). When the tasks from the "medium" group run out of budget, the "low" group starts. The budgets are reset every system tick (1 ms typically).

(2) The "medium" group may also run opportunistically when the "low" group is idle and there is still time available until the next system tick when the "high" group starts running.

kv2019i, 2023-03-14:
Looks great @aborisovich ! One thing I was still looking (might be there, just didn't caught my eye) is what is the primary motivation to use the SOF scheduler abstraction for this? Was the primary reason to make it easier to facilitate IPC3/non-Zephyr usage, or do we have more reasons? This would be good to write down as it's sure to come up in code reviews.

aborisovich, 2023-03-14:
> Looks great @aborisovich ! One thing I was still looking (might be there, just didn't caught my eye) is what is the primary motivation to use the SOF scheduler abstraction for this? Was the primary [...]

I've provided new section in the issue description for the motivation on using SOF scheduler.

lgirdwood, 2023-03-14:
@dbaluta I assume also good for NXP IPC on Zephyr ?
@aborisovich pls also looks for areas where we can align sof scheduler APIs to be more like Zephyr APIs i.e. any low hanging fruit. The general direction should be to align the scheduler API to be thin and map nicley to Zephyr.

lgirdwood, 2023-03-14:
@aborisovich one more thing, do you have an ETA ? Shall we say v2.6 (end of June) ?

lyakh, 2023-03-16:
Can we have an even higher level goal / design description? I.e. what design are we targeting? Like all IPCs should be processed in Zephyr thread context, with a certain (or varied?) priority,... All IDCs should also run in Zephyr threads. Which ones should preempt which other ones, what priorities should be, what should preempt what, what we need deadlines for etc.?

dbaluta, 2023-03-16:
> @dbaluta I assume also good for NXP IPC on Zephyr ?

@lgirdwood I'm trying to wrap my head around this. I still need to read this several times. Anyhow, we are happy to test any change.

From what I understand this has something to do with changing the way we schedule IPC messages comming from Host processor?

aborisovich, 2023-03-16:
> Can we have an even higher level goal / design description? I.e. what design are we targeting? Like all IPCs should be processed in Zephyr thread context, with a certain (or varied?) priority,... [...]

Unfortunately I think just as @dbaluta indicated:

> I'm trying to wrap my head around this. I still need to read this several times. (...)

Is unfortunately the best way to get the answers for the all questions. I can try to provide some answers to some questions here, but if further questions arise please try to get answers from the diagrams I uploaded. I'll do this in the form of Q&A list.

Q: How does the comm_scheduler looks like statically?
A: comm_scheduler is a one instance object for the whole system. Inside this instance we have a "set of mechanics" used by each Core. A single "set of mechanics" is:
* a list `comm_scheduler.sof_tasks` of `task` structs (that are IPC or IDC or AMS, have their UUIDs and custom callbacks for `run()`, `complete()` and `get_deadline()` fptrs).
* `comm_scheduler.sof_tasks` list has elements added by the order of occurrence (the underlying list of `comm_scheduler.message_queue` is not - elements of that list `k_p4wq_work` are added to specific group (prepended or appended). Groups are regions of the  `comm_scheduler.message_queue` which define the execution priority for items).
* a member `struct task* executed_task` pointing to currently executed `task` (`k_p4wq_work` task is actually executed but it maps back to `task` and calls its `task.run()` and `task.complete()` callbacks over the `static void comm_task_handler()` function).
* a single worker thread provided to k_p4wq implementation of `thread_priority=0` that picks up elements from `comm_scheduler.message_queue` list head.
* logic of providing group_indexes to newly created k_p4wq_work items that thanks to this will be placed in the proper order inside the `comm_scheduler.message_queue` (this happens inside the static `void comm_task_handler()` body. There are 3 groups - Group0 (highest priority) AMS `task` objects, Group1 for `task` objects of "ipc_task" type which is ipc response to host interrupt ipc message, and Group2: same as Group1 but those are ipc notifications (we will distinguish those by ipc header)).
* each `task` scheduled for execution may have CoreCount - 1 child task, that it waits for to complete on other cores.

Q: What is preemptible?
A: If the `task` will exceed deadline its thread priority will be lowered and it may be preempted until tasks from other schedulers are completed (eg. DP scheduler).

Q: Can `task` be rescheduled?
A: Yes, it can be rescheduled eg. when during sending IPC response to host, host register state does not allow incoming IPC messages. This will cause a `task` to be rescheduled (while its parent task may complete execution eg. IDC on other core).
Rescheduled task will be put again into the `comm_scheduler.message_queue` to his matching group, but this time it will be prepended to the group not appended as new tasks created for scheduling.

Q: How are deadlines calculated?
A: This is not implemented now but we are going to have predefined deadlines for the `task` type. Please correct me If I am wrong @marcinszkudlinski .

So... yeah. There is a lot to take.

aborisovich, 2023-03-16:
> @dbaluta I assume also good for NXP IPC on Zephyr ?
> @aborisovich pls also looks for areas where we can align sof scheduler APIs to be more like Zephyr APIs i.e. any low hanging fruit. The general direction should be to align the scheduler API to be thin and map nicely to Zephyr.

Unfortunately I did not see any low hanging fruits here. Well maybe except one that we will put existing zephyr_idc
that already uses k_p4wq into the new comm_scheduler that will use the same abstraction.
As I see it that different schedulers perform different tasks and are allowed to use different zephyr tools for underlying implementation. We can probably look for some common parts to make code smaller but doing it too forcefully will cause spaghetti code again.

aborisovich, 2023-03-16:
> one more thing, do you have an ETA ? Shall we say v2.6 (end of June) ?

This has lower priority than MTL beta, so no ETAs for now. Job this size I would expect to be done by the Autumn to be honest.

lyakh, 2023-03-17:
> I can try to provide some answers to some questions here, but if further questions arise please try to get answers from the diagrams I uploaded.

@aborisovich it still seems to me that all this information isn't answering my question. My question isn't about how this scheduler should work. It's about why we think that IPC and other messaging protocols need it? All the documentation I see here so far is built like: "here's a new scheduler, this is how it will work. And we want to use it for X." Whereas it seems to me, that a good approach would be: "we have services A, B, C, we're unhappy about how they're implemented now, and we think a good implementation should have features 1, 2, 3."
As a side note, not related to the main question above, but just as a reminder: we also had an idea of using zbus for them.

aborisovich, 2023-03-19:
> "we have services A, B, C, we're unhappy about how they're implemented now, and we think a good implementation should have features 1, 2, 3."

@lyakh I've tried my best to describe those things in the issue description, in those two sections:
* _Is your feature request related to a problem? Please describe._
* _Motivation to use SOF Scheduler abstraction_

They contain:

- description on how messaging services are currently implemented
- what we do not like about it and what are their downsides
- what we would like to imrpve/unify/refactor/take control of

abonislawski, 2023-03-20:
Example set of changes in Zephyr P4Q to support SOF_SCHEDULE_COMM:

- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/55983

This will allow to simplify P4Q changes and solve all @aborisovich requirements with few assumptions: group prepend/append done via group_id extending (like subgroups), FIFO order will be maintained because of SOF_SCHEDULE_COMM design (constant deadlines)

andyross, 2023-05-20:
Just so folks are aware, I NAK'd the specific set of changes to p4wq, they really don't match the metaphor of the data structure.  There are some suggestions there for how to get the same features in an app-managed compound data structure.

That said... upleveling this to SOF and not Zephyr: this protocol design seems to be getting extremely thick.  Having been in the IPC layer(s) recently, it seems like a better starting point might be some cleanup at the protocol layer?  Lots of the top level complexity seems poorly justified to me, and having to do surgery to the bottom-most bits to support the requirements of the knots at the top of the stack is likely to be counterproductive.

aborisovich, 2023-06-21:
> That said... upleveling this to SOF and not Zephyr: this protocol design seems to be getting extremely thick. Having been in the IPC layer(s) recently, it seems like a better starting point might be [...]

Sure, please submit RFC with idea on how to refactor the IPC protocol in SOF with some concrete design.
It would probably include strong refactor of IPC3 which nxp guys are using, I'm not sure how welcome they would be to this idea. In my opinion it would be best to avoid any refactor now in SOF just because we would like everyone to migrate to IPC4 and remove IPC3 completely. Then it would be much easier to clean up lost of facade code in commons.
However @andyross you are keen on changes to IPC3 and IPC4 then please do submit your RFC.

kv2019i, 2023-08-17:
@lgirdwood @mwasko Development of this has stopped, assigned this to Michal you and clearing the milestone for now until we have clarity on how this will be taken forward.