thesofproject/sof #7246: [FEATURE] Use beamforming algorithm in the sof-hda-dsp topologies

perexg, 2023-03-10:
We have the TDFB algorithm already implement. Would be possible to enable this processing in the sof-hda-dsp firmware / topologies pipes ? There is already hardware on the market with 4 microphones, so we can improve the microphone input on this hardware.

The necessary parameters may be set through the standard "mixer" controls (if there's not a better solution - e.g. ACPI).

perexg, 2023-03-10:
Thank you for this quick comment.

> the quality of the DMIC capture is far from great in Linux

Only note that I would keep the "raw" streaming mode for the eventual software processing. But for the normal use, it makes sense to improve things in the hardware DSP.

> IIRC the position of the microphones is included in the NHLT tables

Could you point me to the specification / documentation for this, please? I would like to consider to extend the user space API to export this information to the user space for the additional software DSP processing. My understanding is that the 3D position and microphone orientation (angles) should be provided for the (beamforming) DSP algorithms. Actually, we just provide API for the sound placement in the PCM API (like "rear left") which may not be sufficient.

> I think PulseAudio/PipeWire mostly downmix to 2channels

Adding @wtay (PipeWire main maintainer) for further comments, but I think that many applications expect only stereo input, so this is probably true.

perexg, 2023-03-10:
> The NHLT public spec is here: https://01.org/sites/default/files/595976_intel_sst_nhlt.pdf
> 
> Section 2.5 report the mic coordinates, but reading that part again I am not sure how it works if there are mics on the lid and keyboard parts, and we don't have the hinge angle....

It seems that the structure at page 17 explains that:

```C
typedef struct _VENDOR_MIC_CONFIG
{
    BYTE Type;
    BYTE Panel;
    WORD SpeakerPositionDistance;   // mm
    WORD HorizontalOffset;          // mm
    WORD VerticalOffset;            // mm
    BYTE FrequencyLowBand;          // 5*Hz
    BYTE FrequencyHighBand;         // 500*Hz
    SHORT DirectionAngle;           // -180 - + 180
    SHORT ElevationAngle;           // -180 - + 180
    SHORT WorkVerticalAngleBegin;   // -180 - + 180 with 2 deg step
    SHORT WorkVerticalAngleEnd;     // -180 - + 180 with 2 deg step
    SHORT WorkHorizontalAngleBegin; // -180 - + 180 with 2 deg step
    SHORT WorkHorizontalAngleEnd;   // -180 - + 180 with 2 deg step
} VENDOR_MIC_CONFIG;
```

The origin 3D point (0, 0, 0) is the display centre (page 20).

plbossart, 2023-03-10:
Right, but if you see Figure 20, not all microphones are located on the display, so the microphone at the bottom left is not at a constant position wrt the 3 others. 
Anyways, probably a 2rd order detail, there are lots of items to check before worrying about this....

perexg, 2023-03-10:
True, but I doubt that laptops have an open lid angle sensor. But this angle is usually around ~110 degrees, so it may be a constant for the initial implementation.

plbossart, 2023-03-10:
Newer laptops do have an angle reported, and there's even the term 'Posture' used to describe how the device is used (normal, tent, tablet, etc). but as far as I know this angle is not exposed by ISH so far in Linux.

lyakh, 2023-03-13:
> @lyakh did we add multi-capture in stable-2.2, or is this limited to MTL/topology2?

@plbossart don't see topology1 multi-capture having being added to any of the stable branches. We added mux/demux fixes in #6762 but I don't see any such topologies there now

singalsu, 2023-03-13:
The fixed beamformer need very accurate mic locations. I doubt an angle sensor in the lid is accurate enough to get e.g. 1 mm level precision for array (x,y,z) positions. Such would work better with adaptive beamformer. But it also needs to know the (az, el) direction(s) for distortion-less capture.

Also complex mechanics with lid & body adds up. But beamforming with the fixed array part channels with line array should work. This some user space service that can upload suitable settings to FW. The settings can be pre-compured like today or make the user space tools calculate them on the fly (ref matlab tool for tdfb).

singalsu, 2023-03-13:
Also one thing that could be done would be to add DRC to microphone capture pipeline, then making it louder would be safe. The current +20 dB boost with IIR high-pass is at about limit what can be done. Both DRC and TDFB need to be converted to IPC4 so they are not available in current git main.

kv2019i, 2025-12-17:
Triaging our open tickets, I noticed this is opened. @singalsu Can you confirm this is done?

I think the main implementation was in commit 707c0c5bcebbf10b289c0bdaa6bd628ccab7ce75 which was included since 2.11 release (sof-bin-2024.09 and newer).

FYI @perexg as reporter.