thesofproject/sof #7249: [BUG] cavs platform_init should not write to DSP shim registers in Zephyr build

kv2019i, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
Fix invalid hw access in platform_init() of intel/cavs platform_init in Zephyr builds.

See https://github.com/thesofproject/sof/issues/5602 and https://github.com/thesofproject/sof/issues/7248

This is not correct
``
»       io_reg_write(DSP_INIT_GENO,
»       »       GENO_MDIVOSEL | GENO_DIOPTOSEL);

»       io_reg_write(DSP_INIT_IOPO,
»       »       IOPO_DMIC_FLAG | IOPO_I2S_FLAG);

»       io_reg_write(DSP_INIT_ALHO,
»       »       ALHO_ASO_FLAG | ALHO_CSO_FLAG);

»       io_reg_write(DSP_INIT_LPGPDMA(0),
»       »       LPGPDMA_CHOSEL_FLAG | LPGPDMA_CTLOSEL_FLAG);
»       io_reg_write(DSP_INIT_LPGPDMA(1),
»       »       LPGPDMA_CHOSEL_FLAG | LPGPDMA_CTLOSEL_FLAG);
``

kv2019i, 2023-03-10:
Started working on this. Some of the registers are written in Zephyr platfom boot code, but not all, so we cannot simply remove this yet. I also didn't want to add ifdef ZEPHYR on individual register writes, but rather wait until we have this init done correctly in Zephyr main and then remove this. 

Also consider how to do this cleanly for intel/cavs. Once  @juimonen complete the runtime_pm refactoring, we are starting to get close to the point where we could add a thin "platform/cavs-zephyr" that gets rid of all platform specific code, or drop XTOS support altogether for intel/cavs in SOF main.

@lgirdwood @juimonen any thoughts?

lgirdwood, 2023-03-13:
@nashif fyi
@kv2019i I assume we have this flow in the Zephyr init and are executing it twice today ?

kv2019i, 2023-03-14:
@lgirdwood Nopes, we are missing bits on Zephyr side -> https://github.com/zephyrproject-rtos/zephyr/pull/55738

lgirdwood, 2023-05-11:
@kv2019i @juimonen any update ?

kv2019i, 2023-05-25:
Need to move to v2.7. I have a PR in Zephyr side, but I can't complete it in SOF2.6 timeline (Zephyr 3.4 feature freeze starts tomorrow).

closed 2023-06-06 by kv2019i