thesofproject/sof #7261: [FEATURE] Pipeline 2.0 with native IPC4, Module API and required windows features

marcinszkudlinski, 2023-03-13:
**Is your feature request related to a problem? Please describe.**
 
We're heading fast for firmware fully compatible with current windows driver

I order to achieve this: 
 - IPC4 was introduced, IPC3 was set as depreciated
 - SOF component API was set as depreciated, and "Module API" was introduced 
 - New way of scheduling (Data processing) was introduced (PR still open)
 
To fit the requirements into current SOF code:
 - the pipeline itself is still controlled in IPC3 way, using a wrapper/translator to be compatible with IPC4
 - the pipeline is still using component API only, a heavy and slow module adapter must be used
 
Till now such approach was fine but is getting out of hands. Is consuming a lot of memory, is slow and with high risk of regression in legacy platforms
 
**Describe the solution you'd like**
 
I think there's a right moment to introduce **PIPELINE 2.0**
 - using IPC4 natively
 - using module API as main API 
 - without any buffering between LL modules
 - with no cache invalidation (both for audio data and metadata) unless necessary
- with producer-consumer safe manageable DP Queues instead of simple buffers between chunks of LL 
- in case when modules are on separate cores - cache coherent shared DP-queue
 - and more - the list is not 100% complete
 
 
**Describe alternatives you've considered**
Keep modifying current pipeline implementation. 
I my opinion 
-	technical debt is now too high 
-	high risk of introducing regression in legacy platforms
 
**Additional context**

plbossart, 2023-03-13:
@marcinszkudlinski The list of suggested improvements seem appealing, we should always try to do better, but allow me to put on my 'bad cop' hat to help clarify the directions.

a) What exactly does "Till now such approach was fine but is getting out of hands. Is consuming a lot of memory, is slow"
You will have to bring a lot more data to make a case for a change. Most of the memory issues we'd had were because of very large 3rd party modules, it's not clear to me what the pipeline infrastructure costs. The slowness is not clear either.

b) The wording also suggests that it's not possible to support DP scheduling without changing the pipeline level, which is not something I've heard before.

c) "The right moment" looks to me as the "wrong time and wrong place". We are half-way through the IPC4 introduction, there's no way any engineering manager in their right mind would accept to change the plumbing and have to redo all the validation, unless there's a clear showstopper preventing us from reaching the committed goals.

d) Also before we talk about a flag day, have you considered breaking down this feature into smaller more achievable ones? Can we e.g. drop the component API and use the module API? what can be improve in steps to reduce the revalidation cost?

e) And last, please do not forget that there is no replacement for IPC3 for non-Intel platforms. It's not deprecated as you state it. IPC4 is actually the outlier, not the norm.

lgirdwood, 2023-03-14:
@marcinszkudlinski I've edited and added some check boxes above to make this like an Epic. It can link to smaller features/PRs if needed.
Some of the items above are WIP (module API) and some sound like a regression (the unnecessary cache invalidations), the DP queues are new alongside the single LL pipeline buffer (which I think could be made to the current struct buffer APIs). 

I would definitely suggest adding more context on the pipeline APIs you want to change/add/delete, this can make the topic easier to plan/align and move forward.

marcinszkudlinski, 2023-03-15:
Lets me clarify

- Pipeline memory consumption. There're buffers between LL modules, in fact not needed at all. Module adapter is even worse - it uses double buffering. And - double buffering in ModuleAdapter is not for nothing, I agree with ModuleAdapter author that probably there're no other way to "marry" component ifc with module ifc. In case of pipeline with many components processing 7.1 in 96Khz - there's no way to fit it in in memory. 
- IPC3 depreciation. Indeed it went a little to far. What I meant is - the new pipeline need to support full features from IPC4. In fact, IPC3 is not so different, it should be possible to keep compatibility with IPC3
- DP Scheduling. In current model of buffering it is hard (and risky) to make it thread/cache safe. I understand, however, we need DP soon so there's a temporary solution coming - with some limitations: DP may be used for components with module interface (using ModuleAdapter), and in front/back of each DP component there must be LL components.
- "have you considered breaking down this feature into smaller more achievable ones" Oh yes. That's why I started this discussion :) looking for ideas/constructive reviews.

lgirdwood, 2023-04-05:
> **Changing scheduler orientation from buffers to modules is a great opportunity to get rid of direction usage in IPC4.**

Added this to the item list.

plbossart, 2023-04-11:
@marcinszkudlinski changes to the module API are already massively invasive, can we try to make smaller steps and avoid combining changes to module, pipeline, scheduler in the same feature request? Things need to be phased and aligned both with validation resources and program timelines. thank you.