thesofproject/sof #7267: [FEATURE][IPC4] Adjust DSP frequency based on pipeline cost data (CPC/CPS)

kv2019i, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
The pipeline create messages provide enough data to describe the cycle budget for a particular pipeline and its configuration. DSP should use this information to adjust the DSP operation point on platforms that support this.

SOF 
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
One version outside main in
https://github.com/thesofproject/sof/commit/553b2a646bb1172d274429aefb8a64d2fd98256d

**Additional context**
Blocking merge of https://github.com/thesofproject/sof/pull/7233

FYI @abonislawski @mwasko @ujfalusi @lgirdwood @kfrydryx @pblaszko

kv2019i, 2023-08-08:
Implemented in https://github.com/thesofproject/sof/pull/6649 , closing.

kv2019i, 2023-08-09:
Sorry, was a bit too fast on this. We still need the pipeline changes in https://github.com/thesofproject/sof/pull/8019 to complete this.

kv2019i, 2023-09-25:
Assigning as per owner of https://github.com/thesofproject/sof/pull/8019

btian1, 2023-11-15:
As we all know, current MTL clock only have two options, 38.4mhz and 393.216mhz, seems we only need take Wov as 38.4mhz and others all run at max freq, we may implement this as now for current solution.

Later, we still need a way to adjust each core dsp clock based on overall pipeline performance when HW support 100/200, we can think about how to implement this in future.

is above thinking reasonable and applicable?

plbossart, 2023-11-15:
The WoV clock only makes sense if ONLY core0 is used.

Once you have more than 1 core used, then the assumption is that you need quite a bit of processing power.

I would be surprised anyways if we can select the clock individually for each core. IIRC this a global setting for all cores. Please check.

kv2019i, 2023-11-24:
This seems to need more time, pushing to v2.9.

kv2019i, 2024-03-04:
Implemented available and reviewed at https://github.com/thesofproject/sof/pull/8019 but merge blocked due to failing test cases. Pushing to 2.10.

kv2019i, 2024-04-24:
With #8019 merged, we can now close this.

closed 2024-04-24 by kv2019i