thesofproject/sof #7268: [BUG][IPC4] Incorrect CPC/CPS values for MTL

kv2019i, 2023-03-14:
**Describe the bug**
The CPS/CPS values for MTL are incorrect for multiple modules. Currently the DSP FW does not adjust DSP speed based on this data, but when this feature is enabled, the incorrect data will lead to running DSP at too low/high frequency, leading to either audio artifacts, or too high power consumption.

**To Reproduce**
Need in-progress code for FW -> https://github.com/thesofproject/sof/issues/7267

**Reproduction Rate**
100%

**Expected behavior**
Values should be closer to those used on mtl-004 branch of rimage

https://github.com/thesofproject/rimage/tree/mtl-004-drop-stable
```
96ed61016a6fb3551b6013ccbb0a4935ae494f07 (origin/mtl-004-drop-stable) mtl: fill in cps and cpc data for smart_amp_test  module
f59cf6fa05156ed9e5f404f2e599746d5e8573b5 mtl: fill in cps and cpc data for ASRC module
0bcef6fea4d9592d7d9ee4b5847c1960a97d6052 mtl: fill in subsequent measured cps and mcps data
```

FYI @abonislawski @mwasko @ujfalusi @lgirdwood @kfrydryx @pblaszko

kv2019i, 2023-03-14:
Note, updated titles, "struct ipc4_base_module_cfg" has only the CPC value, so CPS is not relevant here (unless it's used as a source to calculate CPC).

mengdonglin, 2023-07-31:
Will update CPC value based on MCPS values captured in CI test.

mengdonglin, 2023-08-16:
Move to v2.8.  
sof-test PR to measure performance in under review: https://github.com/thesofproject/sof-test/pull/1090
It will take weeks to run extensive test before we can update CPC values in manifest.

btian1, 2023-11-15:
https://github.com/thesofproject/sof-test/pull/1127

This is the PR under review, according to Liam's suggestion:
"Liam suggest to use max peak + 10% margin as clock scaling cannot bring much power saving unless using 38.4MHz `WoV` clock"

After this was merged, we can update part of module CPC based on CI performance data.

kv2019i, 2023-11-29:
Closing this bug. the CPC (cycle budget per audio module) values are now close enough and there are no kernel warnings from missing values with mainline firmware and topology anymore. Continue maintenance of accurate CPC data with https://github.com/thesofproject/sof/pull/8019 which is enabling DSP speed control driver by the cycle-budgets.

closed 2023-11-29 by kv2019i