thesofproject/sof #7284: [FEATURE] Mixer with different bit widths?

paulstelian97, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
Hello, I'm planning on using the pipeline from tools/topology/topology1/sof-imx8-compr-wm8960-mixer.m4. The pipeline works fine when only one input is used (any format) or if the PCM input is given in S32_LE format. It doesn't work if the PCM input is e.g. S16_LE and the compress pipeline is also active (the CODEC_ADAPTER component outputs solely in 32-bit format)

**Describe the solution you'd like**
I would like either a pipeline configuration or an enhancement to some component (the former is probably the appropriate one) which allows me to have proper playback in the specific situation.

**Describe alternatives you've considered**
I have attempted to add a volume control on the compress pipeline. While that would be useful independent of this issue, it doesn't solve the problem as the volume component forces format_in = format_out, so the issue still stands.

I have considered, but not actually attempted, to use the copier component to do the adjustment for the compress output before the mixer input. The reason I have not attempted it is that the component _appears_ to me to be IPC4-specific, but I'm working on an IPC3 platform.

**Additional context**
The current topology looks like this:

[image]

dbaluta, 2023-03-16:
@lgirdwood @plbossart @ranj063 

I wonder how Intel handles scenarios where input to a mixer have different formats (e.g S32_LE mixed with a S16_LE).

Maye mixer should fix this before calling mix_func?

iuliana-prodan, 2023-03-17:
> In two-PCMs situation it just works because you have the host component for both inputs, which has this internal adjustment internally automatically. In our situation this breaks because the codec_adapter component _doesn't_ make/support any such adjustments, and the mixer component fails to recognize that its bit width must come from the inflexible input.
> 
> I would expect in two-PCM situations the mixer to be either S16_LE or S32_LE, dependent on what file was played initially (and yes, I expect the S32_LE to be converted to S16_LE if the latter was played first, inside the corresponding host component)

@paulstelian97 even with PCM this is not working, at least not on i.MX:
```
root@imx8qmmek:~# aplay -Dhw:1,0 -f S16_LE -c 2 -r 48000 -d 30 -t raw /dev/urandom & aplay -Dhw:1,1 -f S16_LE -c 2 -r 48000 -d 10 -t raw /dev/urandom
[1] 711
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
root@imx8qmmek:~# 
root@imx8qmmek:~# 
root@imx8qmmek:~# 
root@imx8qmmek:~# aplay -Dhw:1,0 -f S16_LE -c 2 -r 48000 -d 30 -t raw /dev/urandom & aplay -Dhw:1,1 -f S32_LE -c 2 -r 48000 -d 10 -t raw /dev/urandom
[2] 719
Playing raw data '/dev/urandom' : [  103.524436] sof-audio-of-imx8 556e8000.dsp: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
Playing raw data '/dev/urandom' : Signed 32 bit Little Endian, Si[  103.534568] sof-audio-of-imx8 556e8000.dsp: HW params ipc failed for stream 0
gned 16 bit Little Endian, Rate 48000 Hz, Rate 48000 Hz, StereoSt[  103.547645] sof-audio-of-imx8 556e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_params on 556e8000.dsp: -22
ereo

[  103.563108]  PCM Deep Buffer: ASoC: error at __soc_pcm_hw_params on PCM Deep Buffer: -22
[  103.572069]  PCM Deep Buffer: ASoC: error at dpcm_fe_dai_hw_params on PCM Deep Buffer: -22
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (42666 42667)
PERIOD_SIZE: 2048
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (170666 170667)
BUFFER_SIZE: 8192
BUFFER_BYTES: 65536
TICK_TIME: 0
[1]   Done                    aplay -Dhw:1,0 -f S16_LE -c 2 -r 48000 -d 30 -t raw /dev/urandom
root@imx8qmmek:~#
```

lgirdwood, 2023-03-17:
@mwasko I think this use case is supported today with copier and mixer on IPC4 (and with deep buffer) ?

@paulstelian97 I wonder if It may be easier to add support in the iMX driver to do IPC4 as a lot of the newer processing features are landing on IPC4 ? There is alos a lot of Zephyr related integration happinging around IPC4 re PM and other RTOS utility/debug that will also land with IPC4

Fwiw @ujfalusi has now upstreamed all the generic IPC4 code into the kernel and my understanding is that it would only be iMX related adaptations required now.

paulstelian97, 2023-03-17:
I have lightly considered IPC4 but did not dive into it enough to figure out if it's practical to do it. That said I will keep it in the back of my head. It does seem like there is a bit of work to do that.

I'm going to assume that the copier component isn't usable with IPC3? It does look like that from however superficially I looked at it today. IPC4 will probably take its time to land in i.MX releases, so I was considering finding if there is a different option until that point.

@dbaluta I guess we should mark this scenario as unsupported for the moment, and keep it in mind for when we will eventually transition to IPC4.

dbaluta, 2023-03-17:
@paulstelian97 I think you should look at switching to IPC4. Care to have a look at this starting next week? 

I think Intel will concentrate only on IPC4 from now on. 

@lgirdwood what's your feeling about IPC3? Will be kept in the future? Are you planning to deprecated it.