thesofproject/sof #7285: [BUG] alsabat test failed on IPC4 nocodec platforms

keqiaozhang, 2023-03-16:
**Describe the bug**
This issue only happened on IPC4 nocodec platform with 821Hz sine wav. From the waveform,  it seems that one sample point is inverted. No such issue with 599 or 997 Hz

[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -p hw:sofnocodec,0 -c hw:sofnocodec,0 -C 2 -F 821

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: tip sof-dev
    * SOF: main
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg sof-adl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL/ADL IPC4 nocodec platforms

keqiaozhang, 2023-03-16:
We have a duplicate issue:https://github.com/intel-innersource/drivers.audio.ci.sof-framework/issues/379

lgirdwood, 2023-03-17:
The sine frequency should make no difference to the result, is the glitch periodic or just random in position/channel ?

keqiaozhang, 2023-03-20:
> The sine frequency should make no difference to the result, is the glitch periodic or just random in position/channel ?

Just random in position/channel, no rules.

lgirdwood, 2023-03-21:
@singalsu how can we make the CI testing more clever to reject glitches on the USB testing HW ?

lrudyX, 2023-03-24:
@keqiaozhang we need IPC logs with payloads to reproduce with FW tests, could you prepare?

keqiaozhang, 2023-03-29:
> anyone able to replicate this locally? I just can't reproduce in my upx...

I did a quick check on TGL-UP and ADLP-RVP nocodec devices, it seems that this issue no longer exists on these 2 platforms.

keqiaozhang, 2023-03-30:
This issue no longer exists in CI, closing this bug.

closed 2023-03-30 by keqiaozhang