thesofproject/sof #7298: [BUG][alsabat] glitches happened on MTL-HDA when testing headset recording

keqiaozhang, 2023-03-17:
**Describe the bug**
I'm enabling alsabat test for headset recording and found some glitches on MTL-HDA. The reproduction rate is about 50% with all three test frequencies(599,821 and 997).

[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -c hw:sofhdadsp,0 -p hw:CODEC,0 -C 2 -F 599
Need PR https://github.com/thesofproject/sof-test/pull/1012

**Reproduction Rate**
50%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: tip sof-dev
    * SOF: tip main
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_HDA

lgirdwood, 2023-03-17:
@keqiaozhang periodic issue with stream or just happens once in stream ? (if once when ?)

keqiaozhang, 2023-03-20:
> @keqiaozhang periodic issue with stream or just happens once in stream ? (if once when ?)

From my observation, this is not a periodic issue, it only occurs at the beginning.

keqiaozhang, 2023-03-22:
> Is the timing of the glitch from beginning always consistent ?

Yes, it's consistent and occurs within the first 1-2 milliseconds.
[image]

juimonen, 2023-03-28:
@keqiaozhang could you attach the mtrace also?

lgirdwood, 2023-03-29:
@singalsu fyi - your glith dtetector should be able to emit output if glitch is in initial periods (i.e. trigger start/release) errors.

singalsu, 2023-03-29:
> @singalsu fyi - your glitch dtetector should be able to emit output if glitch is in initial periods (i.e. trigger start/release) errors.

Currently it starts detect after stable signal level (waiting for ramp to complete) and analyze the FFTs after that. I would need to add a time domain glitch detector for the beginning part during ramp. The time domain is not as sensitive as FFT but it should work for glitches like this. I will try in next version for https://github.com/thesofproject/sof-test/pull/1016

lgirdwood, 2023-03-29:
> > @singalsu fyi - your glitch dtetector should be able to emit output if glitch is in initial periods (i.e. trigger start/release) errors.
> 
> Currently it starts detect after stable signal level (waiting for ramp to complete) and analyze the FFTs after that. I would need to add a time domain glitch detector for the beginning part during ramp. The time domain is not as sensitive as FFT but it should work for glitches like this. I will try in next version for [thesofproject/sof-test#1016](https://github.com/thesofproject/sof-test/pull/1016)

Thanks - this will save some triage time as these are two quite different bugs with different solutions when they occur.

keqiaozhang, 2023-04-03:
> @keqiaozhang could you attach the mtrace also?

**mtrace**
<details>
<code><pre>
<summary>mtrace </summary>
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: intel_adsp_gpdma_init: dma dma@7c000 initialized
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: intel_adsp_gpdma_init: dma dma@7d000 initialized
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: intel_adsp_gpdma_init: dma dma@7e000 initialized
[00:00:00.116,326] <inf> init: print_version_banner: FW ABI 0x301a000 DBG ABI 0x5003000 tags SOF:v2.5-stable-branch-85-g362a0781f20b zephyr:zephyr-v3.3.0-1778-g1851950977d5 src hash 0x17f5bfd1 (ref hash 0x17f5bfd1)
[00:00:00.116,350] <inf> clock: clock_set_freq: clock 0 set freq 400000000Hz freq_idx 2
*** Booting Zephyr OS build zephyr-v3.3.0-1778-g1851950977d5 ***
[00:00:00.116,593] <inf> main: main: SOF on intel_adsp_ace15_mtpm
[00:00:00.116,606] <inf> main: main: SOF initialized
[00:00:00.117,673] <inf> ipc: ipc_cmd: rx       : 0x44000000|0x31400008
[00:00:00.118,443] <inf> ipc: ipc_cmd: rx       : 0x44000000|0x3060004c
[00:00:00.407,810] <inf> ipc: ipc_cmd: rx       : 0x11000004|0x0
[00:00:00.407,836] <inf> pipe: pipeline_new: pipeline new pipe_id 0 priority 0
[00:00:00.408,430] <inf> ipc: ipc_cmd: rx       : 0x40000004|0x15
[00:00:00.408,471] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:00:00.408,531] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[00:00:00.408,548] <inf> ipc: dai_config: comp:0 0x40000 dai_config() dai type = 3 index = 0 dd 0x40112100
[00:00:00.408,581] <inf> pipe: pipeline_connect: comp:0 0x40000 connect buffer 0 as sink
[00:00:00.409,208] <inf> ipc: ipc_cmd: rx       : 0x40000010|0xa
[00:00:00.409,255] <inf> eq_iir: eq_iir_init: comp:0 0x100000 eq_iir_init()
[00:00:00.409,740] <inf> ipc: ipc_cmd: rx       : 0x44000010|0x30000058
[00:00:00.409,760] <inf> eq_iir: eq_iir_set_config: comp:0 0x100000 eq_iir_set_config()
[00:00:00.410,343] <inf> ipc: ipc_cmd: rx       : 0x11010002|0x0
[00:00:00.410,368] <inf> pipe: pipeline_new: pipeline new pipe_id 1 priority 0
[00:00:00.410,851] <inf> ipc: ipc_cmd: rx       : 0x40010004|0x10015
[00:00:00.410,905] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[00:00:00.411,516] <inf> ipc: ipc_cmd: rx       : 0x45000010|0x10004
[00:00:00.411,533] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:00:00.411,556] <inf> pipe: pipeline_connect: comp:0 0x100000 connect buffer 0 as sink
[00:00:00.411,571] <inf> pipe: pipeline_connect: comp:1 0x40001 connect buffer 0 as source
[00:00:00.412,271] <inf> ipc: ipc_cmd: rx       : 0x45000004|0x10
[00:00:00.412,288] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:00:00.412,311] <inf> pipe: pipeline_connect: comp:0 0x40000 connect buffer 0 as sink
[00:00:00.412,325] <inf> pipe: pipeline_connect: comp:0 0x100000 connect buffer 0 as source
[00:00:00.417,631] <inf> ipc: ipc_cmd: rx       : 0x13010003|0x0
[00:00:00.417,668] <wrn> ipc: ipc_pipeline_complete(): no scheduling component specified, use comp 262145
[00:00:00.418,251] <inf> ipc: ipc_cmd: rx       : 0x13010004|0x0
[00:00:00.418,318] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 7
[00:00:00.418,335] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa0113f40 0x201b0U priority 0 flags 0x0
[00:00:00.418,376] <inf> ll_schedule: zephyr_domain_register: zephyr_domain_register domain->type 1 domain->clk 3 domain->ticks_per_ms 38400 period 1000
[00:00:00.418,480] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40001 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:00.419,456] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40001 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:00.419,485] <inf> ipc: ipc_cmd: rx       : 0x13000003|0x0
[00:00:00.420,436] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40001 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:00.420,925] <inf> ipc: ipc_cmd: rx       : 0x13000004|0x0
[00:00:00.420,981] <inf> dai_comp: dai_capture_params: comp:0 0x40000 dai_capture_params() src_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
[00:00:00.421,031] <inf> eq_iir: eq_iir_prepare: comp:0 0x100000 eq_iir_prepare(), source_format=2, sink_format=2
[00:00:00.421,048] <inf> eq_iir: eq_iir_init_coef: comp:0 0x100000 eq_iir_init_coef(): 1 responses, 2 channels, stream 2 channels
[00:00:00.421,065] <inf> eq_iir: eq_iir_init_coef: comp:0 0x100000 eq_iir_init_coef(), ch 0 is set to response 0
[00:00:00.421,095] <inf> eq_iir: eq_iir_init_coef: comp:0 0x100000 eq_iir_init_coef(), ch 1 is set to response 0
[00:00:00.421,135] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 7
[00:00:00.421,151] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa0114940 0x201b0U priority 0 flags 0x0
[00:00:00.421,475] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40001 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:00.421,531] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to copy
[00:00:00.422,436] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40001 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:01.441,486] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 2429, max 4480, overruns 0
[00:00:01.448,515] <inf> ipc: ipc_cmd: rx       : 0x13000003|0x0
[00:00:01.448,530] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 2
[... cut ...]
[00:00:04.375,150] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 2
[00:00:04.375,846] <inf> ll_schedule: zephyr_ll_task_done: task complete 0xa0113f40 0x201b0U
[00:00:04.375,865] <inf> ll_schedule: zephyr_ll_task_done: num_tasks 1 total_num_tasks 1
[00:00:04.375,883] <inf> ll_schedule: zephyr_domain_unregister: zephyr_domain_unregister domain->type 1 domain->clk 3
[00:00:05.108,555] <inf> ipc: ipc_cmd: rx       : 0x13010002|0x0
Terminated108,580] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 0
</code></pre>
</details>

keqiaozhang, 2023-04-03:
> @keqiaozhang can you share the logs with payloads IPC logs.

**payloads IPC logs**
<details>
<code><pre>
<summary>mtrace </summary>
[  791.642352] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  791.642370] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  791.645914] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  791.645934] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  791.645985] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D0I0
[  791.645997] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 0 -> 2
[  791.646008] snd_sof:snd_sof_load_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: loading firmware
[  791.646017] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 2 -> 3
[  791.647071] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x1010000 successful
[  791.648144] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1d1c]=0x8151 successful
[  791.648171] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: booting DSP firmware
[  791.648185] snd_sof_intel_hda_common:hda_dsp_cl_boot_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: IMR restore supported, booting from IMR directly
[  791.649237] snd_sof_intel_hda_common:mtl_dsp_core_power_up: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000101 successful
[  791.649252] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-mtl 0000:00:1f.3: Primary core power up successful
[  791.649264] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x73214]=0x80000000 successful
[  791.649275] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x41 successful
[  791.649284] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x1 successful
[  791.763677] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx      : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  791.763700] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 3 -> 6
[  791.763717] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx done : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  791.763968] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: firmware boot complete
[  791.763988] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 6 -> 7
[  791.764005] snd_sof_intel_hda_common:mtl_enable_sdw_irq: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x114c]=0xf successful
[  791.764019] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  791.764497] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x31400008: MOD_LARGE_CONFIG_SET
[  791.764674] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  791.764689] Message payload: 00000000: 772861e4 ffedcdc9
[  791.764698] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  791.765180] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[  791.765367] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  791.765382] Message payload: 00000000: 00000100 00001000 00000001 000001ef
[  791.765388] Message payload: 00000010: 00000000 00000000 00000000 00000000
[  791.765392] Message payload: 00000020: 00000000 00000000 00000000 00000000
[  791.765396] Message payload: 00000030: 00000000 00000000 00000000 00000000
[  791.765399] Message payload: 00000040: 00000000 00000000 00000000
[  792.049739] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 1
[  792.049751] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period min 192 max 16384 bytes
[  792.049756] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period count 2 max 16
[  792.049760] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: buffer max 65536 bytes
[  792.050117] snd_sof_intel_hda_common:hda_calc_stream_format: sof-audio-pci-intel-mtl 0000:00:1f.3: format_val=0x11, rate=48000, ch=2, format=2
[  792.050132] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 1
[  792.050143] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  792.050165] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[  792.050172] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-mtl 0000:00:1f.3: period_bytes:0x2000
[... cut ...]
[  794.683528] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x32000000|0x0: GLB_DELETE_PIPELINE
[  794.683567] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x12010000|0x0: GLB_DELETE_PIPELINE
[  794.683582] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: widget pipeline.3 freed
[  794.683598] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: widget copier.host.3.1 freed
[  794.683993] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 0 dir 1
</code></pre>
</details>

fredoh9, 2023-04-12:
From today's daily, sh-mtlp-rvp-hda-04 is used

- check-alsabat-headset-capture-599.sh : pass
- check-alsabat-headset-capture-821.sh : failed
- check-alsabat-headset-capture-997.sh : failed

821 Hz has a (a few) imperfection in waveform

[image]

However I tried same build with another device, jf-mtlp-rvp-hda-2, all alsabat capture tests are passed.

fredoh9, 2023-04-13:
For today's daily test, sh-mtlp-rvp-hda-02 was used.

- check-alsabat-headset-capture-599.sh : failed
- check-alsabat-headset-capture-821.sh : pass
- check-alsabat-headset-capture-997.sh : pass

fredoh9, 2023-04-14:
today's daily test failed on sh-mtlp-rvp-hda-03.

tested on sh-mtlp-rvp-hda-03 with same alsa settings with jf-mtlp-rvp-hda-2
check-alsabat-headset-capture-599.sh failed 3 out of 5. not 100% failure. not alsa settings issue.

[image]
[image]
[bat_failed_sh-hda03.wav.gz](https://github.com/thesofproject/sof/files/11236747/bat_failed_sh-hda03.wav.gz)

fredoh9, 2023-04-18:
Yesterday @keqiaozhang checked on the device,

> It’s not a device configuration issue, there’re some distortions in the waveform. This issue also can be reproduced on other devices, the reproduction rate is about ~20%.

[image]

fredoh9, 2023-04-20:
Removed P1 as some of devices have not reported the issue recently. The other devices (multiple) have ~20% failure rate though.

fredoh9, 2023-04-21:
After compiling test results of +1 week, we have 5 devices for this model.

These two devices are solid pass for all,

- jf-mtlp-rvp-hda-2.jf.intel.com
- jf-mtlp-rvp-hda-3.jf.intel.com

Below 3 has ~20% failure, I don't see any failure pattern,

- sh-mtlp-rvp-hda-04.sh.intel.com
- sh-mtlp-rvp-hda-02.sh.intel.com
- sh-mtlp-rvp-hda-03.sh.intel.com

keqiaozhang, 2023-04-24:
> After compiling test results of +1 week, we have 5 devices for this model.

Thanks Fred for debugging, I will try to figure it out.

keqiaozhang, 2023-05-15:
Confirmed that this is a Hardware specific issue and only happens on SH DUTs. I'm debugging it.

keqiaozhang, 2023-05-16:
Issue fixed after connecting the stereo USB sound card to other USB port on MB. It seems that using USB HUB to power up the stereo sound card is not very stable.

closed 2023-05-16 by keqiaozhang