thesofproject/sof #7308: simultaneous playback of multiple streams with MUX component not working with latest sof FW (20-march-2023)

maruthi-machani, 2023-03-21:
With latest sof FW (20-march-2023), simultaneous playbacks of 2 input streams using mux component only single stream is working. no audio playback observed on the second stream. Earlier sof FW (02-March-2023) simultaneous playback was working properly.

plbossart, 2023-03-21:
"simultaneous playbacks of 2 input streams"

You already lost me here @maruthi-machani, no idea how you managed to have 'playback' and 'input streams' in the same sentend.

Please provide clearer descriptions of 
- what platform
- what topology
- what test
you are running.

maruthi-machani, 2023-03-23:
with AMD platform, when simultaneous two input streams are played using mux component only single input stream is played. this issue is observed with the latest sof FW.
Topology uses mux component with two input streams. Attaching topology for reference.
[sof-acp-rmb-tdm8ch-dmic4ch_m4.txt](https://github.com/thesofproject/sof/files/11046953/sof-acp-rmb-tdm8ch-dmic4ch_m4.txt)
[pipe-echo-ref-dai-playback_m4.txt](https://github.com/thesofproject/sof/files/11046954/pipe-echo-ref-dai-playback_m4.txt)

dbaluta, 2023-03-23:
@maruthi-machani I bet the problem start at this commit:

commit c399624fcfb7dfaf75d69671bc2a91ec9812a6ad
Author: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
Date:   Tue Dec 20 18:48:10 2022 +0200

    Audio: Mux: Convert mux and demux to module adapter

We have similar issue but with MIXER.

plbossart, 2023-03-23:
> **Usecase**: simultaneous playback of two input streams on headset and speaker.

Did  you mean: playback of one stream on both headset and speaker? 

It's very hard to parse your explanations @maruthi-machani

maruthi-machani, 2023-03-24:
simultaneously playing two different streams one on headset and other on speaker.

plbossart, 2023-03-24:
Never mind, it's described in the topology where a TDM interface is used.

````
# PCM0 ----> buffer ------+
#                         |----> muxdemux ----> buffer ---->  ACPSP_play
# PCM1 ----> buffer ----->
````

plbossart, 2023-03-24:
@maruthi-machani this isn't a topology that's used on the Intel devices, headset and amplifiers use separate I2S links.

maruthi-machani, 2023-03-27:
@plbossart irrespective of intel/amd platforms the mux component should support TDM of multiple input streams. this specific issue is observed recently. earlier (before march 6 sof) this issue is not observed.

bhiregoudar, 2023-05-05:
Hi Liam,
Please assign this issue to proper author

lgirdwood, 2023-05-11:
@dbaluta @bhiregoudar we dont have any working Intel IPC3/xtos configs in main now, but I think we can probably shed some light on this via the ALSA plugin (in the lrg/topic/plugin) branch. i.e. we can run the topology on host PC.

I think in the short term we can get the plugin running in CI in IPC3 mode, but we will probably need some help from NXP/AMD to do this. ALSA plugin would give us some IPC3 & IPC4 coverage on CI. Ideally we would have a UT for each module/component type.

maruthi-machani, 2023-05-17:
@lgirdwood could you please let us know what kind of support you need from AMD.

lgirdwood, 2023-06-07:
@maruthi-machani apologies, was away. We may need AMD help next week to test. I going to ask @lyakh to deep dive. In the mean time can AMD provide latest FW and kernel logs (inc IPC messages) of failure. Thanks.

maruthi-machani, 2023-06-13:
@lgirdwood Attaching the FW and kernel logs (inc IPC messages) during failure.

[debuglog.txt](https://github.com/thesofproject/sof/files/11732821/debuglog.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/11732822/dmesg.log)

maruthi-machani, 2023-06-14:
@lyakh , failure is observed with patch c399624fcfb7dfaf75d69671bc2a91ec9812a6ad. by reverting the patch failure not observed.

[image]

singalsu, 2023-06-14:
dmesg.txt shows "oading topology:amd/sof-tplg/sof-rmb-rt5682s-rt1019.tplg" but it does not seem to contain a mux:

[image]

maruthi-machani, 2023-06-15:
we rename the topology "sof-acp-rmb-tdm8ch-dmic4ch-rt5682-rt1019.tplg" to "sof-rmb-rt5682s-rt1019.tplg" when placing in the target.
mux initialization from above debuglog.txt

[image]

singalsu, 2023-06-15:
@maruthi-machani can you please try if draft fix https://github.com/thesofproject/sof/pull/7811 works for you?

maruthi-machani, 2023-06-15:
with above fix, simultaneous playback of two streams with MUX component is working fine.

PR #7811 (2023-06-16): Audio: Mux: Fix mistake in frames count handling
The source frames count need to be checked from every active input_buffers[n].size). Similarly in consuming the source buffers it needs to be updated to every
mod->input_buffers[n].consumed.

Fixes: #7308
commit afd86e08 (2023-06-16): Audio: Mux: Fix mistake in frames count handling

The source frames count need to be checked from every
active input_buffers[n].size). Similarly in consuming the
source buffers it needs to be updated to every
mod->input_buffers[n].consumed.

Fixes: #7308
Fixes: c399624 ("Audio: Mux: Convert mux and demux to module adapter")
PR #7821 (2023-06-16): [sof-2.6] Audio: Mux: Fix mistake in frames count handling
The source frames count need to be checked from every active input_buffers[n].size). Similarly in consuming the source buffers it needs to be updated to every
mod->input_buffers[n].consumed.

Fixes: #7308
Fixes: c399624 ("Audio: Mux: Convert mux and demux to module adapter")

(cherry picked from commit afd86e08081835b16c960713ae2f44ffd3654ecb)
commit d03ebade (2023-06-16): Audio: Mux: Fix mistake in frames count handling

The source frames count need to be checked from every
active input_buffers[n].size). Similarly in consuming the
source buffers it needs to be updated to every
mod->input_buffers[n].consumed.

Fixes: #7308
Fixes: c399624 ("Audio: Mux: Convert mux and demux to module adapter")

(cherry picked from commit afd86e08081835b16c960713ae2f44ffd3654ecb)
closed 2023-06-16 by lgirdwood