thesofproject/sof #7317: LLVM C Compiler support for building SOF with Zephyr

nishalkulkarni, 2023-03-21:
While I understand there's no official support for LLVM C compiler, I've tried to compile SOF using Espressif's version of LLVM/Clang containing patches for the Xtensa architecture. After a few modifications(mostly in Zephyr repository, containing Xtensa assembly instructions) I did manage to compile SOF successfully with Clang. 

However, trying to test this out on real hardware(UP Xtreme i11 - Intel Tigerlake Platform) gives an error. I'd like some help understanding this error, and where the possible problems might be...

Command used for building with Clang-14
Some details 
* target - xtensa-esp32-elf (I know this is not the platform on which I test, more on this below)
* I use GCC's assembler as currently, the Clang one does not support all required Xtensa assembly instructions
* Clang's version of binutils
```
west -v -v build --build-dir build-tgl-llvm --pristine always --board intel_adsp_cavs25 ./sof/app -- -DZEPHYR_TOOLCHAIN_VARIANT=llvm -DLLVM_TOOLCHAIN_PATH=~/Projects/esp-clang-14 -DCMAKE_C_FLAGS="-isystem /home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf/include/ -target xtensa-esp32-elf -fno-short-wchar -Wno-unused-command-line-argument" -DUSE_CCACHE=0 -DCMAKE_ASM_COMPILER="/home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc" -DCMAKE_ASM_FLAGS="-mtext-section-literals" -DCMAKE_EXE_LINKER_FLAGS="-Wl,--print-memory-usage -Wl,--sort-common=descending -L/home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/../lib/gcc/xtensa-intel_s1000_zephyr-elf/12.1.0" -DCMAKE_OBJDUMP="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-objdump" -DCMAKE_STRIP="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-strip" -DCMAKE_OBJCOPY="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-objcopy" -DCMAKE_AR="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-ar" -DCMAKE_RANLIB="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-ranlib" -DCMAKE_READELF="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-readelf" -DCMAKE_LINKER="~/Projects/esp-clang-14/bin/xtensa-esp32-elf-ld.bfd"
```

While this builds/compiles the project successfully, after signing and running on real hardware I get the below error
```
Mär 20 20:49:37 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Mär 20 20:49:37 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
Mär 20 20:49:37 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:0:0-03b2f
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:26:0 Kernel ABI 3:21:0
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW ABI is more recent than kernel
Mär 20 20:49:38 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware boot failure due to timeout
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: invalid header size 0xf010e0e. FW oops is bogus
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00000000
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -5
Mär 20 20:49:40 sra-upxi11 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -5
```

If I understand it correctly the Core ISA part of Tigerlake is identical to esp32, e.g. has the same features like floating point, code density extension, loops, etc. I was recently told that using GNU binutils for linker might work for generating a working binary for the Tigerlake platform

So I tried the below command, but still get the same error as above.
this still uses LLVM as compiler but uses GNU binutils
```
west -v -v build --build-dir build-tgl-llvm --pristine always --board intel_adsp_cavs25 ./sof/app -- -DZEPHYR_TOOLCHAIN_VARIANT=llvm -DLLVM_TOOLCHAIN_PATH=~/Projects/esp-clang-14 -DCMAKE_C_FLAGS="-isystem /home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf/include/ -target xtensa-esp32-elf -fno-short-wchar -Wno-unused-command-line-argument -Qunused-arguments" -DUSE_CCACHE=0 -DCMAKE_ASM_COMPILER="/home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc" -DCMAKE_ASM_FLAGS="-mtext-section-literals" -DCMAKE_EXE_LINKER_FLAGS="-fuse-ld=bfd -gdwarf-4 -Wl,--sort-common=descending -L/home/2022/nis.kulkarni/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/../lib/gcc/xtensa-intel_s1000_zephyr-elf/12.1.0" -DCMAKE_OBJDUMP="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-objdump" -DCMAKE_STRIP="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-strip" -DCMAKE_OBJCOPY="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-objcopy" -DCMAKE_AR="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-ar" -DCMAKE_RANLIB="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-ranlib" -DCMAKE_READELF="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-readelf" -DCMAKE_LINKER="~/zephyr-sdk-0.15.2/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-ld.bfd"
```

Note: the modifications to the codebase aren't the problem as compiling them with GCC produces a working image.

plbossart, 2023-03-21:
Who knows what the configuration differences between esp32 and TGL might be. If esp32 has one option that TGL doesn't have, there would be an instruction that's not supported ->  panic?

In general the odds of successfully using a Tensilica configuration for a different hardware are very slim.

aiChaoSONG, 2023-03-22:
Did you sign the firmware with rimage? west build only build the raw image, it probably has the wrong sof_ext_man header.

nishalkulkarni, 2023-03-22:
@aiChaoSONG I did sign the image, the command used:

```
west sign --build-dir ./build-tgl-llvm -t rimage --tool-path ./build-rimage/rimage --tool-data ./sof/rimage/config -- -k ./sof/keys/otc_private_key_3k.pem
```

aiChaoSONG, 2023-03-23:
Image loading stopped at extended manifest parsing, which I think should not happen if the image is properly signed.

closed 2025-07-24 by github-actions[bot]