thesofproject/sof #7327: [BUG] ipc timed out for 0x13000004|0x1 GLB_SET_PIPELINE_STATE  when testing multiple-pause-resume on TGL-IPC4-NOCODEC

keqiaozhang, 2023-03-24:
**Describe the bug**
Observed this issue once in CI daily test. IPC timed out 0x13000004|0x1 GLB_SET_PIPELINE_STATE. 
There's a error in mtrace:
```
[   70.300971] <inf> smart_amp_test: comp:5 0x90000 smart_amp_trigger(), command = 7
[   70.300985] <inf> dai_intel_ssp: dai_ssp_get_properties dai_index 0
[   70.300993] <inf> dai_intel_ssp: dai_ssp_get_properties fifo 487440
[   70.301001] <inf> dai_intel_ssp: dai_ssp_get_properties handshake 2
[   70.301010] <inf> dai_intel_ssp: dai_ssp_get_properties init delay 0
[   70.301023] <inf> dai_intel_ssp: dai_ssp_pre_start
[   70.301043] <inf> smart_amp_test: comp:5 0x90000 smart_amp_trigger(), command = 1
[   70.301081] <inf> dai_intel_ssp: dai_ssp_start
[   70.301091] <inf> dai_intel_ssp: dai_ssp_update_bits base 77000, reg 30, mask 100, value 100
ASSERTION FAIL [aligned_addr == addr] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hdTerminated
```

**dmesg**
```
[ 7750.590558] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[ 7750.591278] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[ 7750.591311] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[ 7750.591315] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[ 7751.079650] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 27 dir 1 cmd 0
[ 7751.079665] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 7751.095028] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[ 7751.095045] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 7751.095050] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 7751.095071] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 7751.095078] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 7751.095101] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 7751.095109] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 7751.095115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 7751.095120] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 7751.095125] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 7751.095141] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 7751.095174] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 7751.095180] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 7751.095222] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 7751.095235] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 7751.095244] kernel:  Port0 Aux Playback: ASoC: trigger FE cmd: 1 failed: -110
```

**To Reproduce**
~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
TBD

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[7abb3259b5f0](https://github.com/thesofproject/linux/commit/7abb3259b5f0)}
    * SOF: {[56abcade565e](https://github.com/thesofproject/sof/commit/56abcade565e)}
2) Name of the topology file
    * Topology: {avs-tplg/sof-tgl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGLU_RVP_NOCODEC_IPC4ZPH}

[dmesg.txt](https://github.com/thesofproject/sof/files/11060387/dmesg.txt)

lyakh, 2023-03-24:
Looks like a duplicate of #7191

keqiaozhang, 2023-04-24:
Closing this bug, this issue is not reproducible in CI and the duplicate issue https://github.com/thesofproject/sof/issues/7191 was closed.

closed 2023-04-24 by keqiaozhang